Koenkk/zigbee2mqtt #2763: brightness_move only works in one direction

merlinschumacher, 2020-01-16:
## What happened
I used `brightness_move` to raise or lower the brightness of a bulb. After choosing one direction (eg. raise) zigbee2mqtt will always choose to raise the brighness, even if i specify a negative value. This also happens the other way around. The only workaround is to reset the brightness to an absolute value. 

## What did you expect to happen
`brightness_move` should raise or lower the brightness according to the given numerical value

## How to reproduce it (minimal and precise)
Send `{"brightness_move": 10}` to a lamp, followed by  `{"brightness_move": stop}`. Then send `{"brightness_move": -10}` and the brightness will continue to rise. 

## Debug Info
zigbee2mqtt version: 1.9.0
CC253X firmware version: 20190619 source routing

bifi2090, 2020-01-17:
I just can approve that too. After updating to 1.9.0 brightness_move only works in positive direction.

Koenkk, 2020-01-17:
Should be fixed in latest dev, please let me know.

bifi2090, 2020-01-18:
Latest dev fixed that Problem for me. Thx

Koenkk, 2020-01-18:
Thanks for confirming!

closed 2020-01-18 by Koenkk