Koenkk/zigbee2mqtt #2764: Interviews failing for Sercomm SZ-DWS04 magnetic contact sensors

luckypants123, 2020-01-17:
Hi,

I have paired a Sercomm SZ-DWS04 magnetic contact sensor about a month ago - works great!

From log:
```
zigbee2mqtt:info  2020-01-16 21:02:02: garage_door_sensor (0x000d6f000fdb0453): SZ-DWS04 - Sercomm Magnetic door & window contact sensor (EndDevice)
```
and
```
zigbee2mqtt:info  2020-01-16 21:02:05: MQTT publish: topic 'zigbee2mqtt/garage_door_sensor', payload '{"contact":true,"tamper":false,"battery_low":false,"linkquality":0}'
```

I have tried to pair a couple more today without luck...hoping its just i'm missing something simple.

Pairing fails with the below in the log:
```
zigbee2mqtt:info  2020-01-16 21:05:06: Device '0x000d6f000ed8cb64' joined
zigbee2mqtt:info  2020-01-16 21:05:06: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x000d6f000ed8cb64"}}'
zigbee2mqtt:info  2020-01-16 21:05:06: Starting interview of '0x000d6f000ed8cb64'
zigbee2mqtt:info  2020-01-16 21:05:06: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x000d6f000ed8cb64"}}'
zigbee2mqtt:error 2020-01-16 21:05:18: Failed to interview '0x000d6f000ed8cb64', device has not successfully been paired
zigbee2mqtt:info  2020-01-16 21:05:18: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x000d6f000ed8cb64"}}'
```
After a few minutes:
```
zigbee2mqtt:info  2020-01-16 21:08:19: Configuring '0x000d6f000ed8cb64'
zigbee2mqtt:info  2020-01-16 21:08:19: MQTT publish: topic 'homeassistant/binary_sensor/0x000d6f000ed8cb64/contact/config', payload '{"payload_on":false,"payload_off":true,"value_template":"{{ value_json.contact }}","device_class":"door","state_topic":"zigbee2mqtt/0x000d6f000ed8cb64","json_attributes_topic":"zigbee2mqtt/0x000d6f000ed8cb64","name":"0x000d6f000ed8cb64_contact","unique_id":"0x000d6f000ed8cb64_contact_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x000d6f000ed8cb64"],"name":"0x000d6f000ed8cb64","sw_version":"Zigbee2mqtt 1.8.0","model":"Magnetic door & window contact sensor (SZ-DWS04)","manufacturer":"Sercomm"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2020-01-16 21:08:19: MQTT publish: topic 'homeassistant/sensor/0x000d6f000ed8cb64/temperature/config', payload '{"unit_of_measurement":"°C","device_class":"temperature","value_template":"{{ value_json.temperature }}","state_topic":"zigbee2mqtt/0x000d6f000ed8cb64","json_attributes_topic":"zigbee2mqtt/0x000d6f000ed8cb64","name":"0x000d6f000ed8cb64_temperature","unique_id":"0x000d6f000ed8cb64_temperature_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x000d6f000ed8cb64"],"name":"0x000d6f000ed8cb64","sw_version":"Zigbee2mqtt 1.8.0","model":"Magnetic door & window contact sensor (SZ-DWS04)","manufacturer":"Sercomm"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2020-01-16 21:08:19: MQTT publish: topic 'homeassistant/sensor/0x000d6f000ed8cb64/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/0x000d6f000ed8cb64","json_attributes_topic":"zigbee2mqtt/0x000d6f000ed8cb64","name":"0x000d6f000ed8cb64_battery","unique_id":"0x000d6f000ed8cb64_battery_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x000d6f000ed8cb64"],"name":"0x000d6f000ed8cb64","sw_version":"Zigbee2mqtt 1.8.0","model":"Magnetic door & window contact sensor (SZ-DWS04)","manufacturer":"Sercomm"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2020-01-16 21:08:19: MQTT publish: topic 'homeassistant/sensor/0x000d6f000ed8cb64/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x000d6f000ed8cb64","json_attributes_topic":"zigbee2mqtt/0x000d6f000ed8cb64","name":"0x000d6f000ed8cb64_linkquality","unique_id":"0x000d6f000ed8cb64_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x000d6f000ed8cb64"],"name":"0x000d6f000ed8cb64","sw_version":"Zigbee2mqtt 1.8.0","model":"Magnetic door & window contact sensor (SZ-DWS04)","manufacturer":"Sercomm"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2020-01-16 21:08:24: Succesfully configured '0x000d6f000ed8cb64'
```
After a restart it shows up as a joined device next to the pre-existing one:
```
zigbee2mqtt:info  2020-01-16 21:10:18: garage_door_sensor (0x000d6f000fdb0453): SZ-DWS04 - Sercomm Magnetic door & window contact sensor (EndDevice)
zigbee2mqtt:info  2020-01-16 21:10:18: 0x000d6f000ed8cb64 (0x000d6f000ed8cb64): SZ-DWS04 - Sercomm Magnetic door & window contact sensor (EndDevice)
```
However later in the log we see the output is off. Have a look at the previous sensor, and then the newly paired one:
```
zigbee2mqtt:info  2020-01-16 21:10:21: MQTT publish: topic 'zigbee2mqtt/garage_door_sensor', payload '{"contact":true,"tamper":false,"battery_low":false,"linkquality":0}'
zigbee2mqtt:info  2020-01-16 21:10:21: MQTT publish: topic 'zigbee2mqtt/0x000d6f000ed8cb64', payload '{"temperature":0,"linkquality":70}'
```

If you have any ideas for me to test i'll be happy to give them a shot!

Database entries:
```
{"id":7,"type":"EndDevice","ieeeAddr":"0x000d6f000fdb0453","nwkAddr":90,"manufId":4401,"manufName":"Sercomm Corp.","powerSource":"Battery","modelId":"SZ-DWS04","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,1026,1280,32,2821],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"SZ-DWS04","manufacturerName":"Sercomm Corp.","powerSource":3,"zclVersion":1,"appVersion":35,"stackVersion":81,"hwVersion":18,"dateCode":"20161216","swBuildId":""}}},"binds":[{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b00194adbb1","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b00194adbb1","endpointID":1}]}},"appVersion":35,"stackVersion":81,"hwVersion":18,"dateCode":"20161216","swBuildId":"","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1}}
{"id":8,"type":"EndDevice","ieeeAddr":"0x000d6f000ed8cb64","nwkAddr":29329,"manufId":4401,"manufName":"Sercomm Corp.","powerSource":"Battery","modelId":"SZ-DWS04","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,1026,1280,32,2821],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"SZ-DWS04","manufacturerName":"Sercomm Corp.","powerSource":3}}},"binds":[{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b00194adbb1","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b00194adbb1","endpointID":1}]}},"interviewCompleted":false,"meta":{"configured":1}}
```

Running zigbee2mqtt 1.8.0 via Hassio

Koenkk, 2020-01-17:
Can you provide the herdsman debug logging while pairing this device?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

luckypants123, 2020-01-17:
Absolutely.

This is the log from switching on Allow Join through to failed interview:
https://pastebin.com/HPeseXgD

I updated to 1.9.0 before hand to see if that might resolve - but no such luck.

Anything look out of place?

Koenkk, 2020-01-17:
Can you try with the latest dev branch (contains some fixes related to pairing). If it doesn't work please provide the debug log again.

luckypants123, 2020-01-17:
I'm running through Hassio - so i switched to the zigbee2mqtt-edge add-on....i'm not 100% sure it is the latest.

Unfortunately it looks to be a similar result:
https://pastebin.com/Pb8R6QpD

I also added a USB extension cable in case it was some kind of interference problem.
Is there anything else you think I could try?

Koenkk, 2020-01-17:
Yes edge is the latest, but it takes some hours before the changes are there. Please try again tomorrow by uninstalling and installing the edge addon.

luckypants123, 2020-01-17:
Ok will do thank you

luckypants123, 2020-01-19:
Tried again on zigbee2mqtt-edge but same result unfortunately.

From Permit Join switched on through to Interview Failed - https://pastebin.com/0vzXzPuv

In case it helps, db entry from previously paired device (7) and new one (8):
```
{"id":7,"type":"EndDevice","ieeeAddr":"0x000d6f000fdb0453","nwkAddr":90,"manufId":4401,"manufName":"Sercomm Corp.","powerSource":"Battery","modelId":"SZ-DWS04","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,1026,1280,32,2821],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"SZ-DWS04","manufacturerName":"Sercomm Corp.","powerSource":3,"zclVersion":1,"appVersion":35,"stackVersion":81,"hwVersion":18,"dateCode":"20161216","swBuildId":""}}},"binds":[{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b00194adbb1","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b00194adbb1","endpointID":1}]}},"appVersion":35,"stackVersion":81,"hwVersion":18,"dateCode":"20161216","swBuildId":"","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1}}
{"id":8,"type":"EndDevice","ieeeAddr":"0x000d6f000ed8cb64","nwkAddr":33819,"manufId":4401,"manufName":"Sercomm Corp.","powerSource":"Battery","modelId":"SZ-DWS04","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,1026,1280,32,2821],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"SZ-DWS04","manufacturerName":"Sercomm Corp.","powerSource":3}}},"binds":[{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b00194adbb1","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b00194adbb1","endpointID":1}]}},"interviewCompleted":false,"meta":{"configured":1},"lastSeen":1579392016847}
```

Is there anything else we can try? If there is anything I can do to help let me know.

luckypants123, 2020-01-19:
Interesting twist.

I just tried to pair some different devices.... 2 different Visonic MCT-370 SMA Contact Sensors.

These both also fail the interview process.

I then flashed a new CC2531 and changed them to see if it was maybe a hardware issue, but still same.

Koenkk, 2020-01-19:
Is this still with the latest edge? Can you provide the logs for all devices when pairing? I'm very interested in this as I want to make the interview process for these devices more stable.

luckypants123, 2020-01-19:
Yes - I will verify everything tonight though.

Tonight i'll unpair all devices (interview complete as well) and change wireless channel to start fresh. 

Then i'll try pairing a number of different devices to get the log data from each and report back to you. I have 2 types of magnetic contact sensor, 1 type of occupancy sensor, and 1 type of door lock to test.

luckypants123, 2020-01-20:
Well i'm not sure why now but everything paired successfully. 
I'm expecting some new devices soon and will see if any pairing issues arise in the next couple of weeks.

If they do I will collect log data and open a new issue.

closed 2020-01-20 by luckypants123