Koenkk/zigbee2mqtt #2772: Ikea E1743 on off switch with xiao mi zncz02lm plug

Shanghaiman, 2020-01-18:
# Bug Report
Ikea switch controlling the xiaomi plug after immed
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version:
CC253X firmware version:

Koenkk, 2020-01-18:
Can you share your `database.db`?

Shanghaiman, 2020-01-19:
yes, check attachment.
tks

middelink, 2020-01-19:
I'm thinking both are in group 0...
(and there is no attachment?)

Shanghaiman, 2020-01-19:
``{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0001491a8c","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,8,11,12,110],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[],"outClusterList":[1280],"clusters":{},"binds":[]},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]}},"interviewCompleted":false,"meta":{}}
{"id":2,"type":"EndDevice","ieeeAddr":"0x00158d00039bb6ba","nwkAddr":63987,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.plug","epList":[1,2],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"65281":{"3":32,"5":5,"7":"0x0000000000000000","8":4898,"9":1027,"100":1,"149":0.07810354232788086,"152":0,"154":0},"65282":[{"elmType":16,"elmVal":1},{"elmType":33,"elmVal":3002},{"elmType":33,"elmVal":17320},{"elmType":36,"elmVal":[0,524297]},{"elmType":33,"elmVal":1771},{"elmType":32,"elmVal":105}],"modelId":"lumi.plug","appVersion":1,"dateCode":"09-04-2018"}},"genOnOff":{"attributes":{"61440":127148339,"onOff":1}}},"binds":[]},"2":{"epId":2,"inClusterList":[],"outClusterList":[],"clusters":{"genAnalogInput":{"attributes":{"presentValue":11.380000114440918}}},"binds":[]}},"appVersion":1,"dateCode":"09-04-2018","interviewCompleted":true,"meta":{"configured":1}}
{"id":3,"type":"EndDevice","ieeeAddr":"0x00158d000343dcce","nwkAddr":32260,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_magnet","epList":[1],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"65282":[{"elmType":16,"elmVal":0},{"elmType":33,"elmVal":2995},{"elmType":33,"elmVal":13224},{"elmType":36,"elmVal":[0,1]},{"elmType":33,"elmVal":162},{"elmType":32,"elmVal":109}],"modelId":"lumi.sensor_magnet"}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[]}},"interviewCompleted":true,"meta":{}}
{"id":4,"type":"EndDevice","ieeeAddr":"0x00158d00035f3a10","nwkAddr":23432,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_switch","epList":[1],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"32768":3,"onOff":1}},"genBasic":{"attributes":{"65282":[{"elmType":16,"elmVal":1},{"elmType":33,"elmVal":3002},{"elmType":33,"elmVal":17320},{"elmType":36,"elmVal":[0,1]},{"elmType":33,"elmVal":1771},{"elmType":32,"elmVal":108}],"modelId":"lumi.sensor_switch"}}},"binds":[]}},"interviewCompleted":true,"meta":{}}
{"id":5,"type":"EndDevice","ieeeAddr":"0x00158d0003d127a9","nwkAddr":27229,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_motion","epList":[1],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"65282":[{"elmType":16,"elmVal":0},{"elmType":33,"elmVal":3055},{"elmType":33,"elmVal":424},{"elmType":36,"elmVal":[0,0]},{"elmType":33,"elmVal":8},{"elmType":32,"elmVal":93}],"modelId":"lumi.sensor_motion","appVersion":11}},"msOccupancySensing":{"attributes":{"occupancy":1}}},"binds":[]}},"appVersion":11,"interviewCompleted":true,"meta":{}}
{"id":6,"ieeeAddr":"0xec1bbdfffeaada9b","nwkAddr":37921,"manufName":"IKEA of Sweden","powerSource":"Battery","modelId":"TRADFRI on/off switch","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":2080,"inClusterList":[0,1,3,9,32,4096,64636],"outClusterList":[3,4,6,8,25,258,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI on/off switch","manufacturerName":"IKEA of Sweden","powerSource":3,"zclVersion":3,"appVersion":33,"stackVersion":98,"hwVersion":1,"dateCode":"20190723","swBuildId":"2.2.010"}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0001491a8c","endpointID":1}]}},"appVersion":33,"stackVersion":98,"hwVersion":1,"dateCode":"20190723","swBuildId":"2.2.010","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1}}
{"id":7,"type":"EndDevice","ieeeAddr":"0x000b57fffe96a8dc","nwkAddr":11791,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Battery","modelId":"TRADFRI motion sensor","epList":[1],"endpoints":{"1":{"profId":49246,"epId":1,"devId":2128,"inClusterList":[0,1,3,9,2821,4096],"outClusterList":[3,4,6,25,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI motion sensor","manufacturerName":"IKEA of Sweden","powerSource":3,"zclVersion":1,"appVersion":17,"stackVersion":87,"hwVersion":1,"dateCode":"20170302","swBuildId":"1.2.214"}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0001491a8c","endpointID":1}]}},"appVersion":17,"stackVersion":87,"hwVersion":1,"dateCode":"20170302","swBuildId":"1.2.214","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1}}

Shanghaiman, 2020-01-19:
[database - .txt](https://github.com/Koenkk/zigbee2mqtt/files/4082927/database.-.txt)
this is the db file

Koenkk, 2020-01-20:
Can you try sending to the MQTT topic `zigbee2mqtt/bridge/configure` payload `0x00158d00039bb6ba`, does it still control the plug after that?

Vincent1468, 2020-01-22:
Hi, I'm having the same problem.
I tried sending the payload `0x14b457fffec5f828` (Ikea remote) to `zigbee2mqtt/bridge/configure` and it said successfully configured. I also tried sending the payload `0x00158d00044b3870` (Xiaomi switch) and it said Skipping configure.

My database:
`{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0001537458","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,8,11,12,110],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[],"outClusterList":[1280],"clusters":{},"binds":[]},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]}},"interviewCompleted":false,"meta":{},"lastSeen":null}
{"id":2,"type":"EndDevice","ieeeAddr":"0x00158d000424dd5f","nwkAddr":53048,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sens","epList":[1,2,3],"endpoints":{"1":{"profId":260,"epId":1,"devId":24321,"inClusterList":[0,3,25,65535,18],"outClusterList":[0,4,3,5,25,65535,18],"clusters":{"msTemperatureMeasurement":{"attributes":{"measuredValue":1933}},"msRelativeHumidity":{"attributes":{"measuredValue":5553}},"genBasic":{"attributes":{"65281":{"1":2975,"4":5032,"5":9,"6":[0,1],"10":0,"100":1968,"101":5461},"modelId":"lumi.sensor_ht","manufacturerName":"LUMI","powerSource":3,"zclVersion":1,"appVersion":2,"stackVersion":2,"hwVersion":30,"dateCode":"20160516","swBuildId":"3000-0001"}}},"binds":[]},"2":{"profId":260,"epId":2,"devId":24322,"inClusterList":[3,18],"outClusterList":[4,3,5,18],"clusters":{},"binds":[]},"3":{"profId":260,"epId":3,"devId":24323,"inClusterList":[3,12],"outClusterList":[4,3,5,12],"clusters":{},"binds":[]}},"appVersion":2,"stackVersion":2,"hwVersion":30,"dateCode":"20160516","swBuildId":"3000-0001","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":null}
{"id":3,"type":"EndDevice","ieeeAddr":"0x00158d0004261252","nwkAddr":60924,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.remote.b286acn01\u0000\u0000\u0000","epList":[1,2,3],"endpoints":{"1":{"profId":260,"epId":1,"devId":24321,"inClusterList":[0,3,25,65535,18],"outClusterList":[0,4,3,5,25,65535,18],"clusters":{"genBasic":{"attributes":{"65281":{"1":3045,"3":23,"4":5032,"5":42,"6":[0,1],"8":5129,"10":0},"65520":{"type":"Buffer","data":[170,16,5,65,135,106,1,16,1]},"modelId":"lumi.remote.b286acn01","manufacturerName":"LUMI","powerSource":3,"zclVersion":1,"appVersion":9,"stackVersion":2,"hwVersion":20,"dateCode":"20180809","swBuildId":"3000-0001"}},"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[]},"2":{"profId":260,"epId":2,"devId":24322,"inClusterList":[3,18],"outClusterList":[4,3,5,18],"clusters":{"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[]},"3":{"profId":260,"epId":3,"devId":24323,"inClusterList":[3,12],"outClusterList":[4,3,5,12],"clusters":{"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[]}},"appVersion":9,"stackVersion":2,"hwVersion":20,"dateCode":"20180809","swBuildId":"3000-0001","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1579718262211}
{"id":4,"type":"EndDevice","ieeeAddr":"0x00158d0003f1a35d","nwkAddr":48553,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_motion.aq2","epList":[1],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"65281":{"1":3065,"3":18,"4":5032,"5":6,"6":[0,2],"10":0,"11":0,"100":0},"modelId":"lumi.sensor_motion.aq2","appVersion":5}},"msIlluminanceMeasurement":{"attributes":{"measuredValue":0}},"msOccupancySensing":{"attributes":{"occupancy":1}}},"binds":[]}},"appVersion":5,"interviewCompleted":true,"meta":{},"lastSeen":null}
{"id":5,"type":"Router","ieeeAddr":"0x000d6ffffe451b9c","nwkAddr":12079,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb E14 CWS opal 600lm","epList":[1],"endpoints":{"1":{"profId":49246,"epId":1,"devId":512,"inClusterList":[0,3,4,5,6,8,768,2821,4096],"outClusterList":[5,25,32,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI bulb E14 CWS opal 600lm","manufacturerName":"IKEA of Sweden","powerSource":1,"zclVersion":1,"appVersion":17,"stackVersion":87,"hwVersion":1,"dateCode":"20180410","swBuildId":"1.3.009"}}},"binds":[]}},"appVersion":17,"stackVersion":87,"hwVersion":1,"dateCode":"20180410","swBuildId":"1.3.009","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1579718396119}
{"id":6,"type":"Router","ieeeAddr":"0x00158d00044b3870","nwkAddr":46188,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.ctrl_ln1.aq1","epList":[1,2,3,5],"endpoints":{"1":{"profId":260,"epId" [... cut ...] asic":{"attributes":{"65281":{"1":3055,"3":23,"4":5032,"5":24,"6":[0,1],"8":5129,"10":0},"65520":{"type":"Buffer","data":[170,16,5,65,135,59,1,16,1]}}}},"binds":[]},"2":{"epId":2,"inClusterList":[],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":0}}},"binds":[]},"3":{"epId":3,"inClusterList":[],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":0}}},"binds":[]}},"appVersion":9,"interviewCompleted":true,"meta":{},"lastSeen":null}`

Koenkk, 2020-01-23:
Issue should be fixed in the latest dev branch, to fix it for existing users (which already paired it):

- Update to latest dev
- Force remove the device trough https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove
- Pair the E1743 again.

Vincent1468, 2020-01-23:
Thank you for taking the time to look into this. 
I've updated to the latest dev (1.9.0-dev, commit 93e96a2), removed the device and repaired. 
I also tried repairing the Xiaomi switch.
Unfortunately the problem persists. 
I'm using the IKEA E1810 not the E1743.

Koenkk, 2020-01-25:
@Vincent1468 unfortunately I don't have the E1810 but can you add the following line here: `await endpoint.bind('genOnOff', 901);` https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L1161

After that force remove the device and repair it (https://github.com/Koenkk/zigbee2mqtt/issues/2772#issuecomment-577850910)

How to get to `devices.js` on your system: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

Vincent1468, 2020-01-26:
@Koenkk Yes that did the trick! Thank you very much!

Shanghaiman, 2020-01-27:
> Issue should be fixed in the latest dev branch, to fix it for existing users (which already paired it):
> 
> * Update to latest dev
> * Force remove the device trough https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove
> * Pair the E1743 again.
This did solve my problem.
it was difficult removing the device, had to use the "zigbee2mqtt/bridge/config/force_remove" mqtt message and to delete all the device entities, after that to factory reset the on off switch by pressing it for few times on the link button.
also had to restart my hassio few times and only then pair the device again and it stopped controlling my xiao mi plugs (which i have 2)

Thank, i hope this will help others

Shanghaiman, 2020-01-27:
> Can you try sending to the MQTT topic `zigbee2mqtt/bridge/configure` payload `0x00158d00039bb6ba`, does it still control the plug after that?

This option didnt make any change for me

Koenkk, 2020-01-27:
Implemented in latest dev branch now, added instructions how to unbind it from the default group here: https://github.com/Koenkk/zigbee2mqtt.io/blob/develop/docs/devices/E1743.md#binding

Users which paired this device before zigbee2mqtt 1.9.0-dev need to follow the instructions from https://github.com/Koenkk/zigbee2mqtt/issues/2772#issuecomment-577850910 first.

closed 2020-01-27 by Koenkk

octavious-01, 2020-02-01:
Updated to latest dev of Zigbee2MQTT, Force removed the device and repaired the device several times. The best result I get is the following:

E1743- Click Off - If ZNCZ02LM is in off state then ZNCZ02LM changes state to on. This only happens with one of two the smart power plugs. Otherwise there is no effect. I am at a loss 

https://community.home-assistant.io/t/someone-got-the-ikea-e1743-working-with-zigbee2mqtt/130669/10

Koenkk, 2020-02-02:
Can you share your database.db and zigbee2mqtt startup log?

octavious-01, 2020-02-02:
Hi

thanks for your quick reply. Please see log file attached.

`{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0014d9e566","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]}},"meta":{},"lastSeen":null}
{"id":2,"type":"Router","ieeeAddr":"0xd0cf5efffe7b83d4","nwkAddr":17696,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb E27 W opal 1000lm","epList":[1],"endpoints":{"1":{"profId":49246,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[5,25,32,4096],"clusters":{"genBasic":{"dir":{"value":1},"attributes":{"10":{},"65533":1,"zclVersion":1,"appVersion":17,"stackVersion":87,"hwVersion":1,"manufacturerName":"IKEA of Sweden","modelId":"TRADFRI bulb E27 W opal 1000lm","dateCode":"20170302","powerSource":1}},"genIdentify":{"dir":{"value":1},"attributes":{"65533":1,"identifyTime":0}},"genGroups":{"dir":{"value":1},"attributes":{"65533":1,"nameSupport":0}},"genScenes":{"dir":{"value":3},"attributes":{"65533":1,"count":0,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0}},"genOnOff":{"dir":{"value":1},"attributes":{"16387":1,"65533":1,"onOff":1,"globalSceneCtrl":1,"onTime":0,"offWaitTime":0}},"genLevelCtrl":{"dir":{"value":1},"attributes":{"15":0,"16384":254,"65533":1,"currentLevel":254,"remainingTime":0,"onOffTransitionTime":5,"onLevel":255}},"genOta":{"dir":{"value":2},"attributes":{}},"genPollCtrl":{"dir":{"value":2},"attributes":{}},"haDiagnostic":{"dir":{"value":1},"attributes":{"65533":1,"numberOfResets":0,"macRxBcast":0,"macTxBcast":0,"aPSRxBcast":0,"aPSTxBcast":0,"nwkFcFailure":0,"packetBufferAllocateFailures":0,"averageMacRetryPerApsMessageSent":0,"lastMessageLqi":116,"lastMessageRssi":-71}},"lightLink":{"dir":{"value":3},"attributes":{"65533":1}}},"binds":[]}},"appVersion":17,"stackVersion":87,"hwVersion":1,"dateCode":"20170302","swBuildId":"1.2.214","zclVersion":1,"meta":{},"lastSeen":1580627552190}
{"id":3,"type":"Router","ieeeAddr":"0x14b457fffefda308","nwkAddr":56870,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI control outlet","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,8,4096,64636],"outClusterList":[5,25,32,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI control outlet","manufacturerName":"IKEA of Sweden","powerSource":1,"zclVersion":3,"appVersion":32,"stackVersion":98,"hwVersion":1,"dateCode":"20190308","swBuildId":"2.0.022"}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0014d9e566","endpointID":1}]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[]}},"appVersion":32,"stackVersion":98,"hwVersion":1,"dateCode":"20190308","swBuildId":"2.0.022","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1580630385066}
{"id":4,"ieeeAddr":"0x086bd7fffe5f345a","nwkAddr":61468,"epList":[1],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]}},"interviewCompleted":false,"meta":{},"lastSeen":1580578661648}
{"id":5,"type":"EndDevice","ieeeAddr":"0x086bd7fffe5f345a","nwkAddr":50541,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Battery","modelId":"TRADFRI on/off switch","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":2080,"inClusterList":[0,1,3,9,32,4096,64636],"outClusterList":[3,4,6,8,25,258,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI on/off switch","manufacturerName":"IKEA of Sweden","powerSource":3,"zclVersion":3,"appVersion":34,"stackVersion":98,"hwVersion":1,"dateCode":"20190410","swBuildId":"2.2.008"}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":47}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0014d9e566","endpointID":1}]}},"appVersion":34,"stackVersion":98,"hwVersion":1,"dateCode":"20190410","swBuildId":"2.2.008","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1580627494348}
{"id":6,"type":"EndDevice","ieeeAddr":"0x00158d0001d87a1b","nwkAddr":17179,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_switch","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,3,25,65535],"outClusterList":[0,3,4,5,6,8,25],"clusters":{"genBasic":{"dir":{"value":3},"attributes":{"65282":[{"elmType":16,"elmVal":1},{"elmType":33,"elmVal":3012},{"elmType":33,"elmVal":17320},{"elmType":36,"elmVal":[0,1]},{"elmType":33,"elmVal":145},{"elmType":32,"elmVal":91}]}},"genIdentify":{"dir":{"value":3},"attributes":{}},"genGroups":{"dir":{"value":2},"attributes":{}},"genScenes":{"dir":{"value":2},"attributes":{}},"genOnOff":{"dir":{"value":2},"attributes":{"onOff":1}},"genLevelCtrl":{"dir":{"value":2},"attributes":{}},"genOta":{"dir":{"value":3},"attribu [... cut ...] "attributes":{}},"genOnOff":{"dir":{"value":2},"attributes":{}},"genLevelCtrl":{"dir":{"value":2},"attributes":{}},"genOta":{"dir":{"value":3},"attributes":{}},"manuSpecificCluster":{"dir":{"value":1},"attributes":{}},"msOccupancySensing":{"attributes":{"occupancy":1}}},"binds":[]}},"appVersion":11,"stackVersion":2,"hwVersion":2,"dateCode":"20160524","swBuildId":"3000-0001","zclVersion":1,"meta":{},"lastSeen":1580630766574}`

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/4143697/log.txt)

Koenkk, 2020-02-02:
You seem to have multiple (3) E1743 switches, are you sure you remove them all according to https://github.com/Koenkk/zigbee2mqtt/issues/2772#issuecomment-577850910 ?

octavious-01, 2020-02-02:
> You seem to have multiple (3) E1743 switches, are you sure you remove them all according to [#2772 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/2772#issuecomment-577850910) ?

I definitely only have one E1743 switch. Where do you see three?

Koenkk, 2020-02-02:
- Can you set the log level to debug (https://www.zigbee2mqtt.io/information/debug.html#zigbee2mqtt-debug-logging)
- Provide the log when pressing the IKEA switch once

octavious-01, 2020-02-03:
Embarrassingly I have found the problem.

Whilst initially investigating the problem with the E1743 switch, I created a test automation that toggled the ZNCZ02LM if the E1743 was clicked (sort of overriding the original problem). Unfortunately when I updated to the latest dev release, force removed the E17443 and re-paired, I forgot to remove the automation. Hence the behaviour I described continued.

All solved now. Sorry for the misinformation.