Koenkk/zigbee2mqtt #2773: Eurotronic Spirit issues z2m 1.9.0

TASSDevon, 2020-01-18:
I updated to zigbee2mqtt v1.9.0 yesterday.
Today I wanted to install a second eurotronic spirit zigbee valve. It seems somewhat successful but I do not get any zigbee messages. Tried removing and re-pairing the device, that works (see log below) but that's the last reaction I see for this device. Any ideas? FYI I'm using the big TI dev board as zigbee coordinator and a CC2530 as router.

`Jan 18 12:59:22 beaglebone npm[9026]: zigbee2mqtt:info  2020-01-18 12:59:22: Removing '0x00158d0001ffd433'
Jan 18 12:59:22 beaglebone npm[9026]: zigbee2mqtt:info  2020-01-18 12:59:22: Clearing Home Assistant discovery topic for '0x00158d0001ffd433'
Jan 18 12:59:22 beaglebone npm[9026]: zigbee2mqtt:info  2020-01-18 12:59:22: MQTT publish: topic 'homeassistant/climate/0x00158d0001ffd433/climate/config', payload 'null'
Jan 18 12:59:22 beaglebone npm[9026]: zigbee2mqtt:info  2020-01-18 12:59:22: MQTT publish: topic 'homeassistant/sensor/0x00158d0001ffd433/battery/config', payload 'null'
Jan 18 12:59:22 beaglebone npm[9026]: zigbee2mqtt:info  2020-01-18 12:59:22: MQTT publish: topic 'homeassistant/sensor/0x00158d0001ffd433/linkquality/config', payload 'null'
Jan 18 12:59:22 beaglebone npm[9026]: zigbee2mqtt:info  2020-01-18 12:59:22: Successfully removed 0x00158d0001ffd433
Jan 18 12:59:22 beaglebone npm[9026]: zigbee2mqtt:info  2020-01-18 12:59:22: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"0x00158d0001ffd433"}'

Jan 18 12:59:46 beaglebone npm[9026]: zigbee2mqtt:info  2020-01-18 12:59:46: Starting interview of '0x00158d0001ffd433'
Jan 18 12:59:46 beaglebone npm[9026]: zigbee2mqtt:info  2020-01-18 12:59:46: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00158d0001ffd433"}}'
Jan 18 12:59:53 beaglebone npm[9026]: zigbee2mqtt:info  2020-01-18 12:59:53: Successfully interviewed '0x00158d0001ffd433', device has successfully been paired
Jan 18 12:59:53 beaglebone npm[9026]: zigbee2mqtt:info  2020-01-18 12:59:53: Device '0x00158d0001ffd433' is supported, identified as: Eurotronic Spirit Zigbee wireless heater thermostat (SPZB0001)
Jan 18 12:59:53 beaglebone npm[9026]: zigbee2mqtt:info  2020-01-18 12:59:53: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x00158d0001ffd433","model":"SPZB0001","vendor":"Eurotronic","description":"Spirit Zigbee wireless heater thermostat","supported":true}}'
Jan 18 12:59:53 beaglebone npm[9026]: zigbee2mqtt:info  2020-01-18 12:59:53: Configuring '0x00158d0001ffd433'
Jan 18 12:59:55 beaglebone npm[9026]: zigbee2mqtt:info  2020-01-18 12:59:55: Successfully configured '0x00158d0001ffd433'
`

TASSDevon, 2020-01-18:
Just a small update:
I changed both valves' friendly name in the configuration yaml file, restarted zigbee2mqtt and now the second valve is being picked up in HA.

However when trying to change the setpoint I get following timeout again:

`Jan 18 14:00:51 beaglebone npm[9168]: zigbee2mqtt:error 2020-01-18 14:00:51: Publish 'set' 'current_heating_setpoint' to 'Spirit Valve Living' failed: 'Error: Timeout - 52944 - 1 - 3 - 513 - 4 after 10000ms'
Jan 18 14:00:51 beaglebone npm[9168]: zigbee2mqtt:info  2020-01-18 14:00:51: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'current_heating_setpoint' to 'Spirit Valve Living' failed: 'Error: Timeout - 52944 - 1 - 3 - 513 - 4 after 10000ms'","meta":{"friendly_name":"Spirit Valve Living"}}'`

TASSDevon, 2020-01-18:
Got it to work in the end.
Since I had a spare CC2350 after replacing it as a coordinator, I flashed the router fw on it. There is an outlet less than 50cm from the valve, so I plugged in the router there.

Re-paired the valve and look at that, it started working:

[image]

Fun fact: the sig quality of the router is 50ish, the Xiaomi occupancy that is literally 20cm away also shows about 50 sig quality. The valve also shows about 50 sig quality now.

Bottomline: I have no idea why it refused to work properly when it paired with an Ikea tradfri, while another Spirit valve happily talks to Tradfri for weeks already.

mariantr, 2020-02-06:
I have a similar issue:
I have 4 Eurotronic Sprint TRVs. 1 is working flawlessly, I never had problems with it.
The other 3, on the other hand, randomly refuse to work, especially after Z2M restarts.
Yesterday I had a power failure and one TRV didn't respond after the system was up. I could see it in the network map, the signal was good, but I could not get or set anything. Tried removing the batteries, resetting, nothing worked, although it joined successfully even after reset.

This morning I updated to 1.10 and surprise: the TRV that yesterday was unreachable, works like a charm. But another one has now the same problem:
```
feb 06 12:05:58 smartHub npm[9703]: zigbee2mqtt:debug 2020-02-06 12:05:58: Received MQTT message on 'zigbee2mqtt/0x00158d00019238df/set/system_mode' with data                           'heat'
feb 06 12:05:58 smartHub npm[9703]: zigbee2mqtt:debug 2020-02-06 12:05:58: Publishing 'set' 'system_mode' to '0x00158d00019238df'
feb 06 12:06:08 smartHub npm[9703]: zigbee2mqtt:error 2020-02-06 12:06:08: Publish 'set' 'system_mode' to '0x00158d00019238df' failed: 'Error: Read 0x00158d00019238df/1 hvacThermostat([16392], {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":4151,"disableDefaultResponse":true}) failed (Error: Timeout - 35537 - 1 - 9 - 513 - 1 after 10000ms)'
feb 06 12:06:08 smartHub npm[9703]: zigbee2mqtt:debug 2020-02-06 12:06:08: Error: Read 0x00158d00019238df/1 hvacThermostat([16392], {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":4151,"disableDefaultResponse":true}) failed (Error: Timeout - 35537 - 1 - 9 - 513 - 1 after 10000ms)
feb 06 12:06:08 smartHub npm[9703]:     at Endpoint.<anonymous> (/srv/homeautomation/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:173:23)
feb 06 12:06:08 smartHub npm[9703]:     at Generator.throw (<anonymous>)
feb 06 12:06:08 smartHub npm[9703]:     at rejected (/srv/homeautomation/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)
feb 06 12:06:08 smartHub npm[9703]: zigbee2mqtt:info  2020-02-06 12:06:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'system_mode' to '0x00158d00019238df' failed: 'Error: Read 0x00158d00019238df/1 hvacThermostat([16392], {\"timeout\":10000,\"defaultResponseTimeout\":15000,\"manufacturerCode\":4151,\"disableDefaultResponse\":true}) failed (Error: Timeout - 35537 - 1 - 9 - 513 - 1 after 10000ms)'","meta":{"friendly_name":"0x00158d00019238df"}}'

```

closed 2020-04-14 by stale[bot]