Koenkk/zigbee2mqtt #2774: After a power cut how i can't select a favourute color - philips hue -

kitopopo, 2020-01-18:
After a Power cut the bulbs always power on with the default philips colour (yeyow colour). 
The philips bridge have a option for select the start favourite colour after power cut.

I test add this in configuration.yaml file  but doesn't work, the zigbe2mqtt sevice doen's star due to some errors ((I dont have home assistant, only domoticz) )

'0x0017880104dc1f87':
    friendly_name: '0x0017880104dc1f87'
    retain: false
    light:
      transition:4
      "rgb": "46,102,193"

I also tryed send a mqtt messages, no errot messages i receive but the bulb doen't keep the color temp sended:

`sudo mosquitto_pub -h 127.0.0.1 -t zigbee2mqtt/0x0017880104dc1xxx/set -m "{\"hue_power_on_color_temperature\": \100,\"}"
`
`sudo mosquitto_pub -h 127.0.0.1 -t zigbee2mqtt/0x0017880104dc1xxx/set -m "{\"hue_power_on_color_temperature\": \100\"}"
`

`sudo mosquitto_pub -h 127.0.0.1 -t zigbee2mqtt/0x0017880104dc1xxx/set -m "{\"hue_power_on_color_temperature\": \100,\"}"
`
Is there a way to light the bulb in a favorite color through the configuration.yaml file or any other method? Maybe you can send a command to the bulb through a mqqtt message and customize the ignition color

I've tried select off state (hue power on behavior) and after run a domoticz script using "json" so that the light bulb comes on in my favorite color and it works but unfortunately first the lamp is lit a few milliseconds in the philips default colour and after change to my favorite colour.

Thanks for your work.

Best regards

## What did you expect to happen

to be able to light all my philips hue lamps in my favorite color after a power cut just like the option included in the philips bridge

## How to reproduce it (minimal and precise)

Making a power cut. Wait 15 secons and and give power to the bulb again. Always the lamp start in default philips colour.

## Debug Info
zigbee2mqtt version:version 1.9.0.
CC253X firmware version:CC2531_DEFAULT_20190608.zip.
bulb 929001953101

Koenkk, 2020-01-18:
This has not been implemented yet.

kitopopo, 2020-01-18:
Thanks.

Best regards

kitopopo, 2020-01-19:
Hi,

Why does the lamp after a voltage cut retain the brightness level but not the color before the cut?

Thanks in advanced, 

Best regards

Koenkk, 2020-01-22:
@kitopopo could you sniff the traffic of the hue bridge when pairing the bulb and setting the restore options from the hub? If you share the wireshark file with me I can implement this. https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

kitopopo, 2020-01-23:
Ok, I will try to do what you say to see if it is not very difficult for me. i will  try to follow your instructions. thanks

kitopopo, 2020-01-23:
hello @Koenkk 
I am already sniffing the traffic, I have configured wireshark with the zigbee protocol and I have only put 2 keys. The first key is the philips hue bridge key, and the second key is the network encryption key (Transport Key). I understand that the Trust Center link key I have to replace it with that of the Philips Hue. It is right? I begin the process of adding a Philips Hue lamp with hue bridge and changing the energy loss configuration options? How do I send you the sniffed wireshark file?

Thanks in advance

Koenkk, 2020-01-23:
You can send the file on telegram (@koenkk)

kitopopo, 2020-01-23:
Dear @Koenkk ,

I send you the file wireshark in 2 minuts. I explain the process:

After make reset to my 4 bulbs i switch on wireshark:

- In philips hue app i press discover new devices
- Philips app have detected 4 bulbs
- I only add 1 bulb to my room
- After add the bulb to the room i enter in the menu app power on behavior
-  by default philips default was selected
- After i selected "last state after loss power"
- After i selected " Custom colour " and i have selected cold colour
- I power of the bulb, wait 15 seconds and after power on
- The bulb start in my "custom colour" --> colour color

(I have make the power on behavior process two times.)

Can you check the wireshark file is is correct? if do you need sniff the traffic again a canr reapeat the process.

Thanks in advanced

kitopopo, 2020-01-24:
Hello dear friend, I send you de philips hue decrypted 5 minutes ago telegram way.
I have managed to find the transport key. It has really been found by the wireshark search engine. The decoded information is already there and I can see the On / off states. I sniffed the data frame. Around the second 50 I set the option of ignition behavior (recovery after loss of energy), around the second 70 I configure the option of ignition behavior (custom color), then I choose the desired color and around the 90 minute I make a power cut . The bulb starts in the desired color. I hope it serves. Best regards and now that I know how to do it if you need anything else, do not hesitate to ask me. Thanks in advance

Koenkk, 2020-01-25:
Supported now, updated docs: https://www.zigbee2mqtt.io/devices/9290022169.html#power-on-behavior   thanks!

closed 2020-01-25 by Koenkk