Koenkk/zigbee2mqtt #2776: "Color" mqtt topic doens't works with philips hue with mosquitto

kitopopo, 2020-01-18:
Hi dear friend,

What would be the correct message to send to change the color to my Philips Hue lamps? I have tried the following messages but none works:

`sudo mosquitto_pub -h 127.0.0.1 -t zigbee2mqtt/0x0017880104dc1f87/set -m "{\"color\": {"r":255,"g":133,b:24}\"}"`

`sudo mosquitto_pub -h 127.0.0.1 -t zigbee2mqtt/0x0017880104dc1f87/set -m "{\"color\": \"{"x":0.6876,"y":0.3003}\"}"`

`sudo mosquitto_pub -h 127.0.0.1 -t zigbee2mqtt/0x0017880104dc1f87/set -m "{\"color\": \"{"hue":118,\"saturation:93}\"}"
`
The response message of the log is always the same and the bulb powers off.

> ene 19 00:30:42 raspberrypi npm[454]: zigbee2mqtt:info  2020-01-19 00:30:42: MQTT publish: topic 'zigbee2mqtt/0x0017880104dc1f87', payload '{"state":"OFF","brightness":0,"color":{},"color_temp":null,"linkquality":31}'

if the bulb ever responds it always stays in a blue tone with different RGB values.

I tried with all possible commands, with quotes, without quotes, using the counter-bar but the bulb does not change color. 

The "brightness" and "color_temp" commands seem to work correctly, but the "color" command does not work

Thank you for your help, receive a cordial greeting

## What did you expect to happen

Change the bulb colour sending a mqtt message

## How to reproduce it (minimal and precise)

sending my messages mentioned above

## Debug Info
zigbee2mqtt version: version 1.9.0.
CC2531 firmware version: 20190608.zip

sjorge, 2020-01-19:
try using single quotes for the -m flag and use I escape double quotes for the json payload.

Or if you want to avoid json, you can call zigbee2mqtt/bulb/set/color with payload of #FF0000 too (IIRC)

kitopopo, 2020-01-26:
Hi dear @sjorge ,

the next message works: 

`sudo mosquitto_pub -h 127.0.0.1 -t zigbee2mqtt/0x0017880104dc1f87/set/color -m "#FF5112"`

why the next message doen't work?

`
sudo mosquitto_pub -h 127.0.0.1 -t zigbee2mqtt/0x0017880104dc1f87/set -m "{\"color\": {"r":255,"g":133,b:24}\"}"`

`sudo mosquitto_pub -h 127.0.0.1 -t zigbee2mqtt/0x0017880104dc1f87/set -m "{\"color\": \"{"x":0.6876,"y":0.3003}\"}"`

Would you be so kind as to give several examples using this color method?

Thanks in advanced. Best regards

sjorge, 2020-01-26:
Try:
```
mosquitto_pub -h 127.0.0.1 -t zigbee2mqtt/0x0017880104dc1f87/set -m '{"color": {"x": 0.6876, "y": 0.3003}}'
```

or

```
mosquitto_pub -h 127.0.0.1 -t zigbee2mqtt/0x0017880104dc1f87/set -m '{"color": {"r": 255, "g": 133, "b": 24}}'
```

kitopopo, 2020-01-28:
Hi dear friend,

Thanks , both commands works correctly. I close the issue.

Best regards

closed 2020-01-28 by kitopopo