Koenkk/zigbee2mqtt #2779: CC1352P-2 won't pair IKEA GU10 bulbs

HA-TB303, 2020-01-19:
I just (re)installed the CC1352p-2 again, changed the pan-ID. Seems to work, but I can't pair IKEA Bulbs and pairing in general is problematic.

My config:

````
{
  "data_path": "/share/zigbee2mqtt",
  "devices": "devices.yaml",
  "groups": "groups.yaml",
  "homeassistant": true,
  "permit_join": false,
  "mqtt": {
    "base_topic": "zigbee2mqtt",
    "server": "mqtt://192.168.2.101:1883",
    "user": "zigbee2mqtt",
    "password": "*secret*"
  },
  "serial": {
    "port": "/dev/ttyACM0"
  },
  "advanced": {
    "pan_id": 6151,
    "channel": 11,
    "network_key": [*secret_key*],
    "availability_blacklist": []
  },
  "ban": [],
  "whitelist": [],
  "queue": {},
  "socat": {
    "enabled": false,
    "master": "pty,raw,echo=0,link=/dev/ttyZ2M,mode=777",
    "slave": "tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5",
    "restartdelay": 1,
    "initialdelay": 1,
    "options": "-d -d",
    "log": false
  }
}
```

## Debug Info
Version of Zigbee2Mqtt: 1.9.0
Coordinator version: 20191106  (launchxl CC1352P-2, REV E)

log:
```
{"ieeeAddr":"0x90fd9ffffe793f91","friendlyName":"0x90fd9ffffe793f91","type":"Router","networkAddress":44301,"manufacturerName":"IKEA of Sweden","modelID":"TRADFRI bulb GU10 W 400lm","failed":["lqi"],"lastSeen":null},{"ieeeAddr":"0x90fd9ffffe14a247","friendlyName":"0x90fd9ffffe14a247","type":"Router","networkAddress":21233,"manufacturerName":"IKEA of Sweden","modelID":"TRADFRI bulb GU10 W 400lm","failed":["lqi"],"lastSeen":null},{"ieeeAddr":"0x000b57fffed8a6dc","friendlyName":"0x000b57fffed8a6dc","type":"Router","networkAddress":39999,"manufacturerName":"IKEA of Sweden","modelID":"TRADFRI bulb GU10 W 400lm","failed":["lqi"],"lastSeen":1579372542760},
```

HA-TB303, 2020-01-19:
I've just erased the devices and database.db and started from scratch.

Pairing a Hue remote works fine, when pairing the IKEA bulbs (in this case 2 bulbs on 1 light switch):

```
}}","state_topic":"zigbee2mqtt/0x90fd9ffffe14a247","json_attributes_topic":"zigbee2mqtt/0x90fd9ffffe14a247","name":"0x90fd9ffffe14a247_linkquality","unique_id":"0x90fd9ffffe14a247_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x90fd9ffffe14a247"],"name":"0x90fd9ffffe14a247","sw_version":"Zigbee2mqtt 1.9.0","model":"TRADFRI LED bulb GU10 400 lumen, dimmable (LED1650R5)","manufacturer":"IKEA"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2020-01-19 12:23:51: Successfully interviewed '0x90fd9ffffe14a247', device has successfully been paired
zigbee2mqtt:info  2020-01-19 12:23:51: Device '0x90fd9ffffe14a247' is supported, identified as: IKEA TRADFRI LED bulb GU10 400 lumen, dimmable (LED1650R5)
zigbee2mqtt:info  2020-01-19 12:23:51: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x90fd9ffffe14a247","model":"LED1650R5","vendor":"IKEA","description":"TRADFRI LED bulb GU10 400 lumen, dimmable","supported":true}}'
zigbee2mqtt:info  2020-01-19 12:24:00: MQTT publish: topic 'zigbee2mqtt/bridge/config/devices', payload '[{"ieeeAddr":"0x00124b001ca5ea81","type":"Coordinator","networkAddress":0,"friendly_name":"Coordinator","softwareBuildID":"zStack3x0","dateCode":"20191106","lastSeen":1579433040033},{"ieeeAddr":"0x0017880103c9d20f","type":"EndDevice","networkAddress":10868,"model":"324131092621","friendly_name":"0x0017880103c9d20f","manufacturerID":4107,"manufacturerName":"Philips","powerSource":"Battery","modelID":"RWL021","hardwareVersion":1,"softwareBuildID":"6.1.1.28573","dateCode":"20190410","lastSeen":1579432896681},{"ieeeAddr":"0x90fd9ffffe793f91","networkAddress":30844,"friendly_name":"0x90fd9ffffe793f91","lastSeen":1579433021271},{"ieeeAddr":"0x90fd9ffffe14a247","type":"Router","networkAddress":37058,"model":"LED1650R5","friendly_name":"0x90fd9ffffe14a247","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelID":"TRADFRI bulb GU10 W 400lm","hardwareVersion":1,"softwareBuildID":"1.2.214","dateCode":"20170302","lastSeen":1579433031946}]'
zigbee2mqtt:info  2020-01-19 12:24:00: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"devices","message":[{"ieeeAddr":"0x00124b001ca5ea81","type":"Coordinator","networkAddress":0,"friendly_name":"Coordinator","softwareBuildID":"zStack3x0","dateCode":"20191106","lastSeen":1579433040036},{"ieeeAddr":"0x0017880103c9d20f","type":"EndDevice","networkAddress":10868,"model":"324131092621","friendly_name":"0x0017880103c9d20f","manufacturerID":4107,"manufacturerName":"Philips","powerSource":"Battery","modelID":"RWL021","hardwareVersion":1,"softwareBuildID":"6.1.1.28573","dateCode":"20190410","lastSeen":1579432896681},{"ieeeAddr":"0x90fd9ffffe793f91","networkAddress":30844,"friendly_name":"0x90fd9ffffe793f91","lastSeen":1579433021271},{"ieeeAddr":"0x90fd9ffffe14a247","type":"Router","networkAddress":37058,"model":"LED1650R5","friendly_name":"0x90fd9ffffe14a247","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelID":"TRADFRI bulb GU10 W 400lm","hardwareVersion":1,"softwareBuildID":"1.2.214","dateCode":"20170302","lastSeen":1579433031946}]}'
zigbee2mqtt:error 2020-01-19 12:25:01: Failed to interview '0x90fd9ffffe793f91', device has not successfully been paired
zigbee2mqtt:info  2020-01-19 12:25:01: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x90fd9ffffe793f91"}}'
```

1 pairs succesfully, the other fails. This is more luck, since mostly both fail.

HA-TB303, 2020-01-19:
When trying to pair it seperately:

```
zigbee2mqtt:info  2020-01-19 12:36:08: Starting interview of '0x90fd9ffffe793f91'
zigbee2mqtt:info  2020-01-19 12:36:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x90fd9ffffe793f91"}}'
zigbee2mqtt:warn  2020-01-19 12:37:23: Device with modelID 'undefined' is not supported.
zigbee2mqtt:warn  2020-01-19 12:37:23: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
zigbee2mqtt:info  2020-01-19 12:37:23: MQTT publish: topic 'zigbee2mqtt/0x90fd9ffffe14a247', payload '{"state":"OFF","linkquality":7,"brightness":204}'
```

HA-TB303, 2020-01-19:
I'm starting to think the proces only fails when there is 1 IKEA router active.
I try to pair 2 lights, 1 succeeds, 1 fails.

Every next attempt to pair a bulb fails. Maybe it's not possible to pair a IKEA bulb on a IKEA router because of some sort of bug....

Koenkk, 2020-01-20:
Can you try resetting the bulb through touchlink? (https://www.zigbee2mqtt.io/information/touchlink.html) Should star pairing immediately after that.

timdonovanuk, 2020-02-17:
FWIW I experienced a similar thing. Hard to pin it down to one thing. I ended up reflashing my CC1352P-2 and things got easier to pair, except for the GU10's. Moving them to closer sockets maybe helped? There is one bulb that still won't pair, even when only a few meters LOS from the coordinator.

HA-TB303, 2020-03-11:
Tried again with the lastest EDGE and DEV firmware, but still no go. I LOVE zigbee2mqtt but this (re)paring process is extremely frustrating and takes me at least half A day each time I try due to the large amount of devices and device locations.

```
zigbee2mqtt:info  2020-03-11 22:59:40: Logging to console and directory: '/share/zigbee2mqtt/log/2020-03-11.22-59-40' filename: log.txt
zigbee2mqtt:info  2020-03-11 22:59:40: Starting zigbee2mqtt version 1.11.0-dev (commit #ce48bee4cd0eac6cb36ed36b20a39024e1ecf0de)
zigbee2mqtt:info  2020-03-11 22:59:40: Starting zigbee-herdsman...
zigbee2mqtt:info  2020-03-11 22:59:41: zigbee-herdsman started
zigbee2mqtt:info  2020-03-11 22:59:41: Coordinator firmware version: '{"type":"zStack3x0","meta":{"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20200306}}'
zigbee2mqtt:info  2020-03-11 22:59:41: Currently 0 devices are joined:
zigbee2mqtt:info  2020-03-11 22:59:41: Zigbee: disabling joining new devices.
zigbee2mqtt:info  2020-03-11 22:59:41: Connecting to MQTT server at mqtt://192.168.2.101:1883
zigbee2mqtt:info  2020-03-11 22:59:41: Connected to MQTT server
zigbee2mqtt:info  2020-03-11 22:59:41: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
zigbee2mqtt:info  2020-03-11 22:59:41: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.11.0-dev","commit":"ce48bee4cd0eac6cb36ed36b20a39024e1ecf0de","coordinator":{"type":"zStack3x0","meta":{"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20200306}},"log_level":"info","permit_join":false}'
zigbee2mqtt:info  2020-03-11 22:59:46: MQTT publish: topic 'zigbee2mqtt/bridge/config/devices', payload '[{"ieeeAddr":"0x00124b001ca5ea81","type":"Coordinator","networkAddress":0,"friendly_name":"Coordinator","softwareBuildID":"zStack3x0","dateCode":"20200306","lastSeen":1583963986904}]'
zigbee2mqtt:info  2020-03-11 22:59:46: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"devices","message":[{"ieeeAddr":"0x00124b001ca5ea81","type":"Coordinator","networkAddress":0,"friendly_name":"Coordinator","softwareBuildID":"zStack3x0","dateCode":"20200306","lastSeen":1583963986913}]}'
zigbee2mqtt:info  2020-03-11 23:00:00: MQTT publish: topic 'zigbee2mqtt/bridge/config/devices', payload '[{"ieeeAddr":"0x00124b001ca5ea81","type":"Coordinator","networkAddress":0,"friendly_name":"Coordinator","softwareBuildID":"zStack3x0","dateCode":"20200306","lastSeen":1583964000011}]'
zigbee2mqtt:info  2020-03-11 23:00:00: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"devices","message":[{"ieeeAddr":"0x00124b001ca5ea81","type":"Coordinator","networkAddress":0,"friendly_name":"Coordinator","softwareBuildID":"zStack3x0","dateCode":"20200306","lastSeen":1583964000018}]}'
zigbee2mqtt:info  2020-03-11 23:00:01: Zigbee: allowing new devices to join.
zigbee2mqtt:info  2020-03-11 23:00:01: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.11.0-dev","commit":"ce48bee4cd0eac6cb36ed36b20a39024e1ecf0de","coordinator":{"type":"zStack3x0","meta":{"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20200306}},"log_level":"info","permit_join":true}'
zigbee2mqtt:info  2020-03-11 23:00:11: Starting network scan (includeRoutes 'false')
zigbee2mqtt:info  2020-03-11 23:00:11: Network scan finished
zigbee2mqtt:info  2020-03-11 23:00:11: MQTT publish: topic 'zigbee2mqtt/bridge/networkmap/raw', payload '{"nodes":[{"ieeeAddr":"0x00124b001ca5ea81","friendlyName":"Coordinator","type":"Coordinator","networkAddress":0,"failed":[],"lastSeen":null}],"links":[]}'
zigbee2mqtt:info  2020-03-11 23:00:30: Device '0x90fd9ffffe793f91' joined
zigbee2mqtt:info  2020-03-11 23:00:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x90fd9ffffe793f91"}}'
zigbee2mqtt:info  2020-03-11 23:00:30: Starting interview of '0x90fd9ffffe793f91'
zigbee2mqtt:info  2020-03-11 23:00:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x90fd9ffffe793f91"}}'
zigbee2mqtt:info  2020-03-11 23:00:30: Device '0x90fd9ffffe14a247' joined
zigbee2mqtt:info  2020-03-11 23:00:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x90fd9ffffe14a247"}}'
zigbee2mqtt:info  2020-03-11 23:00:30: Starting interview of '0x90fd9ffffe14a247'
zigbee2mqtt:info  2020-03-11 23:00:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x90fd9ffffe14a247"}}'
zigbee2mqtt:error 2020-03-11 23:01:30: Failed to interview '0x90fd9ffffe793f91', device has not successfully been paired
zigbee2mqtt:info  2020-03-11 23:01:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x90fd9ffffe793f91"}}'
zigbee2mqtt:error 2020-03-11 23:01:30: Failed to interview '0x90fd9ffffe14a247', device has not successfully been paired
zigbee2mqtt:info  2020-03-11 23:01:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x90fd9ffffe14a247"}}'
zigbee2mqtt:info  2020-03-11 23:01:38: Switching log level to 'debug'
zigbee2mqtt:info  2020-03-11 23:01:38: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.11.0-dev","commit":"ce48bee4cd0eac6cb36ed36b20a39024e1ecf0de","coordinator":{"type":"zStack3x0","meta":{"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20200306}},"log_level":"debug","permit_join
```

timdonovanuk, 2020-03-13:
Still an issue for me too, particularly with the CC1352P-2 instabilities I have to repair often.

HA-TB303, 2020-03-13:
> Can you try resetting the bulb through touchlink? (https://www.zigbee2mqtt.io/information/touchlink.html) Should star pairing immediately after that.

I have tried this, the lights blink, but no pairing.

HA-TB303, 2020-03-13:
I wonder if it's not time to change the website information until these issues are resolved to prevent any one else from buying this rather expensive module....

timdonovanuk, 2020-03-13:
Indeed, bit sad I spent £50 trying to get better zigbee performance :(

Koenkk, 2020-03-14:
- please don`t start a discussion of the same issue in multiple issues (#2376), this makes it hard for me to regain context
- I assume you already tried: https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair
    - When migrating from a previous network, especially make sure that the old network is completely powered off (including the bulbs that are not paired to the new network). These could still form a network where your devices would join to.
- try touchlink (but I guess you already tried) https://www.zigbee2mqtt.io/information/touchlink.html
- make sure you are on the latest zigbee2mqtt dev branch (normally not needed but it contains an important fix regarding use of wrong network addresses)
- try pairing them to your e1524 remote and after that to your cc1352p-2
- if none if this helps, the only way to investigate deeper is to sniff the network when pairing https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

@timdovanuk I read that you also have some instability with the CC1352P-2 and thus you need to require to repair. After fixing this issue I would like to check what's actually happening.

timdonovanuk, 2020-03-14:
@Koenkk Could you please help me getting the zigbee-herdsman debug logs? Currently they go to stdout, not the /logs/ dir. I even did a "docker container logs 5e4a2bcea77b > /tmp/docker.log" but this doesn't capture stdout for some reason, only the same stuff as in /logs.

I've now force_removed all my lights as they clutter the debug logs (as 90% of them have stopped working anyway), as soon as I can actually get at the debug logs (other than watching them flash by in Portainer) I can share.

Koenkk, 2020-03-15:
You can get them with `docker logs zigbee2mqtt > my_log 2>&1`

HA-TB303, 2020-03-16:
> * please don`t start a discussion of the same issue in multiple issues (#2376), this makes it hard for me to regain context
> * I assume you already tried: https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair
>   
>   * When migrating from a previous network, especially make sure that the old network is completely powered off (including the bulbs that are not paired to the new network). These could still form a network where your devices would join to. Keep in mind that when changing pan_id, channel or network key you also have to do this!
> * try touchlink (but I guess you already tried) https://www.zigbee2mqtt.io/information/touchlink.html
> * make sure you are on the latest zigbee2mqtt dev branch (normally not needed but it contains an important fix regarding use of wrong network addresses)
> * try pairing them to your e1524 remote and after that to your cc1352p-2
> * if none if this helps, the only way to investigate deeper is to sniff the network when pairing https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html
> 
> @timdonovanuk I read that you also have some instability with the CC1352P-2 and thus you need to require to repair. After fixing this issue I would like to check what's actually happening.

I have tried everything you listed. 
- I switched all router devices off.
- I used touchlink to reset the ikea bulbs and the Philips bulbs.
- I used the on/off (x5) method for the ikea bulbs
- I tried moving closer to the coordinator where possible (e.g. the ikea signal repeater)
None of that works!
The issue is that NONE of my router devices will connect to the CC1352P-2 coordinator. All battery powered sensors connect just fine.

Sniffing is a bridge too far for me.  :(

Koenkk, 2020-03-16:
- If I understand correctly, you didn't manage to pair any of your router devices to your CC1352P-2?
- In that case I would recommend starting fresh:
  - Delete `data/database.db` and `data/coordinator_backup.json`
  - Choose a different `channel` in `configuration.yaml`
  - Update zigbee2mqtt to 1.11.0
  - Reflash your CC1352P-2 with https://github.com/Koenkk/Z-Stack-firmware/blob/develop/coordinator/Z-Stack_3.x.0/bin/CC1352P_2_20200312.zip while carefully following https://www.zigbee2mqtt.io/information/flashing_via_uniflash.html (step 5 is important here)
  - Try to pair a router device (e.g. GU10 bulb) first.

HA-TB303, 2020-03-16:
Ok, thanks, I'll try tonight!

HA-TB303, 2020-03-16:
Thanks, this worked, or at least a lot better then before.

Most of the Ikea and Philips bulbs required multiple resets before they got paired.

The only things I can not get paired are the Ikea signal repeater https://www.zigbee2mqtt.io/devices/E1746.html and the Xiaomi temperature/humidity sensor https://www.zigbee2mqtt.io/devices/WSDCGQ11LM.html

HA-TB303, 2020-03-17:
I'm stil having some issues. Yesterday evening the entire network was gone without any obvious reason.
I had to reboot the linux machine and after that it sort of worked again.

I also still have 2 devices in my network that have long been removed (months ago) and are NOT online.

They keep reappearing. It's both the devices which appear to be not connected.

[image]

Ans I have the issue that the xiaomi temperature and humidity sensors will not pair.

Koenkk, 2020-03-17:
- Did you try replacing the battery of the weather sensor? (do you see no logging at all?)
- Can you remove these entries manually from `database.db`?

HA-TB303, 2020-03-17:
I don't think it's in the database because I removed that completely before starting over new. I first removed the hassio addon, then removed the fysical folder in /share and then reinstalled.

I'll try replacing the battery, but both devices did in fact pair to the CC2530 just a few days before.

HA-TB303, 2020-03-17:
zigbee2mqtt:info  2020-03-17 19:50:05: Starting interview of '0x0017880102139e87'
zigbee2mqtt:info  2020-03-17 19:50:05: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x0017880102139e87"}}'
zigbee2mqtt:error 2020-03-17 19:51:05: Failed to interview '0x0017880102139e87', device has not successfully been paired

HA-TB303, 2020-03-17:
After replacing the battery it still didn’t work, so I brough them closer to the coordinator and kept trying. After several attempts, they both work now.

closed 2020-05-23 by stale[bot]