Koenkk/zigbee2mqtt #2780: Trust ZLED-RGB9 not pairing

denvro, 2020-01-19:
# Bug Report
## What happened
Trying to pair my Trust ZLED-RGB9 (switch-on/-off 5 times) it's recognized. The lamp starts burning green. Then the zigbee2mqtt log shows
`zigbee2mqtt:info  2020-01-19 14:39:04: Starting interview of 'tv_lamp'
zigbee2mqtt:info  2020-01-19 14:39:04: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"tv_lamp"}}'
zigbee2mqtt:error 2020-01-19 14:39:15: Failed to interview 'tv_lamp', device has not successfully been paired` 

It's added to the devices.yaml file. But not to my devices in Home Assistant. I SSH-ed into my R-PI to try to add the device the 'how_to_support_new_devices' way but cannot locate the file `node_modules/zigbee-herdsman-converters/devices.js` 

When restarting the zigbee2mqtt service the log states 1 device is joined. But it's not supported
`zigbee2mqtt:info  2020-01-19 14:14:23: Currently 1 devices are joined:
zigbee2mqtt:info  2020-01-19 14:14:23: tv_lamp (0x00158d00011e8ec4): Not supported (Router)`

I tried with MQTT.fx to send a message to the topic and that works but the log stated

## What did you expect to happen
I would have expected that my Trust led bulb was supported. Or that it would pair like described in the 'how_to_support_new_devices' manual and that I could add the device manually.
`zigbee2mqtt:warn  2020-01-19 14:32:16: Device with modelID 'undefined' is not supported.
zigbee2mqtt:warn  2020-01-19 14:32:16: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html`

## Debug Info
Home Assistant version: 0.100.3
Mosquitto add-on version: 5.1
zigbee2mqtt add-on version: 1.9.0
CC253X firmware version: 20190608

Configured HA and zigbee2mqtt to automatically discover and join new devices.

denvro, 2020-01-19:
herdsman debug info states there is a timeout?

`
2020-01-19T15:21:55.640Z zigbee-herdsman:controller:device Interview - failed for device '0x00158d00011e8ec4' with error 'Error: Timeout - 61141 - 1 - 6 - 0 - 1 after 10000ms
    at Timeout._onTimeout (/zigbee2mqtt-1.9.0/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at listOnTimeout (internal/timers.js:531:17)
    at processTimers (internal/timers.js:475:7)'
2020-01-19T15:21:55.644Z zigbee-herdsman:controller:error Interview failed for '0x00158d00011e8ec4 with error 'Error: Timeout - 61141 - 1 - 6 - 0 - 1 after 10000ms'
`

Koenkk, 2020-01-20:
Can you try pairing the device with the latest dev branch (hassio **edge** addon). 

If it doesn't work please provide the log when running with herdsman debug logging.

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

denvro, 2020-01-20:
I've installed the dev branch (edge addon). Same result. herdsman debugging is on. 

Logging on pastebin https://pastebin.com/22EytJKY

Koenkk, 2020-01-22:
I think I've fixed the problem in the latest dev branch, please try updating to the latest edge (wait a few hours before it's published) and try pairing it again.

denvro, 2020-01-22:
Wow that's great news. Will try tomorrow!

denvro, 2020-01-24:
I can confirm it is fixed in the latest dev branch. It pairs and recognizes the bulb now. Thank you so much for your help!

Koenkk, 2020-01-24:
💯  thanks for confirming!

closed 2020-01-24 by Koenkk