Koenkk/zigbee2mqtt #2781: Saswell TRV SEA801-ZIGBEE support

m4tek, 2020-01-19:
I'm looking to get some support with adding SEA801-ZIGBEE TRV.

I managed to connect it to the network and it is reporting state changes correctly. It sends updates in with cluster '61184' and type 'raw'. Those are HEX words that I got sorted out for the most part.

I can see hvacThermostat which covers all the options that are available for this value, but my device uses raw data in this custom cluster

Issues I'm struggling right now:

Should I start with defining new cluster in ZCL?
Should I do any bindings in the initial devices.js config, as there is no documentation that I was able to find that would describe this part
I don't have SASWELL hub, so I cannot fully capture 2-directional communication, but I'm working in the assumption that the same binary/hex format is used in both direction

Capture data:
https://docs.google.com/spreadsheets/d/1BaKJdkmumf3S8rP4JfhNATyeryAGX9wh416llWWFACQ/edit#gid=0

rotragit, 2020-01-19:
I asked SASWELL for documentation, but I've got no response.

m4tek, 2020-01-20:
We can get far without it, but @Koenkk or someone from the core DEV team needs to share some guidance on how to approach it. I don't want to start moving in wrong direction.

sjorge, 2020-01-20:
Starting with an existing TRV is probably a good jump of point. Just stay clear of the schneider or eurotronic ones as they have a lot of custom converters

Koenkk, 2020-01-20:
Raw messages is not a good sign, it means that the TRV uses some custom cluster (which you need to figure out). Does wireshark show anything interesting when sniffing? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

Koenkk, 2020-01-21:
@m4tek good, then the next step would be to define it here: https://github.com/Koenkk/zigbee-herdsman/blob/master/src/zcl/definition/cluster.ts

m4tek, 2020-01-23:
I need to do some more research, as cluster 61184 is now defined after:
https://github.com/Koenkk/zigbee-herdsman/pull/125
and syntax of the message is not exactly the same :/

Math1010, 2020-02-03:
Hi,
I just received a SASWELL SEA801 (no gateway) and I cannot pair it with the cc2530 module.
I press the button "+" and "Auto / manu" and the text "----" appears but no message on  ZigBee2MQTT, the allow_join parameter is true.
Can you explain how you did to read the raw TRV data?
Thanks.

gkilcaus, 2020-02-04:
> Hi,
> I just received a SASWELL SEA801 (no gateway) and I cannot pair it with the cc2530 module.
> I press the button "+" and "Auto / manu" and the text "----" appears but no message on ZigBee2MQTT, the allow_join parameter is true.
> Can you explain how you did to read the raw TRV data?
> Thanks.

I have the same result with cc2531

@m4tek, did you progress with integration?

rotragit, 2020-02-04:
You have first to check:
1) that you have the zigbee version. It should be written on the valve
"SEA801-zigbee". If you find "SEA801-APP" or other, it's not zigbee.
2) that valve has horrible wireless range. Make tests just 1 or 2 meters
away from the dongle. Less is better.

romano

gkilcaus, 2020-02-04:
> You have first to check: 1) that you have the zigbee version. It should be written on the valve "SEA801-zigbee". If you find "SEA801-APP" or other, it's not zigbee. 2) that valve has horrible wireless range. Make tests just 1 or 2 meters away from the dongle. Less is better. romano
> […](#)

Makes sense. I didn't pay attention to this when ordering. In product page it said sea801-app and on device itself it's sea801-df1. So I guess I can trash it..

rotragit, 2020-02-04:
sea801-app use 868Mhz. I think sea801-df1 is just another name for the 868Mhz version. You have to buy a gateway to use them. Tuya protocol is based on MQTT (gateway to cloud server), so it could be possible to integrate at that level. I don't know.
Anyway, as far as I can see, SASWELL has only implemented the zigbee protocol as an envelope around their own protocol on the sea801-zigbee version. If you look for any TRV I think it's better to look for something other and already tested.
I have 5 sea801-zigbee but I'm unable to use them anyway :-(

Math1010, 2020-02-10:
Actually I have a df1 model :(
Anyone have tried to disassembly the device ?
Thank.

rotragit, 2020-02-10:
No at the moment, but I plan to do that if I cannot return my to
AliExpress. Anyway DF1 is not zigbee and the RF chip is different. One is
868Mhz, zigbee is 2,4Ghz...

rotragit, 2020-04-15:
Ping
OK, I have 5 TRV devices of this kind, but very little time to investigate, and at the moment I'm missing a CC2531 dongle with zigbee2mqtt firmware.
At the moment I have sniffed the traffic when the TRV is paired within deconz (but it's not functional), and I'm comparing with the captured data by @m4tek . I confirm the layout and semantic of data. I'm not sure if the type messages are the same to send commands from zigbee2mqtt to the TRV.

Damme, 2020-05-19:
Saswell messaged me the other day with a new model, SEA802-Zigbee, and they claim it talks zigbee to base station, I asked for price, resonable 15USD each + the not so resonable double for shipping.

I asked for protocol but only got specs. I'll try to convince them to release protocol since they claim it to be open protocol.

rotragit, 2020-06-03:
The protocol is "open" in the sense that zigbee is open che they use the "Tuya protocol" that it's also "open". As far as I have understood, the Tuya protocol is based on a mqtt cloud server. Tuya claims to be a easy protocol to be implemented. The problem that it's like a envelope, and I haven't found anywere the description of the mqtt messages the Tuya gateway send to set the parameters on each SEA801-Zigbee (hoping the SEA801-Zigbee uses the same message types).
Because I have five of them, it's possible I'll get a Tuya gateway just to sniff and learn the messages exchanged between the TRV and the gateway.

ThreeOlives, 2020-07-27:
I have same TRV devices. They successfully connected to the network but i cant manage them.
I tied to add device to devices.js like this
`    {
        zigbeeModel: ['GbxAXL2'], 
        model: 'SEA801',
        vendor: 'SASWELL',
        description: 'SASWELL _TYST11_KGbxAXL2', 
        supports: 'temperature',
        fromZigbee: [fz.tuya_thermostat_on_set_data],
        toZigbee: [],
    },`

and there is some logs from z2m
https://pastebin.com/FdHXHLUa

Maybe it helps )

rotragit, 2020-07-29:
I should receive a Tuya Zigbee Gateway in a few days, If it works with this device I can sniff the traffic.

rotragit, 2020-08-01:
Here are some sniffing (pairing, temperature set, auto/manu, on/off).
I have not a zigbee2mqtt installation at the moment. I have to search where are my spare cc2531 dongles ;-)

[SEA801.zip](https://github.com/Koenkk/zigbee2mqtt/files/5011033/SEA801.zip)

rotragit, 2020-08-01:
@Koenkk @m4tek it's possible to reopen this issue as we have a sniffed traffic to investigate?

wizardnl, 2020-08-15:
just wondering if there was any update also have the same device

rotragit, 2020-08-16:
it has been implemented in deconz, so we have now all the codes to port it.  I can try to do that but I have to find my dongles and setup a test environment :-)

m4tek, 2020-09-07:
Looking at the situation, I'm in favour of duplicating functions. 
Trying make it work all the time will be very hard. With practical lack of unit testing it would be easy to break and hard to troubleshoot, while non obvious change in one place could have effect on the other. This is where Tuya brought us :/

miecio45, 2020-09-25:
Hello!
Can I somehow help you in implementing support for Tuya thermostat valve?

Sthopeless, 2020-09-26:
> I'm going to implement this device. I'm created the entry in devices.js and looking in fromZigbee.js where there is
> 
> `const tuyaThermostat = (model, msg, publish, options, meta)`
> 
> I see that model is never used and the switch case on dp is quite difficult to distinguish which dp is about which device. Many dp are common, but other are not and because the tuya protocol implements manufacturer functions among standard and cloud functions, we should expect that the same dp could have different meaning on different devices (also if all of them are TRV).I don't know what is worse or better: duplicate the function for every device, leave as it is (with no worry if a dp is about a device or another) or translate in a difficult to read if/else (because switch/case doesn't work on two variables model and dp).
> 
> Just for example, SEA801-ZIGBEE uses dp=364 instead of dp=372 for auto/manual mode. Lowbattery status id dp=1385.

I have you successfully implemented it?

I have received mine today. from database.db:
```
{"id":44,"type":"EndDevice","ieeeAddr":"0xec1bbdfffe5b835b","nwkAddr":27171,"manufId":0,"manufName":"_TYST11_KGbxAXL2","powerSource":"Battery","modelId":"GbxAXL2\u0000","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[0,3],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"GbxAXL2\u0000","manufacturerName":"_TYST11_KGbxAXL2","powerSource":3,"zclVersion":3,"appVersion":73,"stackVersion":0,"hwVersion":1,"dateCode":"20180727"}}},"binds":[],"meta":{}}},"appVersion":73,"stackVersion":0,"hwVersion":1,"dateCode":"20180727","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1601149622346}
```

I have added to my device.js:
```
    {
        zigbeeModel: ['GbxAXL2'],
        fingerprint: [{modelID: 'GbxAXL2\u0000', manufacturerName: '_TYST11_KGbxAXL2'}],
        model: 'GbxAXL2',
        vendor: 'TuYa',
        description: 'Saswell TRV',
        supports: 'temperature, heating system control',
        fromZigbee: [
            fz.tuya_thermostat,
            fz.tuya_water_leak,
            fz.ignore_basic_report,
        ],
        toZigbee: [
            tz.tuya_thermostat_min_temp,
            tz.tuya_thermostat_max_temp,
        ],
   
    },
```

And this is my log after pairing:
```
Zigbee2MQTT:debug 2020-09-26 22:51:02: Device '0xec1bbdfffe5b835b' announced itself
Zigbee2MQTT:info  2020-09-26 22:51:02: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xec1bbdfffe5b835b"},"type":"device_announced"}'
Zigbee2MQTT:debug 2020-09-26 22:51:12: Received Zigbee message from '0xec1bbdfffe5b835b', type 'commandSetDataResponse', cluster 'manuSpecificTuyaDimmer', data '{"data":{"data":[1],"type":"Buffer"},"dp":357,"fn":0,"status":0,"transid":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-09-26 22:51:12: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe5b835b', payload '{"brightness":-3,"holidays":[{"hour":0,"minute":0,"temperature":0},{"hour":0,"minute":0,"temperature":0},{"hour":0,"minute":0,"temperature":0},{"hour":0,"minute":0,"temperature":0},{"hour":0,"minute":0,"temperature":0},{"hour":0,"minute":0,"temperature":0}],"level":0,"linkquality":66,"max_temperature":205,"min_temperature":249,"water_leak":true,"workdays":[{"hour":0,"minute":0,"temperature":0},{"hour":0,"minute":0,"temperature":0},{"hour":0},{},{},{}]}'
Zigbee2MQTT:debug 2020-09-26 22:51:12: Received Zigbee message from '0xec1bbdfffe5b835b', type 'commandSetDataResponse', cluster 'manuSpecificTuyaDimmer', data '{"data":{"data":[0,0,0,233],"type":"Buffer"},"dp":614,"fn":0,"status":0,"transid":2}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-09-26 22:51:12: Received Zigbee message from '0xec1bbdfffe5b835b', type 'commandSetDataResponse', cluster 'manuSpecificTuyaDimmer', data '{"data":{"data":[0,0,0,205],"type":"Buffer"},"dp":615,"fn":0,"status":0,"transid":3}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-09-26 22:51:12: Received Zigbee message from '0xec1bbdfffe5b835b', type 'commandSetDataResponse', cluster 'manuSpecificTuyaDimmer', data '{"data":{"data":[0],"type":"Buffer"},"dp":1385,"fn":0,"status":0,"transid":4}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-09-26 22:51:12: Received Zigbee message from '0xec1bbdfffe5b835b', type 'commandSetDataResponse', cluster 'manuSpecificTuyaDimmer', data '{"data":{"data":[0],"type":"Buffer"},"dp":362,"fn":0,"status":0,"transid":5}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-09-26 22:51:12: Received Zigbee message from '0xec1bbdfffe5b835b', type 'commandSetDataResponse', cluster 'manuSpecificTuyaDimmer', data '{"data":{"data":[0],"type":"Buffer"},"dp":364,"fn":0,"status":0,"transid":7}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-09-26 22:51:12: Received Zigbee message from '0xec1bbdfffe5b835b', type 'commandSetDataResponse', cluster 'manuSpecificTuyaDimmer', data '{"data":{"data":[4,0,75,0,50,0,135,0,100,0,210,0,150,0,255,0,200],"type":"Buffer"},"dp":123,"fn":0,"status":0,"transid":255}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-09-26 22:51:12: Received Zigbee message from '0xec1bbdfffe5b835b', type 'commandSetDataResponse', cluster 'manuSpecificTuyaDimmer', data '{"data":{"data":[4,1,104,0,210,2,208,0,200,4,56,0,205,5,160,0,200],"type":"Buffer"},"dp":124,"fn":0,"status":1,"transid":0}' from endpoint 1 with groupID 0
[... cut ...]
"max_temperature":205 is the target temp I have set on the TRV
"min_temperature":249 I'm assuming is the current temp

I have also connected the TRV to wireshark and my findings can be seen here on [Google Sheet Link](https://docs.google.com/spreadsheets/d/e/2PACX-1vSN8czWM_IVTDJ6Ayr8SEU9u1bzMdTVxImeeZ0k6uXhzcOzW1Hc46KipJiObhhC8RD23kV7jr0cMzbh/pubhtml?gid=615915482&single=true)
[image]

I'm still struggling to understand how to add it correctly to Z2M so any help would be extremely appreciated.

skumka, 2020-10-03:
I am struggling with SEA802... I am new to all z2m concept and getting crazy. I would love to help but I need to know what I should do. 
Actually I have found similar topic with more details on similar problem. 
https://community.home-assistant.io/t/i-am-totally-noob-with-zigbee-and-tuya-and-i-need-advice-for-smart-thermostatic-radiator-valves/163607/21

It ended up with pull request.

rotragit, 2020-10-03:
I'm sorry, I have very little time in this period of the year.
This is tuya explained: [https://medium.com/@dzegarra/zigbee2mqtt-how-to-add-support-for-a-new-tuya-based-device-part-2-5492707e882d](https://medium.com/@dzegarra/zigbee2mqtt-how-to-add-support-for-a-new-tuya-based-device-part-2-5492707e882d)
This tuya documentation, where we find each tuya device (could) have custom functions: [https://docs.tuya.com/en/iot/configure-in-platform/function-definition/define-product-features?id=K97vug7wgxpoq](https://docs.tuya.com/en/iot/configure-in-platform/function-definition/define-product-features?id=K97vug7wgxpoq)
Here is the deconz issue about this TRV: [https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3109#issuecomment-678367184](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3109#issuecomment-678367184)
Here is the deconz code implementing this TRV  (look for GbxAXL2): [https://github.com/Smanar/deconz-rest-plugin/blob/master/tuya.cpp](https://github.com/Smanar/deconz-rest-plugin/blob/master/tuya.cpp)

insipiens, 2020-10-06:
I now have 2 of these. i see a lot of the hard work has already been done! 

I like that they have metal cups to screw yo the valve body, some others like moes hy368 are all plastic and can break.

Will be good to have these in Z2M

Sthopeless, 2020-10-07:
glad you are on board @insipiens I tried looking at the deconz implementation but it's all gibberish to me 😞.
And my 'achievements' are documented on my post above. I hope it helps

skumka, 2020-10-07:
@Sthopeless +1 @insipiens I have SEA802. Please let me know if I can help you with testing this one too. Maybe implementation can cover both.

insipiens, 2020-10-09:
@skumka  @Sthopeless Thanks for the welcome.....

I have a few days off next week so I should be able to work on this.

If it follows the usual Tuya format, which the Deconz work shows then even with different codes it should be straight forward........... Famous last words, right!

insipiens, 2020-10-09:
I made a start:

`MQTT publish: topic 'zigbee2mqtt/Saswell TRV', payload '{"current_heating_setpoint":"15.5","linkquality":52,"local_temperature":"16.4","preset_mode":"manual","system_mode":"heat"}'`

Mainly just reading from the device, but have added adjusting the set temperature

[image]

[image]

rotragit, 2020-10-09:
Great!

skumka, 2020-10-09:
Great! Would you mind sharing what is your definition from Devices.js?

insipiens, 2020-10-09:
These are the additions I have made so far, make a back up of each file and use at own risk.

_line numbers are indicative only_

//Device.js 

```
    {
        fingerprint: [{modelID: 'GbxAXL2\u0000', manufacturerName: '_TYST11_KGbxAXL2'}],
        model: 'GbxAXL2',
        vendor: 'Saswell',
        description: 'Saswell Thermostatic Radiator Valve',
        supports: 'thermostat, temperature',
        fromZigbee: [fz.saswell_thermostat],
        toZigbee: [tz.saswell_thermostat_current_heating_setpoint],
        meta: {configureKey: 1},    
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['genBasic']);
        },  
    },

```

//homeassistant.js

```
    'GbxAXL2': [
        climate(5, 30, 'current_heating_setpoint', 0.5, ['off', 'heat'], [], ['manual', 'program']),
    ],

```

//fromZigbee.js

//inserted after Tuya thermostat at line 465
```
const saswellThermostat = (model, msg, publish, options, meta) => {
    const dp = msg.data.dp;
    const data = msg.data.data;
    const dataAsDecNumber = utils.convertMultiByteNumberPayloadToSingleDecimalNumber(data)
    let temperature;
    
    switch (dp) {
    case 357: // Thermostat on off  = off, on = heat
        return {system_mode: dataAsDecNumber ? 'heat' : 'off'};
    case 362: // away mode
    return {preset_mode: dataAsDecNumber ? 'away' : 'auto'};     
    case 364: // Changed program mode 
        return {preset_mode: dataAsDecNumber ? 'auto' : 'manual'};   
    case 614: // MCU reporting room temperature
        temperature = (dataAsDecNumber / 10).toFixed(1);
        return {local_temperature: temperature};
    case 615: // set temperature
        temperature = (dataAsDecNumber / 10).toFixed(1);
        return {current_heating_setpoint: temperature};
    default:
        meta.logger.warn(`zigbee-herdsman-converters:SaswellThermostat: NOT RECOGNIZED DP #${
            dp} with data ${JSON.stringify(data)}`);
    }
};

```
//and down at around line 5595 

```
    saswell_thermostat: {
        cluster: 'manuSpecificTuyaDimmer',
        type: ['commandGetData', 'commandSetDataResponse'],
        convert: saswellThermostat,
    },

```
//toZigbee.js line 3060

```

    saswell_thermostat_current_heating_setpoint: {
        key: ['current_heating_setpoint'],
        convertSet: async (entity, key, value, meta) => {
            const temp = Math.round(value * 10);
            const payloadValue = utils.convertDecimalValueTo2ByteHexArray(temp);
            await sendTuyaCommand(entity, 615, 0, [4, 0, 0, ...payloadValue]);
        },
    },

```

Sthopeless, 2020-10-09:
@insipiens thanks and great job! 
Btw what are your thoughts this vs the Moes TRV's? design aside

insipiens, 2020-10-09:
These are DP codes I haven't figured out yet which are coming from the device.

112
113
114
116
117
118
120
121
122

115
--
110
119

I've analysed the program DPs that are sent: 

123
124
125
126
127
128
129

each one is for a day's program 1-7 divided into 4 sections as indicated by the first values 4, and the time is calculated from the 2nd, 3rd, then 6th & 7th etc. using a formula:  int(a*256+b)/60 for hour and mod 60 for minutes (I think) 

example: 

[4,1,14,0,155,1,224,0,150,4,56,0,150,5,40,0,160]}
--
1*256+14 = 04:30am

insipiens, 2020-10-09:
Ah, how embarrassing, noticed an error :/

homeassistant.js should be manual or auto, **_not_** manual or program

```
    'GbxAXL2': [
        climate(5, 30, 'current_heating_setpoint', 0.5, ['off', 'heat'], [], ['manual', 'auto']),
    ],  //saswell
```

Some additional 'toZigbee.js' clauses - to control system mode (off/on) and preset manual/auto from home assistant or other HA - just add them on to the end of the earlier clauses.
```
    saswell_thermostat_mode: {
        key: ['preset'],
        convertSet: async (entity, key, value, meta) => {
            if( value == 'auto'){ sendTuyaCommand(entity, 364, 0, [1, 1]) }
            else { sendTuyaCommand(entity, 364, 0, [1, 0]) }
        },
    },
    saswell_thermostat_standby: {
        key: ['system_mode'],
        convertSet: async (entity, key, value, meta) => {
            sendTuyaCommand(entity, 357, 0, [1, value === 'heat' ? 1 : 0]);
        },
    },
```
and the replacement "device.js" entry to include those changes:
```
    {
        fingerprint: [{modelID: 'GbxAXL2\u0000', manufacturerName: '_TYST11_KGbxAXL2'}],
        model: 'GbxAXL2',
        vendor: 'Saswell',
        description: 'Saswell Thermostatic Radiator Valve',
        supports: 'thermostat, temperature',
        fromZigbee: [fz.saswell_thermostat],
        toZigbee: [tz.saswell_thermostat_current_heating_setpoint, tz.saswell_thermostat_mode, tz.saswell_thermostat_standby],
        meta: {configureKey: 1},    
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['genBasic']);
        },  
    },
```

insipiens, 2020-10-09:
We will need to clarify the missing fromZigbee.js DP codes before we can PR this. 

How far do we want to go with the toZigbee.js commands?

rotragit, 2020-10-09:
in deconz we didn't solve all the codes. I had the tuya gateway for a while
and I was unable to understand the meaning.

insipiens, 2020-10-09:
I'll take a look sometime this weekend. Would be great for some feedback on functionality etc.

insipiens, 2020-10-09:
@Sthopeless 

> 
> @insipiens thanks and great job!
> Btw what are your thoughts this vs the Moes TRV's? design aside

There are some pros and cons, the Moes HY368 has more functionality perhaps but construction seems weaker. I had one Moes break - Wife came downstairs to find it, its plastic fixing nut and a lug scattered across the hallway - to be fair Moes refunded me.

also the Z2M conversion used is shared with a few other valve models so the look in home assistant isn't as pleasing - doesn't follow the HA requirements...could this be changed? probably but you would need people testing all the related valve models...

Moes:
- lighter and seems more fragile
- reports valve position
- can calibrate the reported temperature
- doesn't report battery level
- not well integrated into lovelace

Of course if I come downstairs tomorrow and find the Saswell has been pebble dashing my hallway I might have a different opinion...

How the two look side by side in HA:

[image]

skumka, 2020-10-10:
@insipiens This is great! thanks a lot. I confirm it works also for model SEA802.
I think that in addtion to what is possible right now the solid baseline for those TRVs will be also:
- battery level
- valve position
- open door detection

Secondary (at least for me) are:
- child lock
- weekly schedule

insipiens, 2020-10-10:
```

10/10/2020 10:54 | #115 | {"type":"Buffer","data":[0,0,0,0,0,0,0,0,0,17,16,17,17,16,15,15,15,14,16,15,14,14,13,0]}
-- | -- | --
10/10/2020 10:54 | #110 | {"type":"Buffer","data":[0,0,0,0,0,0,0,0,0,16,17,17,17,16,15,14,14,14,14,14,14,14,17,0]}
10/10/2020 10:54 | #119 | {"type":"Buffer","data":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,4,35,100,10,44,85,98,100,0]}
10/10/2020 11:54 | #115 | {"type":"Buffer","data":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0]}
10/10/2020 11:54 | #110 | {"type":"Buffer","data":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0]}
10/10/2020 11:54 | #119 | {"type":"Buffer","data":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0]}
10/10/2020 11:54 | #116 | {"type":"Buffer","data":[0,0,0,0,9,0,0]}
10/10/2020 11:54 | #112 | {"type":"Buffer","data":[0,0,0,0,9,0,0]}
10/10/2020 11:54 | #120 | {"type":"Buffer","data":[0,0,0,0,24,0,0]}
10/10/2020 11:54 | #117 | {"type":"Buffer","data":[0,9,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0]}
10/10/2020 11:54 | #113 | {"type":"Buffer","data":[0,9,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0]}
10/10/2020 11:54 | #121 | {"type":"Buffer","data":[0,24,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0]}
10/10/2020 12:54 | #115 | {"type":"Buffer","data":[13,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0]}
10/10/2020 12:54 | #110 | {"type":"Buffer","data":[18,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0]}
10/10/2020 12:54 | #119 | {"type":"Buffer","data":[100,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0]}
10/10/2020 13:54 | #115 | {"type":"Buffer","data":[13,13,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0]}
10/10/2020 13:54 | #110 | {"type":"Buffer","data":[18,18,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0]}
10/10/2020 13:54 | #119 | {"type":"Buffer","data":[100,100,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0]}
```

I left a terminal Grepping on DP overnight and this morning to see what was produced.

Every hour these DPs are sent by the device. It looks like a history of settings, and I'm assuming the following meaning of some:

- 115  - history of local temperature
- 110 - history of set temperature
- 119 - valve position

Seems the other DP codes are related to these as they start populating when the above codes are reset back to zero.

Maybe someone with a Tuya hub can see if these have some use in the App?

Sthopeless, 2020-10-11:
@insipiens the app is very simple and doesn't have a lot of options.  
I have this messages saved from old sniffs and I don't know what were my settings at the time. 

115
```
00 09 73 00 00 18 17 18 18 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```
110
```
00 0d 6e 00 00 18 15 14 14 14 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
```
119
```
00 11 77 00 00 18 0d 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

insipiens, 2020-10-11:
> @insipiens the app is very simple and doesn't have a lot of options.
> I have this messages saved from old sniffs and I don't know what were my settings at the time.
>
> 115
>
> ```
> 00 09 73 00 00 18 17 18 [...]

I logged into the app last night, it only does history (what you posted above), programming, away, and off/on

Which makes me think this device is pretty much done now.

skumka, 2020-10-11:
@insipiens This is great! thanks a lot. I confirm it works also for model SEA802:
zigbeeModel: ['88teujp', '88teujp\u0000'],
fingerprint: [{ modelID: '88teujp\u0000', manufacturerName: '_TYST11_c88teujp' }],

I think that in addtion to what is possible right now the solid baseline for those TRVs will be also:
- battery level

insipiens, 2020-10-11:
So I have put in a conversion for the low battery flag as described by @Sthopeless .

I will need to add further ones to read/describe the history and programming messages just to avoid any log errors.

Sthopeless, 2020-10-11:
> I logged into the app last night, it only does history (what you posted above), programming, away, and off/on
> 
> Which makes me think this device is pretty much done now.

I've tested your integration yesterday and it seemed to be working ok. The App doesn't much more to it, I also agree that will be it.

insipiens, 2020-10-11:
Submitted the PR to Dev

I left out the program and history DPs in the fromZigbee.js , they would have involved many extra lines of coding, IMO for little benefit to end users. 

If anyone has strong feelings on the matter they can always submit a PR, I left some annotation in the code about the structure of the messages.

Cheers

Ins

skumka, 2020-10-12:
@insipiens Thank you for a great job! Much appreciated...

BTW:
**Would you mind adding an input for SEA802 to devices.js** too?
zigbeeModel: ['88teujp', '88teujp\u0000'],
        fingerprint: [{ modelID: '88teujp\u0000', manufacturerName: '_TYST11_c88teujp' }],
        model: 'SEA802',
        vendor: 'Saswell',
        
 and **homeassistant.js**
'SEA802': [
        climate(5, 30, 'current_heating_setpoint', 0.5, ['off', 'heat'], [], ['manual', 'program']),
    ],

insipiens, 2020-10-12:
@skumka As I can't test that it might be better if you create a thread in the PR section yourself - tkink it would be added manually.

insipiens, 2020-10-12:
> 
> 
> @skumka As I can't test that it might be better if you create a thread in the PR section yourself - think it would be added manually.

Scrub that, I see you already did :)

skumka, 2020-10-12:
@insipiens Hi, I have tested it.  My request is already in. Koen has done a merge. Once again - thanks for a great job.

dsltip, 2020-10-13:
Regarding Valve position
I have asked SASWELL.COM: "Does it report valve position(open percentage) via Zigbee protocol? Can I see its position in HA application?"
The answer: "No, it can not."

m4tek, 2020-10-13:
> Regarding Valve position
> I have asked SASWELL.COM: "Does it report valve position(open percentage) via Zigbee protocol? Can I see its position in HA application?"
> The answer: "No, it can not."

Interesting, as this does not go in line with my observations. As mentioned in the first comment there is my capture sheet:
https://docs.google.com/spreadsheets/d/1BaKJdkmumf3S8rP4JfhNATyeryAGX9wh416llWWFACQ/edit#gid=0

message: 
`'{"type":"Buffer","data":[9,0,2,0,2,102,2,0,4,0,0,1,X]}'`
looks like providing opening of the valve in X value. This ranges for me between 1 and 57, while the upper limit changing after calibration.

m4tek, 2020-10-13:
BTW list of DPs as I classified them:

- 101 - Reporting local changes to ON and OFF
- 102 - Status reporting (temp or PIN position, depending on byte #12)
- 103 - Reporting local configuration change (setting parameters)
- 108 - Reporting switches between AUTO and MANUAL
- 110 - Reporting local change to program mode for 1/3 of a day 
- 115 - Reporting local change to program mode for 1/3 of a day
- 119 - Reporting local change to program mode for 1/3 of a day

Note, that I was only sniffing, so I have no data for sending data to the TRV.

insipiens, 2020-10-13:
If it does indeed report any DP codes I haven’t catered for they will appear in the logs (debug), you can grep the log command , remove ‘-f’ and pipe it into grep DP

I blocked out reporting of 110 to 122 in the script...anything else not discovered should appear.

insipiens, 2020-10-14:
Correction, blocked out 110>>>129 DP (reports of history and programming)

insipiens, 2020-10-14:
@m4tek yiu should probably close this thread now. Any follow up issues should be a new thread and any questions can still be asked here after closing.

[1 more reply]

closed 2020-10-15 by m4tek