Koenkk/zigbee2mqtt #2782: E180-ZG120B-TB support

szivgy, 2020-01-19:
Hi!

What code needs to be changed to support the E180-ZG120B-TB modules?

[Documentation](http://www.ebyte.com/en/pdf-down.aspx?id=1618)
[Buy](
https://www.aliexpress.com/item/4000531916570.html?spm=a2g0o.productlist.0.0.4ac63d48zwR2t3&algo_pvid=87e33758-f11b-4f94-8d69-261ed2ff1bf2&algo_expid=87e33758-f11b-4f94-8d69-261ed2ff1bf2-1&btsid=b5de8f90-879b-4d0b-b6d0-a45335b07be6&ws_ab_test=searchweb0_0,searchweb201602_6,searchweb201603_53)

Best Regards,
Gyula

kabturek, 2020-03-18:
One has to write a new adapter (probably for https://github.com/Koenkk/zigbee-herdsman as this is the future)

Koenkk, 2020-03-18:
A new adapter has to be implemented indeed.
To add this the following interface needs to be implemented (https://github.com/Koenkk/zigbee-herdsman/blob/master/src/adapter/adapter.ts).

I will close this as it's not a zigbee2mqtt but a zigbee-herdsman issue.

closed 2020-03-18 by Koenkk

Hedda, 2020-04-22:
@szivgy & @kabturek FYI, I posted a request for zigbee-herdsman (zigbee2mqtt relies on) here:

https://github.com/Koenkk/zigbee-herdsman/issues/168

That request suggests adding support to all EFR32MG1 and not just E180-ZG120B based hardware.