Koenkk/zigbee2mqtt #2787: Philips Hue Dimmer Switch Model 324131137411

shutupflanders, 2020-01-20:
# Bug Report
## What happened
I've successfully paired 3x Philips Hue Dimmer Switches, but when I press any buttons on them they flash the red LED and no output is detected in the logs.
I've been looking through the devices list and noticed that there's a model number mismatch (324131092621 vs 324131137411) - could this be a US/EU thing that produces slightly different results or something?

I've had the zigbee-herdsman running in debug mode and tried pressing buttons, but there's no output. Only when I repair the device (holding all 4 buttons for 5 seconds) does the device announce itself (but then continues to not work)

Logs [here](https://pastebin.com/X8A92M6a)

I've tried removing them using MQTT remove / force_remove, deleting them from the db, from MQTT explorer, HomeAssistant, then repairing them, but I have the same issue.

## What did you expect to happen
An MQTT message of the devices state

## How to reproduce it (minimal and precise)
Reset (hold pin in the back for 10 seconds), Pair the device with Zigbee2MQTT, Press any button after successful pairing

## Debug Info
zigbee2mqtt version: 1.9.0 (Hassio)
CC253X firmware version: 
Coordinator firmware version: '{"type":"zStack12","meta":{"transpor
trev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'

shutupflanders, 2020-02-05:
@Koenkk sorry to tag you mate, but could I get some advice on the above? I tried going down the debug route, dropping console logs everywhere, but I'm out of my depth.
These devices are the only ones left on my proprietary hubs, once I convert them over I'm free from the cloud.

Koenkk, 2020-02-05:
Can you try reconfiguring the devices?

- right before sending the command below press a button on the remote to wake them up
- Send https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure
- Check if you see `Successfully configured...` in the log.

shutupflanders, 2020-02-05:
Cheers @Koenkk tried the suggestion, here are the logs for a couple of those attempts.
[https://pastebin.com/bu2HRHw0](https://pastebin.com/bu2HRHw0)
[https://pastebin.com/dMSV2v2X](https://pastebin.com/dMSV2v2X)

If I force remove the item and re-add it, I sometimes get "Successfully configured" but then it doesn't respond to button presses as mentioned in the original post.

Koenkk, 2020-02-05:
Can you try and provide the log of:
- Remove the device (not force remove!) this should be possible by waking the device up right before sending the command (to wake it up press some buttons on it)
- Pair the device again
- After you see `Successfully configured..` press buttons on the remote.

shutupflanders, 2020-02-05:
Sure, 

Removal [https://pastebin.com/FXRrAfSP](https://pastebin.com/FXRrAfSP)
Pairing [https://pastebin.com/bwMTeSnb](https://pastebin.com/bwMTeSnb)
And pressing the buttons after that logs nothing, the red LED appears on the device as before.

(apologies for the log formatting being different between pastes, the Hassio log only shows so much on screen so I had to dip into the log.txt for the rest)

Koenkk, 2020-02-05:
The pairing log does not contain the herdsman debug logging (removal does), can you provide the herdsman debug log for the pairing?

shutupflanders, 2020-02-05:
I'll try, it flies through in hassio that fast it's hard to get

shutupflanders, 2020-02-05:
[https://pastebin.com/1qg6FeQR](https://pastebin.com/1qg6FeQR)

Managed to get them using `docker logs [container]`
Let me know if you need me to try anything else.

Koenkk, 2020-02-05:
Can you update to the latest dev branch (edge for hassio) and provide the log again? (there are some interview and logging improvements there)

shutupflanders, 2020-02-05:
I've been on edge for about 10 days now, the logs you see above are using that version

Koenkk, 2020-02-05:
Yes, but not the latest edge. Edge is updated on every commit to zigbee2mqtt, to get the latest edge uninstall/install the addon.

shutupflanders, 2020-02-05:
No worries, I've just done that, removed the device and repaired:
[https://pastebin.com/gAt7nvBK](https://pastebin.com/gAt7nvBK)

Koenkk, 2020-02-06:
- Interview looks good now, did the configuring went successful? (the log stops before that).
- If yes, can you provide the log now when pressing buttons on the remote?

shutupflanders, 2020-02-06:
`zigbee2mqtt:info  2020-02-06 14:48:38: Successfully configured '0x0017880104f3f23d'`
Still nothing in the logs after this when pressing buttons, LED still shows red

Koenkk, 2020-02-06:
In that case, I don't see any other option than sniffing the traffic of the whole pairing procedure, configure and pressing the buttons afterwards. See https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

shutupflanders, 2020-02-06:
Ok, I'd hoped it wouldn't come to this as it's more kit I need to buy, might pick it up again at a later date, thanks anyway

shutupflanders, 2020-02-06:
```2020-02-06T15:41:33.348Z zigbee-herdsman:controller:endpoint Bind 0x0017880104f3f23d/2 genPowerCfg from '0x00124b000be5ab6d/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)
zigbee2mqtt:error 2020-02-06 15:41:33: Failed to configure '0x0017880104f3f23d', attempt 6 (Error: Bind 0x0017880104f3f23d/2 genPowerCfg from '0x00124b000be5ab6d/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:235:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))```

I did get that a couple of times whilst trying to reconfigure it, is that of any use?

Koenkk, 2020-02-06:
Btw, when pairing, did you completely turn of your hue bridge and network (devices still attached to it)?

shutupflanders, 2020-02-06:
I haven't tried that, no - I'll give it a go

shutupflanders, 2020-02-06:
No luck unfortunately, same issue

Koenkk, 2020-02-06:
For some reason all of the communications fails and I dont know why, did you also try pairing it close to the coordinator?

shutupflanders, 2020-02-06:
I've been holding it <6 inches away from the coordinator since I began attempting to pair it

Koenkk, 2020-02-06:
Maybe as a last try, can you disconnect the batteries or replace them for a few mins?

shutupflanders, 2020-02-06:
No luck unfortunately, Something I have noticed though - when I trigger a reconfigure, and start mashing the buttons to keep it away, it fails (see [previous timeout log](https://github.com/Koenkk/zigbee2mqtt/issues/2787#issuecomment-582965674) )
If I don't touch the buttons, I get "Successfully configured..."

shutupflanders, 2020-02-06:
I will quite happily post you this device if you could get it working, my house is full of them

Koenkk, 2020-02-07:
I don't think it has to something to do with the device itself (especially if you already tried multiple of them). Probably it has something to do with the environment as they somehow seem to immediately disconnect (and connect to a different network??).

shutupflanders, 2020-02-07:
[image]
Could it be because it's not connected directly to the coordinator? looks like it's connecting to one of my bulbs the floor below me, maybe it's not forwarding the commands correctly?

shutupflanders, 2020-02-07:
I've just setup my old coordinator on a spare rasp pi running the dev branch (no homeassistant) after a few attempts of pairing it with that, I've managed to get it to pair and receive commands!

[https://pastebin.com/iyrDYf4h](https://pastebin.com/iyrDYf4h)

I wonder if it is the problem I suggested above, that because it's not directly connected with the coordinator in my hassio network, it's not receiving data correctly

shutupflanders, 2020-02-07:
Finally got all my switches moved over, the messages switch between the primary and secondary network and seem to be working ok.

Must be my coordinator in some way or form that is the issue, I don't know why - but it's working now so I'm not going to touch it.

Koenkk, 2020-02-07:
Good that its working now (although its still not clear what the actual issue was). Hopefully its a one-time issue

closed 2020-04-14 by stale[bot]

nobodypb, 2020-08-29:
Hi, it seems like I unfortunatly got excatly the same issue with that very model of hue dimmer switch.
I haven't investigated that much yet, but it shows the same symtoms. Pairing goes fine, but then on each button press, only the red LED lights up and no command is recieved.

shutupflanders, 2021-02-08:
Just upgraded to the latest version, upgraded my stick to a CC2652RB, obviously had to repair everything and now I'm back where I started getting these dimmer switches working, even the old separate network fix before didn't work.