Koenkk/zigbee2mqtt #2788: rejoin RM01 breaks zigbee2mqtt

prutseltje, 2020-01-20:
# Bug Report
## What happened
When resetting RM01 to rejoin zigbee network the controller crashes

## What did you expect to happen
RM01 joining the zigbee network

## How to reproduce it (minimal and precise)
rejoin RM01, the controller software crashes and the CC2530 stick needs to be reset (remove/insert stick) before zigbee2mqtt starts again.

## Debug Info
zigbee2mqtt version: 1.9.0-dev (commit #43c745d)
CC253X firmware version: 20190608

Debug log
```
  "0xd85def11a100445a" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{Busch-Jaeger Zigbee Light Link woonkamer|0xd85def11a100445a (12550)failed: lqi|Busch-Jaeger Zigbee Light Link power supply/relay/dimmer (6735/6736/6737)|2020-01-19T14:05:02+01:00}"];
}'
(node:24625) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtLqiReq after 6000ms
    at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at listOnTimeout (internal/timers.js:531:17)
    at processTimers (internal/timers.js:475:7)
(node:24625) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 150)
```

Koenkk, 2020-01-21:
Please try with the source routing firmware, it should improve stability: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

prutseltje, 2020-01-22:
@Koenkk Thnx! after rejoining all devices now everything works as expected!

closed 2020-01-22 by prutseltje