Koenkk/zigbee2mqtt #2790: zigbee-pro v1.2 HA profile support

dafzor, 2020-01-20:
I'm trying to add some "old" devices that seem to be based on the Zigbee Pro v1.2.0 HA profile.

However they fail the interview stage:
```
zigbee2mqtt:info  2020-01-20 19:46:38: Starting interview of 'cloogy_socket'

zigbee2mqtt:info  2020-01-20 19:46:38: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"cloogy_socket"}}'

zigbee2mqtt:error 2020-01-20 19:47:58: Failed to interview 'cloogy_socket', device has not successfully been paired

zigbee2mqtt:info  2020-01-20 19:47:58: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"cloogy_socket"}}'

zigbee2mqtt:debug 2020-01-20 19:48:39: Saving state to file /app/data/state.json
```

Is the previously mentioned stack no supported with a cc2531 coordinator?

Koenkk, 2020-01-21:
Can you provide the herdsman debug logging when pairing the device?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

dafzor, 2020-01-21:
Missunderstood that setting debug log level in configuration was equivalent to DEBUG=*, bellow is the log you wanted.
```
zigbee2mqtt:info  2020-01-21 19:22:27: Starting interview of 'cloogy_socket'

winston:file written true false

zigbee2mqtt:info  2020-01-21 19:22:27: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"cloogy_socket"}}'

winston:file written true false

  zigbee-herdsman:controller:device Interview - start device '0x00124b0001aaa4ec' +3m

  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":40588,"nwkaddrofinterest":40588} +2m

  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,140,158,140,158,35] +2m

  serialport/stream _write 9 bytes of data +34ms

  serialport/binding-abstract write 9 bytes +2m

  serialport/bindings/unixWrite Starting write 9 bytes offset 0 bytesToWrite 9 +2m

  serialport/stream _read reading { start: 427, toRead: 65109 } +10ms

  serialport/binding-abstract read +10ms

winston:file logged 3812 info  2020-01-21 19:22:27: Starting interview of 'cloogy_socket'

winston:file logged 3985 info  2020-01-21 19:22:27: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"cloogy_socket"}}'

  serialport/bindings/unixRead Starting read +47ms

  serialport/bindings/unixWrite write returned: wrote 9 bytes +14ms

  serialport/bindings/unixWrite Finished writing 9 bytes +1ms

  serialport/stream binding.write write finished +6ms

  serialport/bindings/unixRead read error [Error: EAGAIN: resource temporarily unavailable, read] {

  errno: -11,

  code: 'EAGAIN',

  syscall: 'read'

} +4ms

  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +1ms

  serialport/bindings/poller Polling for "readable" +56ms

  serialport/bindings/poller received "readable" +0ms

  serialport/bindings/unixRead Starting read +2ms

  serialport/bindings/unixRead Finished read 6 bytes +1ms

  serialport/stream binding.read finished { bytesRead: 6 } +6ms

  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102] +46ms

  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102] +1ms

  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102 +0ms

  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0} +2m

  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms

[... 487 lines cut ...]

  zigbee-herdsman:controller:device Interview - active endpoints request failed for '0x00124b0001aaa4ec', attempt 2 +10s

  zigbee-herdsman:controller:device Interview - failed for device '0x00124b0001aaa4ec' with error 'Error: Interview failed because can not get active endpoints ('0x00124b0001aaa4ec')

  zigbee-herdsman:controller:device     at Device.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/device.js:273:23)

  zigbee-herdsman:controller:device     at Generator.throw (<anonymous>)

  zigbee-herdsman:controller:device     at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/device.js:6:65)' +2ms

  zigbee-herdsman:controller:error Interview failed for '0x00124b0001aaa4ec with error 'Error: Interview failed because can not get active endpoints ('0x00124b0001aaa4ec')' +5m

zigbee2mqtt:error 2020-01-21 19:23:47: Failed to interview 'cloogy_socket', device has not successfully been paired

winston:file written true false

zigbee2mqtt:info  2020-01-21 19:23:47: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"cloogy_socket"}}'

winston:file written true false

winston:file logged 4089 error 2020-01-21 19:23:47: Failed to interview 'cloogy_socket', device has not successfully been paired

winston:file logged 4261 info  2020-01-21 19:23:47: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"cloogy_socket"}}'
```

Koenkk, 2020-01-22:
Do you perhaps still have the original bridge, if so, could you sniff the traffic when pairing it to that? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

dafzor, 2020-01-22:
I'll see what I can do.

closed 2020-03-29 by stale[bot]