Koenkk/zigbee2mqtt #2791: I am having problem getting data from device

NopeHere, 2020-01-20:
I am trying to get data from my device: WSDCGQ11LM
It is found at sending data to the MQQ
I looks like it is found and sending data:
zigbee2mqtt:info  2020-01-20 20:26:16: MQTT publish: topic 'zigbee2mqtt/0x00158d00044dade0', payload '{"battery":86,"voltage":2975,"temperature":16.49,"humidity":65.76,"pressure":1036.3,"linkquality":147}'

Now my problem is whoich command should i send to get the data on demand?
I am trying with:
 mosquitto_pub -h localhost -p 8883 -m '{"state":""}' -t zigbee2mqtt/0x00158d00044dade0/get -d
I am getting:
zigbee2mqtt:error 2020-01-20 20:28:16: No converter available for 'state' ()

I would like to get the battery or temperature, are there any way to get this, if you could please
på the entire command it would be greate as i am still a newbee at this..

I have been trying to get the command woring for a few dayes now to any help will be gladly accepted.. :)

Kind regards

NopeHere, 2020-01-20:
Just for info i have read https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html

But I just cant convert it to useful commands i can use.. :( i am a newbee for sure.. :)

Kind regards

Koenkk, 2020-01-21:
The WSDCGQ11LM is an end device which is normally sleeping, therefore you cannot retrieve data from it on demand.

closed 2020-01-21 by Koenkk

NopeHere, 2020-01-21:
Ok, thank you :)

If it was a device i could retrieve data from, was the command ok then ?
Just want to know i I am on the right track..

Kind regards

Koenkk, 2020-01-21:
If it was e.g. a bulb it would indeed return the state.