Koenkk/zigbee2mqtt #2792: Instability in my Zigbee Network, Reason unknown

arjanvroege, 2020-01-20:
Bug report:
- First make sure you are on the latest version of zigbee2mqtt.
- Make sure you are running the latest CC253X firmware from https://github.com/koenkk/z-stack-firmware.
- Provide a clear description of the problem.
- If applicable, provide steps how to reproduce the problem.
- Provide a log with `log_level` set to `debug`, see https://www.zigbee2mqtt.io/configuration/configuration.html
- DON'T copy logs directly here, post a link to https://hastebin.com/ or https://pastebin.com/.
-->

# Bug Report
## What happened
From time to time bulbs cannot be controlled with Zigbee2MQTT through HA. Most of the time the bulb can then only be controlled through a remote directly attached with the group functionality. When this occurs I see the following in the logs:

zigbee2mqtt:debug 2020-01-20 22:09:53: Received MQTT message on 'zigbee2mqtt/tradfri-slaapkamer-ouders-led1/set' with data '{"state": "OFF"}'
zigbee2mqtt:debug 2020-01-20 22:09:53: Publishing 'set' 'state' to 'tradfri-slaapkamer-ouders-led1'
zigbee2mqtt:debug 2020-01-20 22:09:53: Received MQTT message on 'zigbee2mqtt/tradfri-slaapkamer-ouders-led2/set' with data '{"state": "OFF"}'
zigbee2mqtt:debug 2020-01-20 22:09:53: Publishing 'set' 'state' to 'tradfri-slaapkamer-ouders-led2'
zigbee2mqtt:debug 2020-01-20 22:09:53: Received MQTT message on 'zigbee2mqtt/tradfri-slaapkamer-ouders-led3/set' with data '{"state": "OFF"}'
zigbee2mqtt:debug 2020-01-20 22:09:53: Publishing 'set' 'state' to 'tradfri-slaapkamer-ouders-led3'
zigbee2mqtt:info  2020-01-20 22:09:53: MQTT publish: topic 'zigbee2mqtt/tradfri-slaapkamer-ouders-led2', payload '{"state":"OFF","brightness":174,"color_temp":380}'
zigbee2mqtt:info  2020-01-20 22:09:53: MQTT publish: topic 'zigbee2mqtt/tradfri-slaapkamer-ouders-led3', payload '{"state":"OFF","brightness":174,"color_temp":380}'
zigbee2mqtt:error 2020-01-20 22:10:08: Publish 'set' 'state' to 'tradfri-slaapkamer-ouders-led1' failed: 'Error: Timeout - 1695 - 1 - 3 - 6 - 11 after 15000ms'
zigbee2mqtt:debug 2020-01-20 22:10:08: Error: Timeout - 1695 - 1 - 3 - 6 - 11 after 15000ms
    at Timeout._onTimeout (/zigbee2mqtt-1.9.0/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at listOnTimeout (internal/timers.js:531:17)
    at processTimers (internal/timers.js:475:7)
zigbee2mqtt:info  2020-01-20 22:10:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'tradfri-slaapkamer-ouders-led1' failed: 'Error: Timeout - 1695 - 1 - 3 - 6 - 11 after 15000ms'","meta":{"friendly_name":"tradfri-slaapkamer-ouders-led1"}}'

Sometimes when I wait a day the bulb start to respond to zigbee2mqtt but when this not happens I need to either:
- Restart Zigbee2MQTT;
- Unplug and reset the CC2531;
- Restart my NUC with the above step;

I've had the same instability 6 till 12 months ago and then changed to the source routing firmware and added the following to the configuration.yaml:

queue: {
  delay: 75,
  simultaneously: 5
}

This improved the situation a bit but the instability increased after adding another set of IKEA bulbs. My network has 32 devices ( 5 Xiaomi sensors and 27 IKEA bulbs, different types). 

## Debug Info
zigbee2mqtt version: 1.9.0
CC2531 firmware version: 20190619

nldroid, 2020-01-21:
Can't help you with this but i have the same with one of my Ikea bulbs. What helps for me is to turn power off from the lamp, wait a few seconds and put power back on.
I have the feeling that it's a bug in the Ikea bulb. I've read more reports here on timeouts with Ikea bulbs.

arjanvroege, 2020-01-21:
Yesterday evening multiple bulbs were failing. After stopping Zigbee2MQTT and unplugging and resetting the CC2531 and starting Zigbee2MQTT the issues were gone. I think it's somewhat related to the CC2531. I've tested another CC2531 stick and having the same issues. My main question at this moment is. Will an upgrade to the CC26X2R1 adapter solve these issues or is this a bug?

Koenkk, 2020-01-21:
@arjanvroege probably yes, improved range and much more processing power/memory.

GrumpyMeow, 2020-01-26:
For me it's the same and it's getting quite annoying. Everything seems to work for a few days and then devices start to become "disconnected". I have sth like 13 Ikea Leptiter lamps which all are routers. Thus "range" shouldn't be the problem.
I get "timeouts when publishing set"  or "No route to host" kind of error messages.

mr2c12, 2020-02-01:
I suffer from the same problem. One of my bulb sometimes become unavailable with the error: "Error: Timeout - 17800 - 11 - 129 - 6 - 11 after 15000ms".
There are a few strange things to point out:
- this used to happen with an IKEA e14 bulb, but the very same occurs now that I've replaced it with a Philips Hue bulb
- it happens maybe once a week
- the coordinator is quite far (link quality should be close to zero), but in sight there are a CC2530 router, an IKEA router, an IKEA bulb (that does not suffer from the same problem)
- a battery powered end device is close to this problematic bulb but it never has a connection problem

The coordinator is a CC2530+CC2591 fith fw 1.2 (no source routing). This bulb seems to be the only one with a problem.

Koenkk, 2020-02-02:
@mr2c12 maybe the problem is with that specific place, could there be a lot of interference there?

mr2c12, 2020-02-02:
@Koenkk I think it's unlikely a problem with interferences. The bulb is quite near a lot of other devices that works.
I am sincerely suspecting some electrical problem with the bulb older.
If I set log to "debug", am I able to understand if the bulb comes back "online" because it just lost power or connectivity?

Koenkk, 2020-02-02:
You may see a `deviceAnnounce` when the bulb comes online again when debug logging is enabled.

mr2c12, 2020-02-06:
> @mr2c12 maybe the problem is with that specific place, could there be a lot of interference there?

@Koenkk  what are the possible interference that can affect the bulb connection?
Is 220v power able to affect the zigbee connection?
I really can't see anything else (since WiFi is far from this bulb and closer to other non-problematic zigbee devices).

dzungpv, 2020-02-18:
@mr2c12 it could cause by wifi, bluetooth devices. You can use channel 26 to minimize interference.
I have instability in my network too, even with most power chips: cc1352/cc2652. Many reason cause that

closed 2020-04-25 by stale[bot]

Martin004, 2020-10-04:
I had similar problems.
After using a 10cm USB extension for the ZIGBEE stick, the connection was better.
If the stick is plugged directly into the device (computer or Pi), I would first use the extension to decouple it.