Koenkk/zigbee2mqtt #2795: CC1352p-2 timeout error stucked at starting zigbee-herdmen

guitar9, 2020-01-21:
I am using zibgee2mqtt version 1.9.0. 
I connected the board with the cable included in the box from texas instruments and flashed it with Uniflash (like in the readme descriped).
The I run the docker container with this command.
```
docker run \
   -it \
   -v $(pwd)/data:/app/data \
   --device=/dev/ttyACM0 \
   -e TZ=Europe/Amsterdam \
   -v /run/udev:/run/udev:ro \
   --privileged=true \
   koenkk/zigbee2mqtt
```
sometimes I am getting timeout error and sometimes the output stucks at 
`zigbee2mqtt:info  2020-01-20 21:34:27: Starting zigbee-herdsman...`

What I am doing wrong?

Koenkk, 2020-01-21:
Can you provide the logging when running with herdsman debug logging?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

guitar9, 2020-01-21:
This is the output. I am using newest firmware 3.x.0 for CC1352p-2

```
zigbee2mqtt:info  2020-01-22 00:43:35: Logging to console and directory: '/app/data/log/2020-01-22.00-43-35'
zigbee2mqtt:debug 2020-01-22 00:43:35: Can't load state from file /app/data/state.json (doesn't exist)
zigbee2mqtt:info  2020-01-22 00:43:35: Starting zigbee2mqtt version 1.9.0 (commit #287f843)
zigbee2mqtt:info  2020-01-22 00:43:35: Starting zigbee-herdsman...
zigbee2mqtt:debug 2020-01-22 00:43:35: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"}}'
zigbee2mqtt:error 2020-01-22 00:43:42: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-01-22 00:43:42: Failed to start zigbee
zigbee2mqtt:error 2020-01-22 00:43:42: Exiting...
zigbee2mqtt:error 2020-01-22 00:43:42: Error: SRSP - SYS - version after 6000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at listOnTimeout (internal/timers.js:531:17)
    at processTimers (internal/timers.js:475:7)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.9.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.9.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-01-21T23_43_42_939Z-debug.log
```

Koenkk, 2020-01-22:
Looks like a communication issue or maybe incorrect flash of the CC1352-p2. Did you also execute step 5 of https://www.zigbee2mqtt.io/information/flashing_via_uniflash.html ?

guitar9, 2020-01-22:
yes I did all step. After a lot of time I found out changing the pan_id solved the problem

closed 2020-01-22 by guitar9

guitar9, 2020-01-23:
Maybe adding a better error message would help to solve the problem faster....

runningman84, 2020-02-26:
Changing the pan_id also means you loose all your existing devices...?!

Is this really the only way to bring the network up again?