Koenkk/zigbee2mqtt #2797: OSRAM Motion+ Sensor (CC2531 Source problem) - Occupancy, Temperature problem

vencakratky, 2020-01-21:
# Bug Report
Osram Smart Motion sensor always shows Occupancy - false, Temperature - false, 
## What happened
After new firmware CC2531 with source routing i cannot change states of this sensor
## What did you expect to happen
With default firmware working well, after source routing Problems began.
## How to reproduce it (minimal and precise)
I tried delete device via MQTT explorer, after delete in "/share/zigbee2mqtt/database.db", new repairing, the problem still remains.. What I found, in database.db is string ""dateCode":"����������������"...Maybe is this problem?
## Debug Info
There is debug log and string:  https://pastebin.com/KuQA5PU6

zigbee2mqtt version:  1.9.0  and i tried last Zigbee2mqtt Edge
CC2531 firmware version: 20190619 - Source routing

Koenkk, 2020-01-22:
Please provide the herdsman debug log when pairing the device with the latest edge.

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

Koenkk, 2020-01-22:
You need to put `zigbee_herdsman_debug` under the root, not under `advanced` (otherwise we won't get debug logging).

vencakratky, 2020-01-22:
I apologize for my incompetence
https://pastebin.com/rPKXUqD4

vencakratky, 2020-01-22:
Today i bought Ikea Tradfri Motion sensor for testing and now i found the same problem like with Osram Motion sensor. Maybe it's my fault. 
I had first USB CC2531 with default firmware (20190608). With this Adapter i had little problem with Osram Bulbs (MAC no Ack)...So i have ordered the second USB CC2531 (from AliExpress) and this second Dongle i tried flash with source routing Firmware (20190619). This second Dongle its now in Home Assistant (on Rpi4 - HassOs).
It´s possible, thats Problems causing this "new" CC2531 Dongle? Other zigbee devices (except motion sensors) function correctly

Koenkk, 2020-01-23:
https://github.com/Koenkk/zigbee2mqtt/issues/2797#issuecomment-577384680 this is only a small part of the log, please provide the complete log.

vencakratky, 2020-01-23:
Today morning i tried again the first USB CC2531 (with flashed source routing). At first there was still a problem, but after 5-6 hours i saw Osram Motion sensor began show data values. I don't know if ZigBee's miracle or "artificial intelligence" :-)  
Anyway I think the problem is when replacing the USB CC2531 coordinator (with default-firmware) for another dongle CC2531 (Source routing).
It´s possible, that exactly same type CC2531 coordinator affects the functionality of the zigbee network (maybe different MAC Address)? If yes, please create some guide, how to do it, if the existing CC2531 is damaged and you need to replace with another piece (of same type CC2531) .
After your answer, I think we can close the Issue. Thank you! 

[image]

Koenkk, 2020-01-24:
Yes, for these kind of devices (ias) it indeed might be needed to repair them after switching coordinator (only when on zstack 1.2)

vencakratky, 2020-01-24:
Thank you verry much :-)

closed 2020-01-24 by vencakratky