Koenkk/zigbee2mqtt #2798: Xiaomi buttons no longer working

BarBaar44, 2020-01-21:
#2789 Bug Report
## What happened
Xiaomi wireless buttons devices no longer work. Nog logging appears with log_level: debug and "zigbee_shepherd_debug": "true". Status in HA is "unknown"

## What did you expect to happen
See some actions from the buttons when pressed.

## How to reproduce it (minimal and precise)
Previously I wat running zigbee2mqtt nightly. When I switched to 1.19.0, the buttons just don't work anymore. Other Xiaomi devices (curtain controllers for example) do work.
## Debug Info
zigbee2mqtt version: 1.19.0
CC253X firmware version: CC2531_DEFAULT_20190608

BarBaar44, 2020-01-24:
Tried so far:

- reinstalled version 1.19.0
- Moved back to edge version
- used a different CC2531

All no result ...

Koenkk, 2020-01-25:
This indicates that the button does not send any messages at all anymore.

Did you try to repair the buttons or replace the batteries?

BarBaar44, 2020-01-26:
No, but the issue is: I have the exact same problem will all 4 of my buttons (2 different types).

Have not yet tried to re-pair the buttons, but I am afraid this will be the only option left.

BarBaar44, 2020-01-26:
Ended up re-paring all buttons. Weird they just stopped working...

BarBaar44, 2020-01-27:
Reopening this one, as I lost both Xiaomi WXKG01LM buttons again today. They just no longer work..

WXKG02LM buttons are still working.

sander816, 2020-01-28:
i have the same. Every day they stop working, no response anymore in zigbee2mqtt
Just updated to latest dev, hopefully this will fix it

Koenkk, 2020-01-28:
When this happens, could you provide the sniff log when pressing the button? (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html) This should give us an idea were the issue is coming from.

BarBaar44, 2020-02-09:
Closing this as, as this seems to be solved since I started using a CC2530 + CC2591 router

closed 2020-02-09 by BarBaar44