Koenkk/zigbee2mqtt #2799: Availability ping reply does not update last_seen-field

Kryzek, 2020-01-22:
I tried to enable `last_seen` -option and I noticed that when using availability function to check device states the response from ping does not update the last seen on pinged device.

```
Received Zigbee message from '0x00124b001d3ab1b9', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 11 with groupID 0
Successfully pinged '0x00124b001d3ab1b9'
```
This does not get followed by mqtt-publish.

I think this feature should also update last_seen.

Koenkk, 2020-01-22:
I don't know yet if this is desired behaviour. In that case we should also update the last seen for every message received from the device (current only done for converted messages).

sjorge, 2020-01-22:
I guess it depends on what you consider 'last_seen'... is it I saw something (anything) of the device... or last time the device send something useful to us (coordinator).

I wasn't even aware this was published in the mqtt payload, I thought it was used for tracking availability.

Kryzek, 2020-01-23:
Oh, I thought that last_seen would be updated every time device is "seen" - meaning any message that is from that particular device.

closed 2020-03-30 by stale[bot]