Koenkk/zigbee2mqtt #2800: Does aqara/ikea end devices connects to bulb-routers?

mr2c12, 2020-01-22:
I currently have:
- cc2530+cc2591 coordinator (firmware 190523 3.x)
- cc2530 router
- two Ikea light bulbs
- one philips hue light bulbs
- a bunch of aqara or ikea end devices: motion sensors, buttons..

It seems that light bulbs can connect each other (as I expect, given that they are routers). But end devices, both aqara and ikea, does not connect to the light bulbs, they connect to the router or the coordinator (depending on their position), even if a bulb is closer.

Is it correct? I thought it was some sort of problem with Ikea bulbs, but the same happens with the Philips Hue bulb.
Can I do something to improve the situation?

ikbenfrank, 2020-01-22:
Hey I had a setup with a cc2531 and around 10 ikea lights and a few aqara sensors/ikea buttons. The endpoints connected to the bulbs after a few days but not always based on physical distance. Sometimes the light was closer than the coordinator but it chose the coordinator.

groenmarsmannetje, 2020-01-22:
I can confirm that they will be able to connect. Just give the mesh netwerk some time to settle. When you have more bulbs as routers you even do not need to keep the aqara sensors near bulb or router for their first time pairing.

mr2c12, 2020-01-24:
After a few days, I can confirm that the answer is YES.
Thanks!

closed 2020-01-24 by mr2c12

mr2c12, 2020-01-29:
Quick addiction: YES and aqara end device connects to ikea bulb, as confirmed today in my setup.