Koenkk/zigbee2mqtt #2801: brightness_move / moveWithOnOff

fnfid, 2020-01-22:
Hi, 
brightness_move works quite fine now (tested on various IKEA bulbs), but tried to figure out how to prevent bulbs from turning to "OFF" after sending {"brightness_move": -100}. 
Tried al lot even to read/alter the converter code (toZigbee.js), but did not suceed unfortunally. 
If there is easy help outside i would be very thankful.

zigbee2mqtt version: latest dev
CC253X firmware version: CC1352P-2 latest

fnfid, 2020-01-23:
First thing i figured out ist, that "brightness_move":"stop" for groups result in: 

error 2020-01-23 09:08:15: Publish 'set' 'brightness_move' to 'Wohn' failed: 'TypeError: entity.read is not a function'
debug 2020-01-23 09:08:15: TypeError: entity.read is not a function
    at Object.convertSet (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:243:30)
    at async EntityPublish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/entityPublish.js:173:36)
    at async Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:327:21)

As code look briliant, this is not clear, but must be somewhere in the converter-file.

Second, i noticed that identical IKEA remotes (same model+firmware), when coupled directly, sometimes send  {"movemode":1,"rate":84}, some other case {"stepmode":1,"stepsize":43,"transtime":5}.... maybe somthing weird in the IKEA firmware for compatibility to history reasons...

Koenkk, 2020-01-23:
Exception should be fixed in latest dev branch, thanks for reporting.

fnfid, 2020-01-24:
Great Work! While removing some workarounds same thing exists for gathering state informations in groups, in this case i send {"state":""} to .../{group]/get, so my guess is same solution in '' at line 113 or removing the emty version of state in line 336.

info  2020-01-24 07:22:58: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'get' 'state' to 'Wohn' failed: 'TypeError: entity.read is not a function'","meta":{"friendly_name":"Wohn"}}'
error 2020-01-24 07:22:58: Publish 'get' 'state' to 'Oben' failed: 'TypeError: entity.read is not a function'
debug 2020-01-24 07:22:58: TypeError: entity.read is not a function
    at Object.convertGet (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:113:26)
    at Object.convertGet (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:336:45)
    at EntityPublish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/entityPublish.js:200:37)
    at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:327:44)
    at Controller.onMQTTMessage (/opt/zigbee2mqtt/lib/controller.js:239:14)
    at MQTT.emit (events.js:223:5)
    at MQTT.onMessage (/opt/zigbee2mqtt/lib/mqtt.js:91:14)
    at MqttClient.emit (events.js:223:5)
    at MqttClient._handlePublish (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:1162:12)
    at MqttClient._handlePacket (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:351:12)

Koenkk, 2020-01-24:
While indeed its an error, `get` commands can never be send to groups.

fnfid, 2020-01-26:
Yep, sounds correct. It was only cause i thougt state request  {"state":""}  via 'get' also works for groups, broadcasting this to devices, as broadcasts reduce traffic and this part simplifies code as one does not have to determine between devices and groups. But as reporting even for sofisticated cases works fine, everythings up and running. Thanks!

closed 2020-01-28 by fnfid