Koenkk/zigbee2mqtt #2802: Failed to execute LQI for '0x...'

arnold-maderthaner, 2020-01-22:
Hi !

I'm running on zigbee2mqtt 1.9.0 and latest CC2531 firmware.
If I try to check my network to see if everything is connected I always get the above error from different routers like: AC10691 - OSRAM Smart+ plug (Router)
and ZNCZ02LM - Xiaomi Mi power plug ZigBee (Router)

If I restart the ZigBee router it sometimes works to clear the error. 
What does the error exactly mean and how can I fix it permanently ?

kennymc-c, 2020-01-22:
I also noticed these log messages (4-8 times a day). But they only appear since the CC2531 firmware update from 20190425 to 20190610. Today I had no new data from all routers in my network (CC2530, Ikea Tradfri Repeater, Innr SP120) and had to restart Zigbee2MQTT and power cycle all routers with allow join enabled. Before the update I didn't have such errors.

Koenkk, 2020-01-22:
If you get this often, you might want to try the source routing firmware (20190425 was also using source routing): https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

kennymc-c, 2020-01-25:
I noticed that after restarting all routers 3 days ago, the LQI error messages no longer appear. I think switching to the source routing firmware doesn't make much sense anymore.

Wuppi68, 2020-02-05:
I have a hint
I got the same LQI messages AND can't control the failed devices

CC2530 with 
commit | 287f843 | 2020-02-05 21:00:40
coordinator_meta_maintrel | 3 | 2020-02-05 21:00:40
coordinator_meta_majorrel | 2 | 2020-02-05 21:00:40
coordinator_meta_minorrel | 6 | 2020-02-05 21:00:40
coordinator_meta_product | 0 | 2020-02-05 21:00:40
coordinator_meta_revision | 20190608 | 2020-02-05 21:00:40
coordinator_meta_transportrev | 2 | 2020-02-05 21:00:40
coordinator_type | zStack12

Ikea Repeater
Ikea Bulbs RGB and WW (bad network connection from repeater and coordinator) using mesh from other bulbs

my solution - just tested - restart the bulb to rejoin the mesh

why this happens I have no clue ;-)

If you need further informations, contact me ;-)

xefil, 2020-02-23:
Hello!
I've the same issue, in that way I'm not able to create correctly the network map.
It happens to my two fresh installed CC2531 with firmware `router-cc2531-diag-usb.hex`.
Any idea?

gimlichael, 2020-02-24:
I have encountered this issue migrating to the docker variant of this otherwise fine software. What to do?

gimlichael, 2020-02-24:
> I have encountered this issue migrating to the docker variant of this otherwise fine software. What to do?

I migrated back to the starting point; same error. This is quite troubling :-(

gimlichael, 2020-02-24:
The weird part is, that i can see the link quality eventually in the log .. but the routers all fail with the `Failed to execute LQI`. Everything worked earlier .. until i migrated to linux and docker .. when that failed big time, i went back to my origin .. just to see the error emerge here as well .. now, that is very strange .. sorry to spam this thread .. just frustrated .. and tired :-(

gimlichael, 2020-02-24:
OK .. latest update .. i went back to my original purchased sniffer (without external antenna) .. now only 1 of 3 of the repeaters/routers is having the LQI error .. the two others are working.

Maybe the sniffer with the external is semi-friend? It failed on all 3 of the repeaters .. BUT everything else was working .. time for bed ..

Gromina, 2020-02-25:
I'm facing the same problem with IKEA E1603 Tradfri Control Outlet Zigbee

gimlichael, 2020-02-25:
Okay, for me the solution was far from optimal; i had to delete the database.db and start over by re-joining all Zigbee devices.

The good thing is, that it seems to be working now .. i have a suspicion that the database.db might have been corrupted or need tweaking otherwise; maybe it would be a good idea to open up for this, @Koenkk ?

Anyway, keep up the excellent work of this zigbee2mqtt tool - I have really come to love it - even with the faced obstacles once in a while :-)

Koenkk, 2020-02-26:
Could someone sniff the traffic of the failed LQI scans? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

Wuppi68, 2020-02-26:
Sorry can’t sniff :-( Only one adapter avail

netweaver1970, 2020-02-28:
Same here ... Multiple devices. I have about 15 devices now in the house, 2 floors, mix of Ikea bulb/switches/sockets/PIR and Xiaomi temp/humidity sensors. 

Coordinator : CC26X2R1 with the Nov 2019 Firmware

I'm on latest version of Z2M 11.0, I can't sniff as I'm only seldom home. Most of my work is remote.

(Potentially) related error logs:
zigbee2mqtt:error 2020-02-28 07:19:55: Publish 'set' 'state' to 'IKEA_Bulb_hall1' failed: 'Error: Command 0x680ae2fffe3cebef/1 genOnOff.off({}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'MAC channel access failure' (225))'
zigbee2mqtt:error 2020-02-28 08:00:11: Failed to execute LQI for 'IKEA_Router_d22'
zigbee2mqtt:error 2020-02-28 08:00:21: Failed to execute LQI for 'IKEA_Bulb_nighthall_781'
zigbee2mqtt:error 2020-02-28 08:00:39: Failed to execute LQI for 'IKEA_Bulb_hall2'
zigbee2mqtt:error 2020-02-28 08:00:49: Failed to execute LQI for 'IKEA_Bulb_hall1'
zigbee2mqtt:error 2020-02-28 08:00:59: Failed to execute LQI for 'IKEA Color 600lm BF'
zigbee2mqtt:error 2020-02-28 08:01:09: Failed to execute LQI for 'IKEA_Bulb_800lm_41'
zigbee2mqtt:error 2020-02-28 08:01:19: Failed to execute LQI for 'IKEA_Router_547'
zigbee2mqtt:error 2020-02-28 08:01:29: Failed to execute LQI for 'IKEA_Socket_e15'

The 2 below bulbs are not powered on right now so it's kind of normal they don't 'behave'. 
  zigbee2mqtt:error 2020-02-28 08:00:59: Failed to execute LQI for 'IKEA Color 600lm BF'.
  zigbee2mqtt:error 2020-02-28 08:01:09: Failed to execute LQI for 'IKEA_Bulb_800lm_41'

But all other devices are powered !
The Xiaomi devices are all ok, just the IKEA stuff is acting up.

Seems to be happening more often.

Also my network map seems to have dropped the texts/labels/colors recently but that might be a Z2M Assistant issue. Just mentioning it here to give extra context.

[image]

Keep up the good work, the product is getting better and better every release.

Thanks.

hormic77, 2020-02-28:
I also have the issue.

I use HomeAssitant in a docker Environment.
My coordinator ist CC2231.
I switched from CC2531_DEFAULT_20190608 to CC2531_SOURCE_ROUTING_20190619 to solve the issue. But not luck with it.

Also I updated Zigbee2mqtt to the Version 1.11.0. The Error ist the same.

zigbee2mqtt:info  2020-02-29 00:40:00: Starting network scan (includeRoutes 'false')
zigbee2mqtt:error 2020-02-29 00:40:10: Failed to execute LQI for 'zahnbuersten'
zigbee2mqtt:error 2020-02-29 00:40:20: Failed to execute LQI for 'zigbee_innr_temp'
zigbee2mqtt:error 2020-02-29 00:40:30: Failed to execute LQI for 'wz_virtine_rgbw'
zigbee2mqtt:error 2020-02-29 00:40:40: Failed to execute LQI for 'temp_rgbw'
zigbee2mqtt:error 2020-02-29 00:40:50: Failed to execute LQI for 'tabletswitch_eg'

The affected devices are not usable anymore.

I hope someone find the root cause for the issue soon.

I also have only one coordinator. So I'm not able to create the Wireshark logs... :-(

Thanks

slashbv, 2020-02-29:
Hi,
  I have the same issue. After I restart the bulbs and router everything works fine for a while and then I am back to the problem:  Failed to execute LQI for ....

slashbv, 2020-02-29:
I have CC2231 as coordinator and CC2530 + CC2591 as router. I have updated Z2M, updated osram bulbs firmware and changed CC2530 + CC2591 from coordinator to router. It worked fine before but becasue of many changes I don't know what exactly causes the problem

Koenkk, 2020-02-29:
For those having this issue, first **update to zigbee2mqtt 1.11.0** and please replace the contents of `/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js` with the contents of https://gist.github.com/Koenkk/10f5febae43f9f5151979279cb60f6ef.

The scan will take longer now, but we'll fix this later. Let me know if it improves things.

xefil, 2020-02-29:
Hello @Koenkk ,
Is there something for Docker already available? In case in devel branch would be good as well!
Thanks!
Simon

slashbv, 2020-02-29:
for me it doesn't seem to work, after I restart everything all is fine for about 30 minutes

Arjenlodder, 2020-03-01:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/4272454/log.txt)
[networkmap.txt](https://github.com/Koenkk/zigbee2mqtt/files/4272455/networkmap.txt) (change .txt to .pcapng)

I have the same issue and have provided logging + sniffer info (pcapng), only 1 router (of my 3 routers) does not return the error.

Relevant info:
**Coordinator**: CC26X2R1
**Router 1**: CC2530
**Router 2**: CC2530
**Router 3**: CC2530-CC2591
**Zigbee2MQTT version**: latest (1.11.0)
**Extra info**: innr_rs128t_light_1 till _5 and xiaomi_socket_1 are unplugged

I hope the attached files can provide any direction, I have no idea how to properly read the sniffer file (any info on how to translate source/destination to a device id?).

Koenkk, 2020-03-02:
@Arjenlodder thanks, is this with the patch from https://github.com/Koenkk/zigbee2mqtt/issues/2802#issuecomment-592951223 ?

Arjenlodder, 2020-03-02:
Yes, it is. I don't have logging without the patch. (P.s.: I stopped Zigbee2MQTT, replaced the .js file and then started Zigbee2MQTT).

slashbv, 2020-03-02:
I have deleted the database and also I put the source routing firmware. seems to be stable for the moment ... 24 hours

xefil, 2020-03-02:
Hello @slashbv !
Where do you have put the source routing firmware?
I've a CC2530+CC2591 (via ESP8266) as coordinator and two CC2531 as Router. All with default firmware. Wich should I try to set to source routing?
@Koenkk suggestions as well?
I'm on docker with latest dev image.
Thanks, Simon

slashbv, 2020-03-02:
I have put source routing on CC2531. I am using CC2530+CC2591 as router. I am on docker also but with stable image.

xefil, 2020-03-03:
Hello @slashbv !
Well, I've seen yes that the `source routing` applies only on the Coordinator. Ok, but the question is: my Coordinator is the only I get always succesfully LQI infos. It's failing on the two routers and on the Ikea Tradfi lamp. It's suggested to move to the `source routing` anyway? Remembering I've a CC2530+CC2591 (via ESP8266) as coordinator.
Thanks all!
Simon

slashbv, 2020-03-03:
for me was failing for everything connected to the router

Koenkk, 2020-03-03:
@Arjenlodder it seems that only the devices connected through the CC2530 router 2 fail. @ptvoinfo do you know where this issue could come from? It looks like a routing issue in the CC2530 firmware.

ptvoinfo, 2020-03-03:
@Koenkk I don't use the 'source routing' coordinator firmware. Therefore this problem is unknown to me. Is it necessary to enable any compiler directives for a source routing? I know, a router response to LQI messages from a coordinator to build a network map.

ptvoinfo, 2020-03-03:
I'm not sure about the version on GitHub. The link to the latest version:
http://ptvo.info/wp-content/uploads/2018/09/cc2530_1.2.2a.44539_firmware.zip

ptvoinfo, 2020-03-03:
Link to the latest version (for all chips)
http://ptvo.info/wp-content/uploads/2018/09/cc2530_1.2.2a.44539_firmware.zip

slashbv, 2020-03-03:
the problem is not happening with source routing

xefil, 2020-03-03:
> Link to the latest version (for all chips)
> http://ptvo.info/wp-content/uploads/2018/09/cc2530_1.2.2a.44539_firmware.zip

Hi @ptvoinfo is this firmware for a Coordinator on CC2530+CC2591 (connected to zigbee2mqtt via ESP8266) ?
Does it make sense try to change to this version for the above issue, related only for LQI fails on my two CC2531 routers and one Ikea Lamp? The actual corrdinator answers on LQI calls.

Thanks

Arjenlodder, 2020-03-03:
@Koenkk Can I further contribute to this issue? Should I install the Router software that **ptvoinfo** has submitted and make another sniffer-dump? If so, the _std_ version, or the _diag_ version? And only the router_2, or all the routers?

slashbv, 2020-03-03:
I have another problem now, even though the link seems to be good the bulbs react very show few seconds or they don't react at all, one of them even showed unavailable. one of the bulbs even appeared unavailable in HA

thehijacker, 2020-03-03:
I've had availability_timeout: 60 and report: true enabled. Now they are disabled and so far got no error. Will check again in morning.

1igeza, 2020-03-04:
I changed coordinator (another CC2530) to source routing (CC2530_SOURCE_ROUTING_20190619.zip). 
It didn't helped. With latest dev: 
Mar 04 07:41:32 zigbee2mqtt npm[3181]: zigbee2mqtt:error 2020-03-04 07:41:32: Failed to execute LQI for 'Router Goscinny'
also with changed  zStackAdapter.js Failed to execute LQI.
Do you need more sniffer traffic or another logs?

Koenkk, 2020-03-04:
So for those having this problem, does this always happen to the devices that are connected to a CC2530 router?

xefil, 2020-03-04:
Hi @Koenkk , good morning all,
Maybe. In my scenario it is so. My setup:

- zigbee2mqtt on NAS (docker) version 1.11.0-dev (commit #68750fa)
- Coordinator: CC2530+CC2591 via ESP8266 (EspEasy serial to tcp), firmware `CC2530_CC2591_DEFAULT_20190608` (**not** source routing)
- Router 2x CC2531, firmware `CC2531_router_2019_02`
- Router 1x Tradfri bulb

I'm having this issue.
Only sometimes I was able to get LQI answers from all deviced, but 99% of the time, no way.

This evening I could sniff some data. I've an extra CC2531 I could use. Let me know.

Bye, Simon

slashbv, 2020-03-04:
good morning,
 my coordinator is CC2531, router CC2530+CC2591 and 2 osram bulbs as routers. Since I have put source routing the error message is almost gone, but sometimes the bulbs react very slow or don't react at all. I am controlling them with 2 ikea remotes. It worked fine for about 24 hours after writing the coordinator with source routing.

thehijacker, 2020-03-04:
I found out something interesting, I get LQI error when network scan starts (every full hour)

`info  2020-03-03 21:00:00: Starting network scan (includeRoutes 'false')
info  2020-03-03 21:00:01: MQTT publish: topic 'zigbee2mqtt/Bulb Bedroom', payload '{"state":"OFF","linkquality":13,"brightness":252,"color_temp":370,"color":{"x":0.38
05,"y":0.3769},"update_available":false}'
info  2020-03-03 21:00:01: MQTT publish: topic 'zigbee2mqtt/Bulb Shramba', payload '{"state":"OFF","linkquality":101,"brightness":254,"color_temp":250,"color":{"x":0.3
805,"y":0.3769},"update_available":false}'
info  2020-03-03 21:00:01: MQTT publish: topic 'zigbee2mqtt/Bulb Shramba', payload '{"state":"OFF","linkquality":101,"brightness":254,"color_temp":250,"color":{"x":0.3
805,"y":0.3769},"update_available":false}'
info  2020-03-03 21:00:04: MQTT publish: topic 'zigbee2mqtt/Bulb Bedroom', payload '{"state":"OFF","linkquality":13,"brightness":252,"color_temp":370,"color":{"x":0.38
05,"y":0.3769},"update_available":false}'
info  2020-03-03 21:00:06: MQTT publish: topic 'zigbee2mqtt/Bulb Upstairs', payload '{"state":"OFF","linkquality":26,"brightness":254,"color_temp":241,"color":{"x":0.3
805,"y":0.3769},"update_available":false}'
error 2020-03-03 21:00:14: Failed to execute LQI for 'Bulb Hallway'
error 2020-03-03 21:00:26: Failed to execute LQI for 'Bulb Nightlamp Andrej'
info  2020-03-03 21:00:26: Network scan finished
info  2020-03-03 21:00:26: MQTT publish: topic 'zigbee2mqtt/bridge/networkmap/graphviz', payload 'digraph G {
node[shape=record];`

Maybe seperated error:

`error 2020-03-03 22:34:29: Publish 'set' 'state' to 'Bulb Nightlamp Simona' failed: 'Error: Command 0x00158d0004ae9d02/1 genOnOff.on({}, {"timeout":6000,"manufacturerC
ode":null,"disableDefaultResponse":false}) failed (Error: Timeout - 58185 - 1 - 238 - 6 - 11 after 6000ms)'
info  2020-03-03 22:34:29: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'Bulb Nightlamp Si
mona' failed: 'Error: Command 0x00158d0004ae9d02/1 genOnOff.on({}, {\"timeout\":6000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Timeou
t - 58185 - 1 - 238 - 6 - 11 after 6000ms)'","meta":{"friendly_name":"Bulb Nightlamp Simona"}}'`

`info  2020-03-04 01:00:00: Starting network scan (includeRoutes 'false')
info  2020-03-04 01:00:10: MQTT publish: topic 'zigbee2mqtt/Bulb Shramba', payload '{"state":"OFF","linkquality":101,"brightness":254,"color_temp":250,"color":{"x":0.3
805,"y":0.3769},"update_available":false}'
info  2020-03-04 01:00:10: MQTT publish: topic 'zigbee2mqtt/Bulb Shramba', payload '{"state":"OFF","linkquality":101,"brightness":254,"color_temp":250,"color":{"x":0.3
805,"y":0.3769},"update_available":false}'
error 2020-03-04 01:00:14: Failed to execute LQI for 'Bulb Nightlamp Simona'
info  2020-03-04 01:00:15: Network scan finished`

Could be related?

1igeza, 2020-03-05:
[logzigbee.txt](https://github.com/Koenkk/zigbee2mqtt/files/4291388/logzigbee.txt)
[log.pcapng.txt](https://github.com/Koenkk/zigbee2mqtt/files/4291389/log.pcapng.txt)
Maybe it would help. Sniffer is between Router kotlownia and coordinator.

xefil, 2020-03-06:
Hi @Koenkk and @ptvoinfo .
I've a pcap decoded I could share just in case. Let me know. (send via mail or here?)
Thanks, Simon

slashbv, 2020-03-08:
after adding report: true seems to be working fine for a few days

xefil, 2020-03-08:
Hi all,
I've tried to set CC2531 (previously router) as coordinator and CC2530+CC2591 (previously coordinator) as router. Before to set the CC2530 as router, I've removed it from the net and leaved only one CC2531 as Coordinator and another CC2531 as Router. Until here all fine, network map ok and, listen, the IKEA Tradfri works perfectly!
Then, after adding the CC2530+CC2591 as only router (firmware router-cc2530-cc2591-diag.hex - CC2530_CC2591_router_2019_02) the LQI failed for this device again.
So seems really the problem relies on the CC2530+CC2591. I would avoid thinking it's a hardware issue, because having it previously as coordinator, I was still able to get sensors informations.
But the problem is there.
I'll try now to leave all 3 devices (2x router and 1x coordinator) online and let's see how it works the next hours.
Suggestions about that?
PS: report: true added in the conf; coordinator firmware NOT source_routing.
Bye!
Simon

xefil, 2020-03-11:
Ok, some news. Now the devices that are registered behind the CC2530 are not visible on the map, maybe because the LQI is not returned by the router.
The Tradfri Lamp, until it's directly connected without passing through the CC2530, works.
Ideas?
Thanks!

Koenkk, 2020-03-11:
@ptvoinfo it seems that is really related to the CC2530 router firmware, do you have any clue?

ptvoinfo, 2020-03-12:
How can I test it? Could you someone describe a scenario? I have 4 different routers in my home. All routers report a link quality for connected devices. I see LQI for all links on the network map.

xefil, 2020-03-12:
Good morning @ptvoinfo and @Koenkk !
I'll try to recap my tests and when the LQI was available and when not.
---
**Test 1**
```
Coordinator: CC2530+CC2591 via ESP8266 and TCP Serial Server (CC2530_CC2591_DEFAULT_20190608)
Router: 2x CC2531 (CC2531_router_2019_02)
```
Results:
No LQI for 99% of the time and no Ikea Tradfri working. BTW in this scenario I can see the decives data, i.e. temperatures and motion detection.

---
**Test 2**
```
Coordinator: CC2530+CC2591 via ESP8266 and TCP Serial Server (CC2530_CC2591_DEFAULT_20190608 - source routing)
Router: 2x CC2531 (CC2531_router_2019_02)
```
Results:
No LQI for 99% of the time and no Ikea Tradfri working.

---
**Test 3**
```
Coordinator: CC2531 (CC2531_DEFAULT_20190608)
Router: 1x CC2531 (CC2531_router_2019_02)
note: CC2530+CC2591 removed
```
Results:
All works perfectly

---
**Test 4**
```
Coordinator: CC2531 (CC2531_DEFAULT_20190608)
Router: 1x CC2531 (CC2531_router_2019_02) AND CC2530+CC2591 (CC2530_CC2591_router_2019_02) without ESP8266, only as router.
```
Results:
LQI failed only for CC2530+CC2591. I lost LQI infos (and map infos) for devices behind the CC2530+CC2591.

So actually I'm on test4. I would think it's not a hardware issue on CC2530-CC2591 because as router and coordinator it doesn't stop to pass informations about values (most of the time), only for LQI. I don't know if LQI data needs more power and I've power issues, but I don't think so, because I've changed 3 times the power supplier.
I should have a pcap sniff about scenario 1, just in case.

Hope it helps, cheers!

Simon

ptvoinfo, 2020-03-12:
@xefil My thoughts:
1. LQI request to a router may fail (for any reason). Then we do not get LQI information for connected devices. But after some time the same request can be completed successfully. 
2. LQI information on the router may outdate if a device does not send anything for a long time.

xefil, 2020-03-12:
Okie, but have no idea why it's not working almost always, really seems worked for really some times. and having the cc2530 near to the coordinator didn't help. how is your setup? are you using the cc2530-cc2591? if yes in which setup (standalone, connected via serial/sub, connected via tcp) and which firmware?
And most of all, how to debug? only via pcap? Which I have and could share...
I would try to have a setup more similar as possible.
ty, Simon

xefil, 2020-03-17:
Hi there, any news?
In addition I've two questions:
1) If a router is NOT able to send LQI info (failed to execute LQI), it's possible it's out of range, even it's only 5-6meters far away from the coordinator? (two CC2531 with external antenna). Possible??
2) if a router is NOT able to send LQI info (failed to execute LQI), does a CHILD item send succesfully informations about it's state (contact/temperature/...) anyway if registered through the affected router?
Thanks,

user514, 2020-03-19:
I fall here because I have purged the HA database creating an automation for it, and right after finishing, I get all zigbee devices disconnected. And when checking the zigbee2mqtt found the error in red: "zigbee2mqtt:error 2020-03-19 02:01:39: Failed to execute LQI for 'cc2530'". Restart HA doesn't help, now I don't see any device in my networkmap page. I'm on zigbee2mqtt 1.11.0.

xefil, 2020-03-19:
Good you've solved it @user514 ...
For me always the same. Here I am with last tests:
- last zigbee2mqtt-dev on docker
- reset database
- starting with CC2530+CC2591 as coordinator (CC2530_CC2591_DEFAULT_20190608 - no source routing) this time connected via ser2net on wired network to a raspberry.
The program starts correctly, but trying to pair a router (CC2531) I'm getting:

```
info  2020-03-19 13:04:36: Device '0x00124b000be87c6a' joined
info  2020-03-19 13:04:36: MQTT publish: topic 'XefilHome/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x00124b000be87c6a"}}'
info  2020-03-19 13:04:36: Starting interview of '0x00124b000be87c6a'
info  2020-03-19 13:04:36: MQTT publish: topic 'XefilHome/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00124b000be87c6a"}}'
debug 2020-03-19 13:04:36: Device '0x00124b000be87c6a' announced itself
(...)
error 2020-03-19 13:04:57: Failed to interview '0x00124b000be87c6a', device has not successfully been paired
(...)
debug 2020-03-19 14:07:36: Received Zigbee message from '0x00124b000be87c6a', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"0/0x00124B001B7D6B1D","inactiveText":"COORD","presentValue":34,"relinquishDefault":0,"minimumOffTime":0}' from endpoint 8 with groupID 0
(...)
```

And of course no LQI.

So seems really related to CC2530+CC2591. I hope it's not a HW issue, BTW, again, I would not get any data, but seems that some data is flowing, looking on the COORD data.

@Koenkk  or @ptvoinfo which setup and firmware version do you have with those devices?

Thanks!

Simon

rajeevan, 2020-03-21:
I would like to share my experience with this erro `Failed to execute LQI for`.

Previously; I have homeassitant & zigbee2mqtt setup separatly in Debian box (Using Python env & custom installation). I didn't have any issues with this set-up.

Yesterday; I Re-installed the box and deviced to install hass.io as supervisord (using install script). This enables me to install zigbee2mqtt via hass-add-on. Which is great and convenient. Setup went good and then I started to see this erro rin zigbee2mqtt log 😢 But Restarting the box seems to have resolved the issue 😕 

I read all the post in this thread, not much of a solution or idnication of the casue. But tonight. When I was adding more automation, I saw that everytime I Restart the `Home assistant` to load the new automation modification; I get this error in the zigbee2mqtt log 🆗. So I decided to restart the `addon` zigbee2mqtt to see if that helps. And guess what. Error is no more...

I try again to restart the home assitant and the error was there and after restarting home assitant and Restarting the zigbee2mqtt addon fixes the issue.

Maybe this helps some of you!

My Setup:
- Home assitant in Debian 10 using hassio Supervisord script
- MariaDB via addon
- Zigbee2Mqtt via addon
- mosquitto broker via addon

xefil, 2020-03-22:
Hello @rajeevan !
Thanks for sharing your experience. In my case I'm not using hass.io, btw I'm starting to think that my CC2530+CC2591 has hardware issues. I would lnly like to know how to check it.
Bye, Simon

rajeevan, 2020-03-22:
@xefil Maybe I spoke too soon 😜 I noticed this morning that error is back again!
Until I switch to the hass.io installation. I didn't have any issues.. I do have another USB stick `CC2531` that need to be flashed. I will give it a try later on

rajeevan, 2020-03-22:
Update: I replaced the usb stick with another. Error still there... I don't think it is related to hardware issues.. maybe software. Prior to update, I didn't have the 'firmware update' , this is new feature (I may have been using old zigbee2mqtt before)..

I also notice that this error only appear when I run the ZigBee network map script to generate the map. I guess it invokes some communication between devices!?

Also, 2 out of 3 of my IKEA repeaters are now showing up as not connected because of the LQI fail error. I see there is an update available for the working one.

I will continue to debug. But slowly, as I have the system working in my house for real application (security, sensors and switches)

highlandr, 2020-04-13:
Hello @Koenkk and every1 with the problem.
I have the same issues.
My network setup is the following:
Zigbee2MQTT 1.12.2-dev
1x CC2531 Coordinator ({'type': 'zStack12', 'meta': {'transportrev': 2, 'product': 0, 'majorrel': 2, 'minorrel': 6, 'maintrel': 3, 'revision': 20190608}})
1x CC2531 Router same version as coordinator
10x Xiaomi Plug Router (and whatever is connect to it physically)
1x Ikea Bulb as Router and Light
30x Sensors, Motion, Temperature, Smoke, Buttons from Xiaomi and Ikea

Since a few weeks and I use always the latest DEV version of zigbee2mqtt I have LQI scan failed.

Yesterday I did an upgrade to the latest DEV and before the upgrade the Coordinator and the CC2531 Router had all the devices connected to them, the Ikea Light and the Xiaomi Plugs had nothing, few devices couldn't get into the network, I'm thinking because both receiving Routers were maxed out.

After I restart the Zigbee2 MQTT the Coordinator, the CC2531 Router and the Ikea Bulb got choosen and every endpoint tried to use those 3, when maxed out the rest was left behind.

After the update to the latest and I only have checked this morning I got the Coordinator and a Xiaomi Plug dealing with all the devices, till they get maxed out.
I can see the other Routers connected to the Coordinator or the working Router atm which is a Xiaomi Plug but they don't get endpoints.

Here's a grep from the log, every device is a Router that is not getting devices... That's my Failed LQI.

highlander@HA:~$ tail -f -n 1000 /opt/zigbee2mqtt/data/log/2020-04-13.03-08-23/log.txt | grep "Failed to execute LQI"
error 2020-04-13 08:55:40: Failed to execute LQI for '2Gang2Light'
error 2020-04-13 08:55:51: Failed to execute LQI for 'USB_Router_0x00124b000fea6e6f'
error 2020-04-13 08:56:01: Failed to execute LQI for 'Piso1_Servidor'
error 2020-04-13 08:56:15: Failed to execute LQI for 'Ikea_Piso0Suite_Light'
error 2020-04-13 08:56:25: Failed to execute LQI for 'Piso0_Sala_Candeeiro'
error 2020-04-13 08:56:35: Failed to execute LQI for 'Piso0_Sala_Globo'
error 2020-04-13 08:56:45: Failed to execute LQI for 'Piso1_Quarto_Candeeiro'
error 2020-04-13 08:56:55: Failed to execute LQI for 'Piso0_Suite_Aquecedor'
error 2020-04-13 08:57:05: Failed to execute LQI for 'Piso1_Router'
error 2020-04-13 08:57:15: Failed to execute LQI for 'Piso0_TV'
error 2020-04-13 08:57:27: Failed to execute LQI for 'Piso0_Aquecedor'
error 2020-04-13 08:57:37: Failed to execute LQI for 'Piso1_Candeeiro_PC'

@Koenkk what can I do to help figure out this problem, what to do next?
Thanks.

Regards.

[56 more replies]

closed 2020-10-04 by github-actions[bot]