Koenkk/zigbee2mqtt #2803: Tradfri LED1624G9 with output: attribute

jmt-cash, 2020-01-22:
Hello,
I run zigbee2mqtt on a raspberry Pi 4 in a docker container with output: attribute. So far everything works well. Yesterday I registered 2 Tradfri LED1624G9. The show the following topics:
- zigbee2mqtt/LED1624G/state
- zigbee2mqtt/LED1624G/brightness
- zigbee2mqtt/LED1624G/linkquality
- zigbee2mqtt/LED1624G/color-x
- zigbee2mqtt/LED1624G/color-y

I can set state and brightness by inserting set in the topic:
- zigbee2mqtt/LED1624G/set/state
- zigbee2mqtt/LED1624G/set/brightness

But how do I set the color of the Bulb? What are the topics and what are the values?

## Debug Info
zigbee2mqtt version: 1.8.0
CC253X firmware version: unknown

Yours Ilu

Koenkk, 2020-01-22:
You can do it via RGB, e.g. topic `zigbee2mqtt/LED1624G/color`, payload `#64C80A`

jmt-cash, 2020-01-23:
Thank you, I'll try this Tonight! 
I'm ashamed that I did not come up with this idea myself… :-(

BTW: You are doing a great Job with zigbee2mqtt!

jmt-cash, 2020-01-23:
Should it not be the Topic 
zigbee2mqtt/LED1624G/set/color ?

Koenkk, 2020-01-23:
Should be `zigbee2mqtt/LED1624G/set/color` indeed.

jmt-cash, 2020-01-23:
It worked, thank you!

closed 2020-01-23 by Koenkk