Koenkk/zigbee2mqtt #2805: CC1352-p2 does not find phillips hue lamp

guitar9, 2020-01-22:
zigbee2mqtt version: 1.9.0
firmware version: CC1352P_2_20191106.hex

I am running zigbee2mqtt with docker.
I switched philips hue lamp on and off but źigbee2mqtt does not find the lamp.
permit_join is in my confuguraration.yaml set to true.
Why in the last line permit_join is set to false?
This is my output:
```
zigbee2mqtt:info  2020-01-22 23:39:05: Logging to console and directory: '/app/data/log/2020-01-22.23-39-05'
zigbee2mqtt:debug 2020-01-22 23:39:05: Removing old log directory '/app/data/log/2020-01-22.23-10-57'
zigbee2mqtt:debug 2020-01-22 23:39:05: Loaded state from file /app/data/state.json
zigbee2mqtt:info  2020-01-22 23:39:05: Starting zigbee2mqtt version 1.9.0 (commit #287f843)
zigbee2mqtt:info  2020-01-22 23:39:05: Starting zigbee-herdsman...
zigbee2mqtt:debug 2020-01-22 23:39:05: Using zigbee-herdsman with settings: '{"network":{"panID":6755,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"}}'
zigbee2mqtt:info  2020-01-22 23:39:18: zigbee-herdsman started
zigbee2mqtt:info  2020-01-22 23:39:18: Coordinator firmware version: '{"type":"zStack3x0","meta":{"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20191106}}'
zigbee2mqtt:debug 2020-01-22 23:39:18: Zigbee network parameters: {"panID":6755,"extendedPanID":"0xdddddddddddddddd","channel":11}
zigbee2mqtt:info  2020-01-22 23:39:18: Currently 0 devices are joined:
zigbee2mqtt:warn  2020-01-22 23:39:18: `permit_join` set to  `true` in configuration.yaml.
zigbee2mqtt:warn  2020-01-22 23:39:18: Allowing new devices to join.
zigbee2mqtt:warn  2020-01-22 23:39:18: Set `permit_join` to `false` once you joined all devices.
zigbee2mqtt:info  2020-01-22 23:39:18: Zigbee: allowing new devices to join.
zigbee2mqtt:info  2020-01-22 23:39:18: Connecting to MQTT server at mqtt://localhost:1883
zigbee2mqtt:info  2020-01-22 23:39:18: Connected to MQTT server
zigbee2mqtt:info  2020-01-22 23:39:18: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
zigbee2mqtt:info  2020-01-22 23:39:18: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.9.0","commit":"287f843","coordinator":{"type":"zStack3x0","meta":{"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20191106}},"log_level":"debug","permit_join":false}'
```

Koenkk, 2020-01-23:
Turning the bulb on/off is not enough to reset it, try following the instructions from https://www.zigbee2mqtt.io/devices/4034031P7.html#pairing

guitar9, 2020-01-23:
I does not own a philips hue bridge. And the bulb was never connected to a philips hue bridge. It was only connected with a CC2531 which runs with an other pan_id. 
So I should reset it nevertheless?

nldroid, 2020-01-23:
Yes

guitar9, 2020-01-23:
Touchlink woked with CC1352 thank you. After first attempt with touchlink I got timeout error. But after second attempt it worked! Thank you !

closed 2020-01-23 by guitar9