Koenkk/zigbee2mqtt #2806: unit_of_measurement missing in HA

HA-TB303, 2020-01-23:
Improvement:

On all my sensors (hue, Ikea and Xiaomi) the unit_of_measurement value for linkquality (dB or % whatever is appropriate) is missing.

## Debug Info
zigbee2mqtt version: 1.9.0

[image]

It would be nice of you could add this.

nanobra1n, 2020-01-24:
Link Quality Indicator (LQI) is the quality of the real data received in a signal.It ranges from 0 to 255, with 255 being the best quality. Therefore, there is no unit_of_measurement, as shown by the "-"

HA-TB303, 2020-01-24:
Would it not still be perfectly possible to translate those values to a more consumer friendly 0-100 and make the unit of measurement ”%”

nanobra1n, 2020-01-24:
Including this functionality into zigbee2mqtt is a philosophical question and beyond my pay grade. But if you really want this now, it's as easy as creating your own template sensor

```
sensor
  - platform: template
    sensors:
      linkqual:
        entity_id: sensor.yoursensor_linkquality
        unit_of_measurement: '%'
        friendly_name: 'My sensor link'
        value_template: '{{states.sensor.yoursensor_linkquality.state| float / 255 * 100}}'
        icon_template: mdi:wifi
```
Voila

HA-TB303, 2020-01-24:
I know i could do that for my 18 sensors, but that's not what I’m asking 😏

groenmarsmannetje, 2020-01-24:
Maybe just use “lqi” as unit. Besides that I would recommend to use device_class “signal_strength” as default class for all these Linkquality sensors. Currently I have to manually customize that each time.

Koenkk, 2020-01-25:
@groenmarsmannetje done, unit will be `lqi` now and icon will be set to a signal icon now. Device class `signal_strength` cannot be used as according to the docs the unit has to be `dB` for this (https://www.home-assistant.io/integrations/sensor/#device-class)

PedroLamas, 2020-01-26:
@Koenkk the dev build got broken after these changes, I assume the tests have failed?

PedroLamas, 2020-01-26:
On a side note, I've asked on Home Assistant discord if there would be an interest in having a "lqi" device class of sorts!

Koenkk, 2020-01-26:
@PedroLamas thanks, fixed the tests.

PedroLamas, 2020-01-27:
This feels a bit unnecessary and kind of "hacky" to me...

All these sensors end on "_linkquality", so it was quite easy to configure them globally anyway with the `customize_glob.yaml` file:

```yaml
"*_linkquality":
  unit_of_measurement: 'lqi'
  icon: mdi:signal
```

You can check [here](https://github.com/PedroLamas/home-assistant-config/blob/011da3185cd3d475e43ccafec299fd18608cb91a/config/customize_glob.yaml) an example in my own configuration.

Koenkk, 2020-01-27:
@PedroLamas why do you think it's hacky?

PedroLamas, 2020-01-27:
Probably a bit of a strong word there (sorry!), but I think this should just be provided on a new device class on Home Assistant rather than having the HA extension supply this metadata.

Koenkk, 2020-01-27:
@PedroLamas yes I agree it should be in a device class, let's see if they'll add it.

groenmarsmannetje, 2020-01-27:
> This feels a bit unnecessary and kind of "hacky" to me...
>
> All these sensors end on "_linkquality", so it was quite easy to configure them globally anyway with the `customize_glob.yaml` file: [...]

Thank you for that tip. I did not know that something like that existed.

carldebilly, 2020-01-27:
Maybe it should be configureable in Z2M.

Koenkk, 2020-03-27:
I assume this can be closed, unit is not set to `lqi`.

closed 2020-03-27 by Koenkk