Koenkk/zigbee2mqtt #2807: CC26X2R1: time-out ob startup after a few days

docbobo, 2020-01-23:
# Bug Report
## What happened

Woke up this morning, seeing the following in the log:

Woke this morning to this:
zigbee2mqtt             |
zigbee2mqtt             | > zigbee2mqtt@1.9.0 start /app
zigbee2mqtt             | > node index.js
zigbee2mqtt             |
zigbee2mqtt             | zigbee2mqtt:error 2020-01-23 07:16:01: Error while starting zigbee-herdsman
zigbee2mqtt             | zigbee2mqtt:error 2020-01-23 07:16:01: Failed to start zigbee
zigbee2mqtt             | zigbee2mqtt:error 2020-01-23 07:16:01: Exiting...
zigbee2mqtt             | zigbee2mqtt:error 2020-01-23 07:16:01: Error: SRSP - SYS - version after 6000ms
zigbee2mqtt             |     at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
zigbee2mqtt             |     at listOnTimeout (internal/timers.js:531:17)
zigbee2mqtt             |     at processTimers (internal/timers.js:475:7)
zigbee2mqtt             | npm ERR! code ELIFECYCLE
zigbee2mqtt             | npm ERR! errno 1
zigbee2mqtt             | npm ERR! zigbee2mqtt@1.9.0 start: `node index.js`
zigbee2mqtt             | npm ERR! Exit status 1
zigbee2mqtt             | npm ERR!
zigbee2mqtt             | npm ERR! Failed at the zigbee2mqtt@1.9.0 start script.
zigbee2mqtt             | npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
zigbee2mqtt             |
zigbee2mqtt             | npm ERR! A complete log of this run can be found in:
zigbee2mqtt             | npm ERR!     /root/.npm/_logs/2020-01-23T06_16_01_909Z-debug.log

Has to disconnect the coordinator. Afterwards everything worked again

## What did you expect to happen
zigbee2mqtt continued to work

## How to reproduce it (minimal and precise)
?

## Debug Info
zigbee2mqtt version: 1.9.0
firmware version: 3.x.0

Koenkk, 2020-01-23:
Do you have the logging leading to the restart of zigbee2mqtt?

docbobo, 2020-01-23:
Well, this is what I got:

> {"log":"\u001b[32mzigbee2mqtt:info \u001b[39m 2020-01-23 05:29:36: MQTT publish: topic 'zigbee2mqtt/kitchen_switch_03', payload '{\"state\":\"OFF\",\"linkquality\":69,\"last_seen\":\"2020-01-23T05:29:36+01:00\",\"device\":{\"friendlyName\
> {"log":"\u001b[34mzigbee2mqtt:debug\u001b[39m 2020-01-23 05:29:36: Received Zigbee message from 'bathroom_switch_02', type 'readResponse', cluster 'genOnOff', data '{\"onOff\":0}' from endpoint 18 with groupID 0\n","stream":"stdout","time
> {"log":"\u001b[32mzigbee2mqtt:info \u001b[39m 2020-01-23 05:29:36: MQTT publish: topic 'zigbee2mqtt/bathroom_switch_02', payload '{\"state\":\"OFF\",\"linkquality\":66,\"last_seen\":\"2020-01-23T05:29:36+01:00\",\"brightness\":18,\"color_
> {"log":"\u001b[34mzigbee2mqtt:debug\u001b[39m 2020-01-23 05:29:41: Received Zigbee message from 'kitchen_switch_03', type 'readResponse', cluster 'genOnOff', data '{\"onOff\":0}' from endpoint 18 with groupID 0\n","stream":"stdout","time"
> {"log":"\u001b[32mzigbee2mqtt:info \u001b[39m 2020-01-23 05:29:41: MQTT publish: topic 'zigbee2mqtt/kitchen_switch_03', payload '{\"state\":\"OFF\",\"linkquality\":72,\"last_seen\":\"2020-01-23T05:29:41+01:00\",\"device\":{\"friendlyName\
> {"log":"\u001b[34mzigbee2mqtt:debug\u001b[39m 2020-01-23 05:29:41: Received Zigbee message from 'bathroom_switch_02', type 'readResponse', cluster 'genOnOff', data '{\"onOff\":0}' from endpoint 18 with groupID 0\n","stream":"stdout","time
> {"log":"\u001b[32mzigbee2mqtt:info \u001b[39m 2020-01-23 05:29:41: MQTT publish: topic 'zigbee2mqtt/bathroom_switch_02', payload '{\"state\":\"OFF\",\"linkquality\":63,\"last_seen\":\"2020-01-23T05:29:41+01:00\",\"brightness\":18,\"color_
> {"log":"\u001b[34mzigbee2mqtt:debug\u001b[39m 2020-01-23 05:29:46: Received Zigbee message from 'kitchen_switch_03', type 'readResponse', cluster 'genOnOff', data '{\"onOff\":0}' from endpoint 18 with groupID 0\n","stream":"stdout","time"
> {"log":"\u001b[32mzigbee2mqtt:info \u001b[39m 2020-01-23 05:29:46: MQTT publish: topic 'zigbee2mqtt/kitchen_switch_03', payload '{\"state\":\"OFF\",\"linkquality\":72,\"last_seen\":\"2020-01-23T05:29:46+01:00\",\"device\":{\"friendlyName\
> {"log":"\u001b[34mzigbee2mqtt:debug\u001b[39m 2020-01-23 05:29:46: Received Zigbee message from 'bathroom_switch_02', type 'readResponse', cluster 'genOnOff', data '{\"onOff\":0}' from endpoint 18 with groupID 0\n","stream":"stdout","time
> {"log":"\u001b[32mzigbee2mqtt:info \u001b[39m 2020-01-23 05:29:46: MQTT publish: topic 'zigbee2mqtt/bathroom_switch_02', payload '{\"state\":\"OFF\",\"linkquality\":66,\"last_seen\":\"2020-01-23T05:29:46+01:00\",\"brightness\":18,\"color_
> {"log":"(node:16) UnhandledPromiseRejectionWarning: Error: Timeout - 19237 - 18 - 81 - 6 - 1 after 10000ms\n","stream":"stderr","time":"2020-01-23T04:30:13.835567197Z"}
> {"log":"    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)\n","stream":"stderr","time":"2020-01-23T04:30:13.835611707Z"}
> {"log":"    at listOnTimeout (internal/timers.js:531:17)\n","stream":"stderr","time":"2020-01-23T04:30:13.835621901Z"}
> {"log":"    at processTimers (internal/timers.js:475:7)\n","stream":"stderr","time":"2020-01-23T04:30:13.835631011Z"}
> {"log":"(node:16) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (r
> {"log":"(node:16) UnhandledPromiseRejectionWarning: Error: AREQ - AF - dataConfirm after 10000ms\n","stream":"stderr","time":"2020-01-23T04:30:13.835676725Z"}
> {"log":"    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)\n","stream":"stderr","time":"2020-01-23T04:30:13.83568839Z"}
> {"log":"    at listOnTimeout (internal/timers.js:531:17)\n","stream":"stderr","time":"2020-01-23T04:30:13.835697738Z"}
> {"log":"    at processTimers (internal/timers.js:475:7)\n","stream":"stderr","time":"2020-01-23T04:30:13.835706655Z"}
> {"log":"(node:16) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (r
> {"log":"(node:16) UnhandledPromiseRejectionWarning: Error: Timeout - 23435 - 18 - 80 - 6 - 1 after 10000ms\n","stream":"stderr","time":"2020-01-23T04:30:19.833797387Z"}
> {"log":"    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)\n","stream":"stderr","time":"2020-01-23T04:30:19.833846155Z"}
> {"log":"    at listOnTimeout (internal/timers.js:531:17)\n","stream":"stderr","time":"2020-01-23T04:30:19.833856452Z"}
> {"log":"    at processTimers (internal/timers.js:475:7)\n","stream":"stderr","time":"2020-01-23T04:30:19.833865523Z"}
> {"log":"(node:16) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (r
> {"log":"(node:16) UnhandledPromiseRejectionWarning: Error: AREQ - AF - dataConfirm after 10000ms\n","stream":"stderr","time":"2020-01-23T04:30:19.833907586Z"}
> {"log":"    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)\n","stream":"stderr","time":"2020-01-23T04:30:19.833929841Z"}
[... cut ...]
> (node:16) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch().
> (node:16) UnhandledPromiseRejectionWarning: Error: AREQ - AF - dataConfirm after 10000ms
> (node:16) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch().

Bretos, 2020-01-23:
Hello guys!

I am currently facing the exact same issue.

The error (attached at the end) is identical when testing with:
* bare-metal 1.9.0
* bare-metal 1.8.0
* bare-metal dev branch
* bare-metal master branch

Due to the error content I though that maybe this is an environmental issue, so I tried running zigbee2mqtt in docker with:

* latest
* latest-dev
* 1.9.0
* 1.8.0

issue was reproductable with every single one of these versions. Between versions I made sure that no other z2m process is running, nor that any other process uses the serial port.

Then I though that on startup too many zigbee devices try to connect to the coordinator, so I turned off some of them and tried again. Same error.

As a friend flashed the CC2531 for me, I am not exactly sure which FW version he used, however I am sure that it was not the source routing one.

Error contents:

`> zigbee2mqtt@1.9.0-dev start /app
> node index.js

zigbee2mqtt:info  2020-01-23 18:32:55: Logging to console and directory: '/app/data/log/2020-01-23.18-32-55'
zigbee2mqtt:info  2020-01-23 18:32:56: Starting zigbee2mqtt version 1.9.0-dev (commit #145f8a5)
zigbee2mqtt:info  2020-01-23 18:32:56: Starting zigbee-herdsman...
zigbee2mqtt:error 2020-01-23 18:33:03: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-01-23 18:33:03: Failed to start zigbee
zigbee2mqtt:error 2020-01-23 18:33:03: Exiting...
zigbee2mqtt:error 2020-01-23 18:33:03: Error: SRSP - SYS - version after 6000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at listOnTimeout (internal/timers.js:531:17)
    at processTimers (internal/timers.js:475:7)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.9.0-dev start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.9.0-dev start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-01-23T17_33_03_191Z-debug.log`

Koenkk, 2020-01-24:
@docbobo looks like the CC26xR1 crashed or the communication was interrupted, let's see if it happens another time because now I don't know were to start investigating.

@Bretos as you have a CC2531 it's not related to this issue, please try flashing the source routing firmware.

harisosmancevic, 2020-03-15:
Seems like i am having the same issue, also with an CC26X2R1, running the latest version..

harisosmancevic, 2020-03-15:
Also having issues with other zigbee devices since 3 hours ago today, no changed made last couple of days:
zigbee2mqtt:error 2020-03-15 22:53:11: Publish 'set' 'state' to '0x7cb03eaa0a011664' failed: 'Error: Command 0x7cb03eaa0a011664/3 genOnOff.off({}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'MAC channel access failure' (225))'
zigbee2mqtt:info  2020-03-15 22:53:11: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to '0x7cb03eaa0a011664' failed: 'Error: Command 0x7cb03eaa0a011664/3 genOnOff.off({}, {\"timeout\":6000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Data request failed with error: 'MAC channel access failure' (225))'","meta":{"friendly_name":"0x7cb03eaa0a011664"}}'

Bretos, 2020-03-24:
Reflashing helped

closed 2020-05-30 by stale[bot]