Koenkk/zigbee2mqtt #2809: Setting brightness to 1 turns off philips bulb.

CypherMK, 2020-01-23:
# Bug Report
I updated from 1.60 to 1.90 yesterday, and without changing anything in node red I noticed this behaviour:

So when I set the brightness to 1 (using node red), my philips bulbs (https://www.zigbee2mqtt.io/devices/8718696449691.html) turn off (no more light visible), but they stay in the ON position in homeassistant.

When I do the same thing with any ikea bulb I have, they stay on (light visible) at the lowest brightness.

## What did you expect to happen
I expect the light to stay on in the lowest brightness possible

## How to reproduce it (minimal and precise)
Set brightness to 1 with a philips bulb and the light turns off.

## Debug Info
The logs say this: zigbee2mqtt/Bijkeuken', payload '{"state":"ON","brightness":1,"linkquality":52}
zigbee2mqtt version: 1.90 (commit #287f843)
CC253X firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190619}}'

TilmanK, 2020-01-23:
I can confirm this issue with a Hue Bulb (LCT015), nevertheless this is not a Z2M issue from what I understand. 

I've sent all values from 5 to 1 and it stays on until I send brightness 1 (so 2 works...). There is no difference in the Zigbee message I sniffed, this is a problem of the bulb itself that switches off a brightness 1. 

**Brightness: 2**
ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x01)
    Sequence Number: 33
    Command: Move to Level with OnOff (0x04)
    Payload
        Level: 2
        Transition Time: 0.0 seconds

**Brightness: 1**
ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x01)
    Sequence Number: 34
    Command: Move to Level with OnOff (0x04)
    Payload
        Level: 1
        Transition Time: 0.0 seconds

An easy way to fix this could be to filter on brightness and replace a 1 value with a 2.

CypherMK, 2020-01-23:
I tried it with 1.60 and same behaviour, so it is not version related.

So I guess that is how the bulb works. If I set the brightness to 2, it stays on.

But still it is weird, that the behaviour is like that. I guess it is not in line with other bulbs. I have multiple ikea bulbs and one xiaomi bulb. They all stay on with brightness 1.

Can't zigbee2mqtt adjust this? So if a user uses brightness 1, it converts it to 2? So it stays in line with the rest of the bulbs?

CypherMK, 2020-01-23:
Also, if I use brightness 1, the bulb stays on in homeassistant but light turns off. If I use brightness 0, the bulb goes to off in homeassistant and the light goes off. 
With brightness 1 I can use transition time, with brightness 0 I can't use transition time, it immediately turns off.

TilmanK, 2020-01-23:
That's the behavior intended. HomeAssistant doesn't know, that the light isn't on, since the bulb acts as it would be...

Same behavior in HomeKit with NodeRed + Z2M...

CypherMK, 2020-01-23:
One thing has changed though between 1.60 and 1.90.
When I set this payload in node red: msg.payload.state: OFF, msg.payload.brightness: 0, msg.payload.transition: 20
Then the light makes a smooth transition to full off in 20 seconds. So light off and switch to off in Homeassistant.

Now I have to do this: msg.payload.brightness: 1, msg.payload.transition: 20 and after 20 seconds I need to do:msg.payload.state: OFF; to get the off position in Homeassistant.

So it didn't improve in my opinion.

ryanbeaton, 2020-01-24:
I'm sure I remember some discussion about this and it creating a change.

Maybe this:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/633

CypherMK, 2020-01-24:
I think that change made things more complicated. Like I said 1 should be on in HA and light should be visible and 0 should be all off. 
If that is not possible, then make it possible to set a transition, when setting the brightness to 0.

TilmanK, 2020-01-24:
@ryanbeaton From what I understand this affects what a Hue dimmer can send when pressing brightness up/down buttons.

closed 2020-03-31 by stale[bot]

PrzemekSkw, 2022-02-21:
Hello, I have similar situation but on Ledvance zigbee bulbs. Everything was OK with that buls until I buy Philips Smart Hue button. I set automation and my two bulbs turn off to minimal brightness. Until that day everytime I turn off that bulbs thay are minimal brightness when turned on.
Regards.