Koenkk/zigbee2mqtt #2811: Sercomm SZ-ESW01

coolguymatt, 2020-01-23:
# Bug Report
## What happened
After upgrading to Zigbee2MQTT to 1.9 in hassio, both of my Sercomm SZ-ESW01 plugs started reporting energy readings for the first time (did not occur in prior versions). After a period of time being on, both stop responding to on/off commands. In the log below, you will see an energy reading every few seconds while it is on. Then you can see that it fails to turn off, but continues to report power readings. It would not respond to on/off commands until I repaired it, and then started to display the same behaviour again a few hours later. I do not know if it is related, but several other of my devices also started to display odd behaviour (not responding mostly) and were fixed by repairing those devices. Currently both my Sercomm plugs are unplugged and the network is back to behaving properly. 

info  2020-01-22 09:00:50: MQTT publish: topic 'zigbee2mqtt/0x000d6f0004f3ba9f', payload '{"state":"ON","linkquality":26,"last_seen":1575450948563,"power":4.35}'
info  2020-01-22 09:00:53: MQTT publish: topic 'zigbee2mqtt/0x000d6f0004f3ba9f', payload '{"state":"ON","linkquality":28,"last_seen":1575450948563,"power":4.4}'
info  2020-01-22 09:00:54: MQTT publish: topic 'zigbee2mqtt/0x000d6f0004f3ba9f', payload '{"state":"ON","linkquality":26,"last_seen":1575450948563,"power":4.35}'
error 2020-01-22 09:00:55: Publish 'set' 'state' to '0x000d6f0004f3ba9f' failed: 'Error: Data request failed with error: 'No network route' (205)'
info  2020-01-22 09:00:55: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to '0x000d6f0004f3ba9f' failed: 'Error: Data request failed with error: 'No network route' (205)'","meta":{"friendly_name":"0x000d6f0004f3ba9f"}}'
info  2020-01-22 09:00:57: MQTT publish: topic 'zigbee2mqtt/0x000d6f0004f3ba9f', payload '{"state":"ON","linkquality":28,"last_seen":1575450948563,"power":4.4}'
info  2020-01-22 09:00:58: MQTT publish: topic 'zigbee2mqtt/0x000d6f0004f3ba9f', payload '{"state":"ON","linkquality":28,"last_seen":1575450948563,"power":4.35}'

## What did you expect to happen
Devices to continue to operate properly when power readings occur.

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: 1.9
CC253X firmware version: CC2531_DEFAULT_20190608.zip

Koenkk, 2020-01-24:
Can you sniff the traffic while the device is working and when it stopped working? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

coolguymatt, 2020-01-30:
I bought a second CC2531 so once it arrives I will get this information. Is it possible that too many messages are being sent and things are being dropped? I find when my other power monitoring plugs are on, my network is less stable (plugs and bulbs take 2-3 attempts to turn on). These also didn't report energy until 1.9 so it is a new issue. I got a CC26X2R1 thinking that a more powerful device would help but it still happens.

closed 2020-04-07 by stale[bot]