Koenkk/zigbee2mqtt #2812: aqara opple command bug

lychar460, 2020-01-23:
Good evening @Koenkk , 
I have a problem with one of aqara opple 2 bands, that was functional for several days, but since yesterday nothing.  I'm changing the battery to start with, that's not it.  I delete it from my configuration, relaunch the pairing, during the pairing the commands are recognized, but once pairing successfull, the commands are no longer recognized, they do not appear in the log?  an idea? 
For information, I have 2 others that work perfectly.

Cordially

zigbee2mqtt version:1.8.0
CC253X firmware version: CC2531_DEFAULT_20190608

rduivenvoorde, 2020-01-24:
Not sure if this is related, but I just paired a NEW Aqara Opple 2 Bands device succesfully, but with button 1 and 2 only single clicks are sent over mqtt, while button 3 and 4 do both single AND double clicks.

Not sure if this is malfunctioning or by design? @lychar460 can you maybe check with your working devices?

Or should I create a separate issue for this? 
When enabling debug I see that button 1 and 2 do NOT have any data in the zigbee messages.
Below I enabled debug and clicked on all buttons first single and then double:
```
zigbee2mqtt:debug 2020-01-24 15:10:29: Received Zigbee message from 'dubbelknop', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-01-24 15:10:29: MQTT publish: topic 'riouw/zbee/dubbelknop', payload '{"battery":"100.00","voltage":3100,"linkquality":70,"action":"button_1_single"}'
zigbee2mqtt:debug 2020-01-24 15:10:32: Received Zigbee message from 'dubbelknop', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-01-24 15:10:32: MQTT publish: topic 'riouw/zbee/dubbelknop', payload '{"battery":"100.00","voltage":3100,"linkquality":78,"action":"button_1_single"}'
zigbee2mqtt:debug 2020-01-24 15:10:34: Received Zigbee message from 'dubbelknop', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-01-24 15:10:34: MQTT publish: topic 'riouw/zbee/dubbelknop', payload '{"battery":"100.00","voltage":3100,"linkquality":94,"action":"button_2_single"}'
zigbee2mqtt:debug 2020-01-24 15:10:36: Received Zigbee message from 'dubbelknop', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-01-24 15:10:36: MQTT publish: topic 'riouw/zbee/dubbelknop', payload '{"battery":"100.00","voltage":3100,"linkquality":89,"action":"button_2_single"}'
zigbee2mqtt:debug 2020-01-24 15:10:39: Received Zigbee message from 'dubbelknop', type 'commandStep', cluster 'genLevelCtrl', data '{"stepmode":1,"stepsize":85,"transtime":7}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-01-24 15:10:39: MQTT publish: topic 'riouw/zbee/dubbelknop', payload '{"battery":"100.00","voltage":3100,"linkquality":102,"action":"button_3_single"}'
zigbee2mqtt:debug 2020-01-24 15:10:41: Received Zigbee message from 'dubbelknop', type 'commandStepColorTemp', cluster 'lightingColorCtrl', data '{"stepmode":1,"stepsize":69,"transtime":7,"minimum":0,"maximum":0}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-01-24 15:10:41: MQTT publish: topic 'riouw/zbee/dubbelknop', payload '{"battery":"100.00","voltage":3100,"linkquality":89,"action":"button_3_double"}'
zigbee2mqtt:debug 2020-01-24 15:10:42: Received Zigbee message from 'dubbelknop', type 'commandStep', cluster 'genLevelCtrl', data '{"stepmode":0,"stepsize":85,"transtime":7}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-01-24 15:10:42: MQTT publish: topic 'riouw/zbee/dubbelknop', payload '{"battery":"100.00","voltage":3100,"linkquality":94,"action":"button_4_single"}'
zigbee2mqtt:debug 2020-01-24 15:10:44: Received Zigbee message from 'dubbelknop', type 'commandStepColorTemp', cluster 'lightingColorCtrl', data '{"stepmode":3,"stepsize":69,"transtime":7,"minimum":0,"maximum":0}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-01-24 15:10:44: MQTT publish: topic 'riouw/zbee/dubbelknop', payload '{"battery":"100.00","voltage":3100,"linkquality":94,"action":"button_4_double"}'

```

lychar460, 2020-01-24:
Hello, 
this is my case too but it is not at all the same problem. Me, after pairing, the commands are no longer recognized.

sjorge, 2020-01-24:
New firmware version maybe?

rduivenvoorde, 2020-01-24:
@sjorge do you mean you think I have to upgrade?
I see this from info:
Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190425}}'

sjorge, 2020-01-24:
I meant on the Aqaba switch, it’s a relatively new device and if the behavior changed since implemented. New unit might have a newer firmware.

~ sjorge

lychar460, 2020-01-25:
Hello, 
after 3 hours of new pairing, it is recognized, by cons orders are read 4 times.
Jan 25 10:27:04 raspberrypi3b npm[557]: zigbee2mqtt:info  2020-01-25 10:27:04: MQTT publish: topic 'zigbee2mqtt/Aqara OPPLE quadruple N2', payload '{"linkquality":99,"action":"button_3_single"}'
Jan 25 10:27:04 raspberrypi3b npm[557]: zigbee2mqtt:info  2020-01-25 10:27:04: MQTT publish: topic 'zigbee2mqtt/Aqara OPPLE quadruple N2', payload '{"linkquality":99,"action":"button_3_single"}'
Jan 25 10:27:04 raspberrypi3b npm[557]: zigbee2mqtt:info  2020-01-25 10:27:04: MQTT publish: topic 'zigbee2mqtt/Aqara OPPLE quadruple N2', payload '{"linkquality":99,"action":"button_3_single"}'
Jan 25 10:27:04 raspberrypi3b npm[557]: zigbee2mqtt:info  2020-01-25 10:27:04: MQTT publish: topic 'zigbee2mqtt/Aqara OPPLE quadruple N2', payload '{"linkquality":99,"action":"button_3_single"}'
Jan 25 10:27:11 raspberrypi3b npm[557]: zigbee2mqtt:info  2020-01-25 10:27:11: MQTT publish: topic 'zigbee2mqtt/Aqara OPPLE quadruple N2', payload '{"linkquality":84,"action":"button_3_double"}'
Jan 25 10:27:11 raspberrypi3b npm[557]: zigbee2mqtt:info  2020-01-25 10:27:11: MQTT publish: topic 'zigbee2mqtt/Aqara OPPLE quadruple N2', payload '{"linkquality":84,"action":"button_3_double"}'
Jan 25 10:27:11 raspberrypi3b npm[557]: zigbee2mqtt:info  2020-01-25 10:27:11: MQTT publish: topic 'zigbee2mqtt/Aqara OPPLE quadruple N2', payload '{"linkquality":84,"action":"button_3_double"}'
Jan 25 10:27:11 raspberrypi3b npm[557]: zigbee2mqtt:info  2020-01-25 10:27:11: MQTT publish: topic 'zigbee2mqtt/Aqara OPPLE quadruple N2', payload '{"linkquality":84,"action":"button_3_double"}'

closed 2020-05-17 by stale[bot]