Koenkk/zigbee2mqtt #2813: Random 'No network route' (205)' on bulb

mr2c12, 2020-01-24:
My actual configuration is:
- coordinator fw 3.0, cc2530+cc2591
- router cc2530
- 2 bulbs (1 ikea, 1 philips hue)
- several end devices

Sometimes, the philips bulb lose the connection and every attempt to send a command return: 'No network route' (205)'
The bulb is close to the other bulb. Then it comes the router and finally the coordinator.
When printing the network map, the bulb is connected to the three of them.

The issue happens quite random. In a few minutes, the bulb is back on.
I even replace the problematic bulb with another one (a new ikea one) but I get the same 
Very close to the bulb, I have installed an Ikea Motion sensor which seems to never have a connectivity problem.

What is my best option now? Should I try to add another router?
Switching the coordinator to "source routing" version could be of any help? And, in this case, should I re-pair everything?

In my current setup, this bulb is the only problematic device I have. Any other device is working just fine.

dzungpv, 2020-01-24:
@mr2c12 I faced the same problem.
Your problem is you are running Zigbee 3.0 with Zigbee 1.2 router. With Zigbee 3.0 it have new Child Aging feature, but  @Koenkk  have been disable it because some xiaomi devices.
Your solution: 
1, If all your bulb and sensor are End Devices: build your own firmware with Child Aging enable, build your own Router firmware 3.0, same with Coordinator.
2, If Ikea and Phillips running Zigbee 1.2, and they are router, you can seperate it to difference network.

mr2c12, 2020-01-24:
Switch ing my coordinator to fw 1.2 with source routing is an option?
As I can see in the faq, in that case I would need to repair every device. I would like to avoid this if possibile.

mr2c12, 2020-01-29:
I spent some time in the weekend to switch to 1.2 without source routing, since at the moment I have ~10 devices.
From that moment, the problem never appeared again.

My bad, I chose the wrong firmware (3.0) for my adapter.

closed 2020-01-29 by mr2c12