Koenkk/zigbee2mqtt #2814: lag / delay, NOT related to mqtt (see post)

prankousky, 2020-01-24:
Hi everybody,

I am running `1.9.0` `
{'type': 'zStack3x0', 'meta': {'transportrev': 2, 'product': 1, 'majorrel': 2, 'minorrel': 7, 'maintrel': 1, 'revision': 20191106}}` on a `Texas Instruments CC1352P-2`.

Recently, **all** my light entities, and **most of** my climate entities have either *a lot* of delay, or *will not work at all*.

I use the `Eurotronic SPZB0001` zigbee thermostat as climate entities; my lights are some `innr` bulbs, some `philips hue`, a `philips aurora`, and some `aqara smart bulbs`. All of these devices are supported by `zigbee2mqtt` according to the official website.

zigbee2mqtt is connected to `Home Assistant` (version `0.104.3`) via `mqtt`. What really goes over my head is the fact that most sensors work immediately; let's say I open the office door... the entity in Home Assistant will turn to `open` right away! However, when I try to switch on my office light, nothing will happen. Home Assistant will display it as `on` after I switched it on, but then, after a few seconds, display it as `off` again. It ***might*** turn on eventually, or it might now. If it does, Home Assistant will display its state accordingly once this happened. 

However, below is my mqtt output. Unfortunately there is no timestamp, so I manually inserted estimated time values.

```
❯ msub zigbee2mqtt/Philips_Hue_01/#

zigbee2mqtt/Philips_Hue_01/availability online
zigbee2mqtt/Philips_Hue_01/set {"state": "ON"} --- 16:42:10
zigbee2mqtt/Philips_Hue_01/set {"state": "ON"} --- 16:42:50 (tried again)
zigbee2mqtt/Philips_Hue_01/set {"state": "OFF"} --- 16:42:52 (manually turned it back off)
zigbee2mqtt/Philips_Hue_01/set {"state": "ON"} --- 16:43:10
zigbee2mqtt/Philips_Hue_01/set {"state": "ON"} --- 16:45:00
zigbee2mqtt/Philips_Hue_01/availability online --- 16:49:30 (first time this came up since I subscribed to the lamp's topic)
```

The light is currently still off. This is usually the light that responds the fastest, so I used that one as an example. Of course now that I want to demo, it does not work at all. Even after it showing its availability as `online`, I am not able to control it!!

However, I can say that most lights work like this. The will eventually react to commands (not just on/off,  but also color/brightness changes), but this might happen between 20 seconds and 10 minutes (just to name a value; it can be multiple minutes...).

What could be the cause of this? Until recently, I did not have this issue. I recently updated both `zigbee2mqtt` as well as `Home Assistant`, but I am not sure this has anything to do with it... when I change anything in Home Assistant, it will display **right away** in the mqtt log, so it is being transmitted *to* zigbee2mqtt immediately. Why does it take zigbee2mqtt so long for the lights and climate entities to work? Sensor values will *usually* be transmitted as soon as they change. I am not sure for all sensors atm, but I am positive for door/window sensors, as I often test them when something is wrong. 

So communication between z2m and hass seems to be immediate. But still, lights and climate entities work very inconsistently. 

This is my configuration. I changed the values for `queue` as well as completely commented them out for testing (and always restarted afterwards)

```
homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://url:1883'
  user: very_mqtt
  password: much_secret
  client_id: z2mbeelink
  include_device_information: true
serial:
  port: >-
    /dev/serial/by-id/usb-Texas_Instruments_XDS110__02.03.00.18__Embed_with_CMSIS-DAP_L430019B-if00
  disable_led: false
advanced:
  pan_id: my pan id
  network_key:
   - my network key
  homeassistant_discovery_topic: homeassistant
  homeassistant_status_topic: hass/status
  last_seen: ISO_8601_local
  elapsed: true
  report: true
  log_level: debug
  availability_timeout: 30
  cache_state: true
queue:
  delay: 350 #previously 250
  simultaneously: 100 # previously 15
ban:
  - '0x14b457fffe779019'
devices:
```
That did not make a difference at all.

# Really weird

When I restart the entire server running zigbee2mqtt, everything will work fine for a couple of minutes. When I switch a climate entity or light entity, it will work as expected: change to the newly set state right away. But after a few minutes, the lag is there again.

I have completely deactivated the `Home Assistant zigbee2mqtt map` addon, as I have read that it causes a lot of traffic. I have also connected my coordinator via USB extender cable to my server; it sits **outside** of my mini-rack, not inside. The antenna is around 2 meters away from the controller itself, as it's cable offered this much range. It has also worked like this just fine until recently.

As I have many automations that work with these light entities, I would like to get back to the state where there is no lag. It had worked fine previously, I did not (knowingly) change anything, yet suddenly this problem occurred and was there to stay.

Of course I can provide additional log files, information, etc.; please just let me know what helps to fix this and I'll post it :)

Here is another log for the same bulb from just now. As you can see, I have turned it on multiple times, but it reported its status as off. Then suddenly it turned on...

```
❯ msub zigbee2mqtt/Philips_Hue_01/#
zigbee2mqtt/Philips_Hue_01/availability online
zigbee2mqtt/Philips_Hue_01/set {"state": "OFF"}
zigbee2mqtt/Philips_Hue_01 {"state":"OFF","linkquality":54,"last_seen":"2020-01-24T17:01:26+01:00","brightness":189,"device":{"friendlyName":"Philips_Hue_01","model":"8718696449691","ieeeAddr":"0x0017880102a2d83f","networkAddress":53354,"type":"Router","manufacturerID":4107,"manufacturerName":"Philips","powerSource":"Mains (single phase)","applicationVersion":2,"stackVersion":1,"zclVersion":1,"hardwareVersion":1,"dateCode":"20170311","softwareBuildID":"1.23.0_r20156"}}
zigbee2mqtt/Philips_Hue_01/set {"state": "ON"}
zigbee2mqtt/Philips_Hue_01 {"state":"OFF","linkquality":54,"last_seen":"2020-01-24T17:01:38+01:00","brightness":189,"device":{"friendlyName":"Philips_Hue_01","model":"8718696449691","ieeeAddr":"0x0017880102a2d83f","networkAddress":53354,"type":"Router","manufacturerID":4107,"manufacturerName":"Philips","powerSource":"Mains (single phase)","applicationVersion":2,"stackVersion":1,"zclVersion":1,"hardwareVersion":1,"dateCode":"20170311","softwareBuildID":"1.23.0_r20156"}}
zigbee2mqtt/Philips_Hue_01/set {"state": "ON"}
zigbee2mqtt/Philips_Hue_01 {"state":"ON","linkquality":54,"last_seen":"2020-01-24T17:02:04+01:00","brightness":189,"device":{"friendlyName":"Philips_Hue_01","model":"8718696449691","ieeeAddr":"0x0017880102a2d83f","networkAddress":53354,"type":"Router","manufacturerID":4107,"manufacturerName":"Philips","powerSource":"Mains (single phase)","applicationVersion":2,"stackVersion":1,"zclVersion":1,"hardwareVersion":1,"dateCode":"20170311","softwareBuildID":"1.23.0_r20156"}}
zigbee2mqtt/Philips_Hue_01 {"state":"ON","linkquality":54,"last_seen":"2020-01-24T17:02:15+01:00","brightness":189,"device":{"friendlyName":"Philips_Hue_01","model":"8718696449691","ieeeAddr":"0x0017880102a2d83f","networkAddress":53354,"type":"Router","manufacturerID":4107,"manufacturerName":"Philips","powerSource":"Mains (single phase)","applicationVersion":2,"stackVersion":1,"zclVersion":1,"hardwareVersion":1,"dateCode":"20170311","softwareBuildID":"1.23.0_r20156"}}

```

Thank you for your help

dzungpv, 2020-01-24:
@prankousky try these setting for queue:
queue: 
delay: 40
simultaneously: 25

prankousky, 2020-01-24:
Thank you @dzungpv 

This kinda works: when I toggle the lamp now, it will take about 25 seconds for it to react to the command. This is a big improvement, but still too slow. I will try playing around with those values more... but is there a general rule I can work with? 

For example, is simultaneously best set to a certain percentage of total devices? I have ~85 total, so is 25 good or should it be a higher/lower value? Same for delay, do either of those values have to do with the amount of devices, or is there another way to tweak it more without (what I will do now) trial and error?

dzungpv, 2020-01-24:
@prankousky younhave 85 devices but rarely you turn it all on or off, so not the problem, the above setting work fine for me. Or you can change delay like 40 and simulate 85, it take only 3.4 seconds to process all.

prankousky, 2020-01-24:
Thank you. Unfortunately, changing those options does not help. The lights now take between 15 - 25 seconds to change. Could something else in my configuration cause this? I have tried commenting differenting things out already, but nothing made a noticable difference as far as I can tell.

dzungpv, 2020-01-24:
@prankousky so it relate to other problem of yours, i reply in difference issue, please check that.

prankousky, 2020-01-25:
@dzungpv I replied in that thread (https://github.com/Koenkk/zigbee2mqtt/issues/2705#issuecomment-578389345). Just a quick summary if somebody found this one via search: the 15-25 seconds delay *was* gone yesterday; however, this was **merely** due to restarting zigbee2mqtt. A few hours after the last restart, the lag was incredibly high again.

So this likely does not have anything to do with the `queue` section in the configuration.yaml.

Koenkk, 2020-01-25:
The queue settings are not used anymore.

Try removing availability_timeout from your configuration.yaml, as the bigger the networks gets, the more stress it will put on it.

prankousky, 2020-01-25:
Thank you @Koenkk 

This is my updated `configuration.yaml`. After I just now restarted, the lights took about 30 seconds to react to any command (switch on/off, color change). Do you know what could cause this delay to grow according to the amount of time zigbee2mqtt is running? It always works best after a restart, then lag expands. And since just a few weeks ago there was **almost no lag at all**, I know that both hardware and software a capable of working faster than they do at the moment.

```
homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://ip:1883'
  user: my_user
  password: actual_password
  client_id: z2mbeelink
  include_device_information: true
serial:
    port: null # changed to this to auto-detect the correct port, as the CC1352P-2 will show up as two different devices (according to official documentation)
  disable_led: false
advanced:
  pan_id: #
  network_key: #
  homeassistant_discovery_topic: homeassistant
  homeassistant_status_topic: hass/status
  last_seen: ISO_8601_local
  elapsed: true
  report: true
  log_level: debug
  cache_state: true
```

prankousky, 2020-01-26:
A little update on the situation: I restarted `zigbee2mqtt` last night because I was not able to control my light at all (or the delay must have been over 10 minutes, that's how long I had waited beforehand). After restart, it would not work, so I restarted the entire server...

**After that**, everything reacted as expected. I'd toggle a light, and it'd actually work as soon as I did. Same goes for the thermostats. However, this morning I had the lag again. I'd say between 30 - 45 seconds after toggling a light in Home Assistant until the light did actually change. As before, when I'd look at the `mqtt` messages, as soon as I'd toggle the light in Home Assistant, it would send the payload. However, the light would very seldomly report it's status, so I assume there is something wrong there. I just don't get why regular sensors would work right on time (I open a door, it will be displayed as open), when **any** light takes so long. 

```

state
OFF
linkquality
63
last seen
2020-01-26T13:14:35+01:00
color
{ "x": 0.177, "y": 0.655 }
elapsed
30845
device
{ "friendlyName": "Schlafzimmer_Innr", "model": "RB 285 C", "ieeeAddr": "0x00158d0002f6ef2c", "networkAddress": 9067, "type": "Router", "manufacturerID": 4454, "manufacturerName": "innr", "powerSource": "Mains (single phase)", "applicationVersion": 16, "stackVersion": 21, "zclVersion": 2, "hardwareVersion": 1, "dateCode": "20180813-1", "softwareBuildID": "2.0" }

```

Last seen is ~3 minutes from when I am typing this; I am not sure what the `elapsed` value is... milliseconds? Either way, `linkquality` is at least okay. It was at about 19 when I unplugged another light (`Philips Aurora`) that is in between the coordinator and this `Innr` light; still, there was a connection... I plugged the Philips right back in though to have a better link quality.

Koenkk, 2020-01-26:
Can you provide the log when running with herdsman debug logging enabled? We need to know what fills up the queue.

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

sjorge, 2020-01-26:
> The queue settings are not used anymore.
> 
> Try removing availability_timeout from your configuration.yaml, as the bigger the networks gets, the more stress it will put on it.

I've been toying with the idea of having a 'lite' version that does what we now do for battery devices... but for powered ones too. So no more pings. That should not put extra load on the network right?

Koenkk, 2020-01-27:
@sjorge good idea, but I'm not sure if e.g. bulbs check in every day when not controlling them so that has to be checked first.

sjorge, 2020-01-27:
Hmmm... true, my ikea publish attReports when using them (on/off, level) my one hue does too... but indeed if not using them they might not, as there is nothing to report.

prankousky, 2020-01-27:
> Can you provide the log when running with herdsman debug logging enabled? We need to know what fills up the queue.

@Koenkk I feel pretty stupid now, but I believe that I have found the problem. 

I have a mini server rack with metal enclosure. To assure that I'd still have decent reception, I used a USB extension cable from my home server (inside the rack) to the CC1352P-2 (outside, on top of the rack). There must have been **something** wrong with the extension cable; perhaps it provided insufficient energy for the CC1352P-2. 

What I did to fix this was substitute the regular USB extension cable with an active USB hub; after I connected the CC1352P-2 to this hub, it worked fine again. Sorry for all this commotion about this, when the problem was actually caused by some crappy cable. I can still provide the logs, but I doubt there will be anything in them now, as everything works fine again.

Unfortunately by now I already ordered an additional `CC26X2R1` (which is already shipped out). This might not be necessary at all, but still: would it be possible to use my existing CC1352P-2 as coordinator and the CC26X2R1 as a router? In case you are working on a coordinator firmware I'd be willing to test it on the CC26X2R1, as I don't have any other use for it (I assume the CC1352P-2 that I am currently -and finally properly- using at the moment already has better coverage than the CC26X2R1 so I don't see why I should change the coordinator hardware - is this correct?).

So this issue is solved. I am not going to close it yet because I'll wait for your reply. :)

Koenkk, 2020-01-27:
@prankousky great that you've found the issue, in terms of coverage I don't think there is a big difference between `CC26X2R1` and `CC1352P-2`. Let's talk about the route firmware here: #1429

prankousky, 2020-02-01:
@Koenkk unfortunately, I experienced this problem again today. So the USB extension cable was not the solution after all.

I will provide the logs you requested in your previous post tomorrow. Stopping / restarting zigbee2mqtt takes a rather long time... It has before as well, but after I thought the issue was solved, restarting and stopping would only take a second or two. Now it takes at least 30 seconds, sometimes considerably longer.

prankousky, 2020-02-02:
@Koenkk can you please tell me how I can log this output? When I run `DEBUG=zigbee-herdsman* npm start > log.log`, it will not write any of those zigbee-herdsman related things that I see in my terminal to that log. Instead, the logfile I piped to `log.log` looks just like a regular logfile that is created without `DEBUG=zigbee-herdsman`. 

The last thing I did that fixed this problem **temporarily** was

* `sudo systemctl stop  zigbee2mqtt.service`
* clearing ram and cache `sudo sync; sudo echo 2 > /proc/sys/vm/drop_caches`
* `sudo reboot now` 

After the reboot, everything reacted just fine again (lights and thermostats did not have any delay). That was yesterday evening after I posted my previous comment. This morning, there was incredible lag again. 

I can also just `sudo reboot now`; this will not get rid of the lag, *but* one particular light that did not respond at all would respond again - with about 45 seconds of lag.

There is so much DEBUG output for zigbee-herdsman that is displayed alongside the regular debug output that I cannot copy and paste it from the terminal. What should I do to provide the information you need?

Koenkk, 2020-02-02:
I think you need to run with `DEBUG=zigbee-herdsman* npm start > log.log  2>&1`

prankousky, 2020-02-03:
That worked, thank you @Koenkk 

The command only ran for a couple of minutes, yet the log is almost 4MB. I used `Philips_Hue_01` as an example, turning it on/off a couple of time while logging. It took quite some time in between. Does the log tell you anything that might indicate what is wrong here and/or what I need to change?

[This](https://github.com/prankousky/z2mlog/raw/master/log.log) is the log. [Alternative Link](https://github.com/prankousky/z2mlog/blob/master/log.log)

Koenkk, 2020-02-03:
There is a ridiculous amount of commands being send to `Zwischenzimmer_Heizung` and `Schlafzimmer_Heizung`

If you search the log for any of these names you see that almost every second a lot of commands are executed.

```
[34mzigbee2mqtt:debug[39m 2020-02-03 12:59:55: Received MQTT message on 'zigbee2mqtt/Zwischenzimmer_Heizung/set/system_mode' with data 'off'
[34mzigbee2mqtt:debug[39m 2020-02-03 12:59:55: Publishing 'set' 'system_mode' to 'Zwischenzimmer_Heizung'
[34mzigbee2mqtt:debug[39m 2020-02-03 12:59:55: Received MQTT message on 'zigbee2mqtt/Schlafzimmer_Heizung/set/system_mode' with data 'off'
[34mzigbee2mqtt:debug[39m 2020-02-03 12:59:55: Publishing 'set' 'system_mode' to 'Schlafzimmer_Heizung'
[34mzigbee2mqtt:debug[39m 2020-02-03 12:59:55: Received MQTT message on 'zigbee2mqtt/Zwischenzimmer_Heizung/set/system_mode' with data 'off'
[34mzigbee2mqtt:debug[39m 2020-02-03 12:59:55: Publishing 'set' 'system_mode' to 'Zwischenzimmer_Heizung'
[34mzigbee2mqtt:debug[39m 2020-02-03 12:59:55: Received MQTT message on 'zigbee2mqtt/Schlafzimmer_Heizung/set/system_mode' with data 'off'
[34mzigbee2mqtt:debug[39m 2020-02-03 12:59:55: Publishing 'set' 'system_mode' to 'Schlafzimmer_Heizung'
[34mzigbee2mqtt:debug[39m 2020-02-03 12:59:55: Received MQTT message on 'zigbee2mqtt/Zwischenzimmer_Heizung/set/system_mode' with data 'off'
[34mzigbee2mqtt:debug[39m 2020-02-03 12:59:55: Publishing 'set' 'system_mode' to 'Zwischenzimmer_Heizung'
[34mzigbee2mqtt:debug[39m 2020-02-03 12:59:55: Received MQTT message on 'zigbee2mqtt/Schlafzimmer_Heizung/set/system_mode' with data 'off'
[34mzigbee2mqtt:debug[39m 2020-02-03 12:59:55: Publishing 'set' 'system_mode' to 'Schlafzimmer_Heizung'
````

You should find out who is sending all of these MQTT messages, e.g. perhaps a bug in a Home Assistant automation?

prankousky, 2020-02-03:
Thank you! Good point. I deactivated that automation and restarted, and currently everything reacts perfectly fine again. I cannot say for sure if this is the only problem, as sometimes after a restart zigbee2mqtt will work just fine and then slow down after a couple of hours. 

I am going to wait until tomorrow without re-activating these automations, nor changing anything. If it is still stable by tomorrow, the automation seems to be the problem. Hopefully it is. If not I will report back and/or provide more zigbee-herdsman logifles. Thank you so much for your help so far, I am sure this can be fixed :)

prankousky, 2020-02-06:
Thank you for all your help and patience, @Koenkk 
The problem definitely lay with Home Assistant, not zigbee2mqtt. My automation is currently still disabled, and *almost* everything works fine.

The only problem is now that my `Eurotronic SPZB0001` thermostats still don't work reliably. Two of them always have to be manually connected, while all the others work without any trouble. I assume this is also not related to zigbee2mqtt itself, but the thermostats (I heard they have been shipped out with different firmwares). Still it is annoying. 

Is it possible to debug only two paired devices instead of all of them? I guess I should close this issue and, if so, open a separate issue for the thermostats. Again, thank you so much.

prankousky, 2020-02-06:
(re-opened just so you can reply if you got the time; it can then be closed as the original issue has been resolved :+1: )

Koenkk, 2020-02-06:
It's not easy to debug only 2 devices, but note that there is already an issue open for this device: https://github.com/Koenkk/zigbee-herdsman-converters/issues/715

closed 2020-02-06 by Koenkk