Koenkk/zigbee2mqtt #2816: innr SP 222 not seem to report state or linkquality

yapweiliang, 2020-01-24:
# Bug Report
## What happened

Although I can get this to switch on/off, I cannot get the state or the linkquality.  Am I missing something, or is this a limitation of the SP 222?

## What did you expect to happen

I expected that the state of the socket would be reported or updated when I press the manual switch on the device itself

## How to reproduce it (minimal and precise)

press the manual switch on SP 222.  Expect a zigbee2mqtt debug log entry about the change in state.  Expect Openhab UI to update the representation of the switch.  No change is received.   No log entry observed about linkquality either.

However setting the state works with no problems. 

## Debug Info
zigbee2mqtt version: 1.9.0 (commit #287f843)
CC253X firmware version: 20190425

yapweiliang, 2020-01-26:
Thank you.  I have manually updated my devices.js, and I also added 

`meta: {configureKey: 2},`

(because the SP 224 and the SP 120 have that line, but I do not know whether it is really needed or not for SP 222)

So it works for me now.  Thank you again

Koenkk, 2020-01-26:
Great, will merge it in the dev branch in a few hours, thanks!

closed 2020-01-26 by Koenkk