Koenkk/zigbee2mqtt #2818: CR Smart Home TS0202 motion sensor always occupied

groenmarsmannetje, 2020-01-25:
# Bug Report
## What happened
I installed a new CR Smart Home TS0202 motion sensor. The status is always occupied in Home Assistant even after leaving the device stay in an empty room for a day.

## What did you expect to happen
The device should switch status to occupied = false after a short time.

## How to reproduce it (minimal and precise)
From to moment that I paired this device. Reconnecting, resetting and repairing doesn’t have a different effect.

## Debug Info
Version of Zigbee2Mqtt: 1.9.0 
Coordinator version: 20190619

Koenkk, 2020-01-25:
CR Smart Home send me this device (send it back so I don't have it anymore now) and I've integrated it.

Thinking now I think the integration is wrong as we can impossible provide an `occupancy` `false` for this device, also discussed this with CR Smart Home and they confirmed it. This is a limitation of the device.

The problem: the device will only send an event when there has no occupancy been detected for 1 minute.

> The device should switch status to occupied = false after a short time.

Let's say the short time = 90 seconds here (this is what is also used for Xiaomi motion sensor).

Example:
- 00:00 device detects occupancy and sends event
- 00:30 device detects occupancy and DOES NOT send an event
- 01:01 device detects occupancy and DOES NOT send an event (<-- as last occupancy was detected 31 seconds ago)
- 01:30 90 seconds have been passed now, at this point zigbee2mqtt cannot know if there is still occupancy as it doesn't now if there have been any occupancy events between 00:01 and 01:29

Proposed solution: remove the binary occupancy sensor and add an action sensor, `action` will then go to `occupancy` once the device send the `occupancy` event.

groenmarsmannetje, 2020-01-26:
I am not sure if I understand your proposed solution. Assume that action has become occupancy what will then cause that the action will switch to something else? Or is the action only one event like a button click?

Koenkk, 2020-01-26:
With this device we cannot determine occupancy false, therefore its not suited to control lights with (as you need both true and false occupancy events).

ChristianPorsche, 2020-01-31:
Hi Koenkk, we had contact earlier about the behavior of this device. You found a solution, but apparently it isn't working. Can you change the code or is this device simply not compatible with Zigbee2MQTT?

Koenkk, 2020-01-31:
@ChristianPorsche I've mentioned this problem in the chat we had and you mentioned that this behavior is the same with the original gateway, it's not about being compatible with zigbee2mqtt, this device just doesn't send events when occupancy is detected within 60 seconds of occupancy detection (https://github.com/Koenkk/zigbee2mqtt/issues/2818#issuecomment-578437995)

groenmarsmannetje, 2020-02-02:
And with that behavior it is actually useless. It turns on once and never turns off again in  Home Assistant because it never gets a state change.

Koenkk, 2020-02-02:
In the latest zigbee2mqtt dev branch it will now provide an occupancy false event (after 90 seconds of no occupancy), @groenmarsmannetje please check if it works.

groenmarsmannetje, 2020-02-03:
> In the latest zigbee2mqtt dev branch it will now provide an occupancy false event (after 90 seconds of no occupancy), @groenmarsmannetje please check if it works.

I checked with the dev branch and it works. Thank you!

Koenkk, 2020-02-04:
@groenmarsmannetje it even works when staying in the room for more then 90 seconds?

groenmarsmannetje, 2020-02-04:
@Koenkk good question. I checked that now. Occupancy turns off after 90 seconds regardless if you are still in the room. If you directly walk out and walk again it will not turn on. You have to wait a short time (probably a minute) when it responds again.

For occupied rooms that will be very annoying because lights will turn on and off while you are still in the room. But for my case it is almost good enough because it is a hallway that you most of the time only pass for a short duration.

Koenkk, 2020-02-05:
Good, as this is the best we can get this device I will close this.

closed 2020-02-05 by Koenkk

Cportelli, 2021-01-10:
As per manual it says that when continuous motion is detected, it won't send an event until no motion is detected again for 1 minute. After that it will restore no motion status and will go into "working mode".

I can see both events, the first motion detected and the other no motion that triggers after 1 minute of last motion detected. However the false event (after 90 seconds of no occupancy) if throwing of the automation if the room is occupied for more then 90 seconds.

_Zigbee2MQTT:info  2021-01-10 22:24:22: MQTT publish: topic 'zigbee2mqtt/Motion PIR Sensor', payload '{"battery":100,"battery_low":false,"linkquality":84,"occupancy":true,"tamper":false}'
Zigbee2MQTT:info  2021-01-10 22:25:52: MQTT publish: topic 'zigbee2mqtt/Motion PIR Sensor', payload '{"battery":100,"battery_low":false,"linkquality":84,"occupancy":false,"tamper":false}' **---FALSE EVENT**
Zigbee2MQTT:info  2021-01-10 22:27:10: MQTT publish: topic 'zigbee2mqtt/Motion PIR Sensor', payload '{"battery":100,"battery_low":false,"linkquality":78,"occupancy":false,"tamper":false}'  ---**Actual 1min after no motion**

groenmarsmannetje, 2021-02-09:
@Koenkk Seems that something is changed (in #5690 ?) and now this earlier change is not working any more in my case for my device.  Device is now also reported as a TuYa TS0202 in front-end UI while it was earlier identified as device from brand CR Smart Home. Could it be that different makes/versions are mixed up now causing this unexpected effect?

Device has become totally useless again to me now.

Koenkk, 2021-02-09:
@groenmarsmannetje can you provide the data/database.db entry of this device?

groenmarsmannetje, 2021-02-09:
@Koenkk  see below

{"id":34,"type":"EndDevice","ieeeAddr":"0xec1bbdfffe782df5","nwkAddr":49299,"manufId":0,"manufName":"_TYZB01_jytabjkb","powerSource":"Battery","modelId":"TS0202","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,1280],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"TS0202","manufacturerName":"_TYZB01_jytabjkb","powerSource":3,"zclVersion":3,"appVersion":64}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b00215ff740","zoneState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1612891670988}

Koenkk, 2021-02-09:
@groenmarsmannetje fixed, Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

groenmarsmannetje, 2021-02-09:
That’s great! I will check with the new dev version when it is available.

groenmarsmannetje, 2021-02-10:
I can confirm that it is working again for my device with latest dev version. Thanks.

DavidFFerreira, 2021-04-27:
@groenmarsmannetje I have the same PIR but apparently, after while it just stop working, and after another 10 or 15m is start to working again :O, do you have the same probleme?