Koenkk/zigbee2mqtt #2826: Sercomm SZ-DWS04's parsing incorrectly

luckypants123, 2020-01-26:
Sercomm SZ-DWS04's magnetic contact sensors have stopped reporting their correct state very recently (maybe since 3ish days ago?).

They report contact: false and tamper: false regardless of the actual state. I've tested dev/edge and stable on both Hassio and Buster lite.

Eg: If i depress the tamper switch or open/close the switch a new message appears, however will always report:
```
zigbee2mqtt:debug 2020-01-26 02:07:21: Received Zigbee message from 'garage-contact-mandoor', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"zonestatus":33,"extendedstatus":0}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-01-26 02:07:21: MQTT publish: topic 'z2m_garage/garage-contact-mandoor', payload '{"contact":false,"tamper":false,"battery_low":false,"linkquality":102}'
```
I'd suspect the battery_low property is also not reporting correctly.

Link quality seems to be reporting properly - however oddly it is showing up in my log every 20 seconds ... with a new message .. i'd imagine its not sleeping?

What can I provide/test to assist?

Ian.

Koenkk, 2020-01-26:
Did you update zigbee2mqtt or something else that could be related to this?

According to the zigbee cluster library specification zoneStatus `33` means that contact, low_battery and tamper are all false. What zoneStatus does it send when the contact is opened?

luckypants123, 2020-01-27:
I don't think so. Its a pretty bare bones installation.

I just did some tests...here are the results.

Sensor mounted and door opened (should be contact:false, tamper: false):
```
zigbee2mqtt:debug 2020-01-27 01:20:07: Received Zigbee message from 'garage-contact-mandoor', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"zonestatus":33,"extendedstatus":0}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-01-27 01:20:08: MQTT publish: topic 'z2m_garage/garage-contact-mandoor', payload '{"contact":false,"tamper":false,"battery_low":false,"linkquality":47}'
```
Sensor mounted and door closed (should be contact:true, tamper: false):
```
zigbee2mqtt:debug 2020-01-27 01:20:20: Received Zigbee message from 'garage-contact-mandoor', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"zonestatus":33,"extendedstatus":0}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-01-27 01:20:20: MQTT publish: topic 'z2m_garage/garage-contact-mandoor', payload '{"contact":false,"tamper":false,"battery_low":false,"linkquality":34}'
```
Sensor dismounted and door closed (should be contact:true, tamper: true):
```
zigbee2mqtt:debug 2020-01-27 01:20:29: Received Zigbee message from 'garage-contact-mandoor', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"zonestatus":33,"extendedstatus":0}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-01-27 01:20:29: MQTT publish: topic 'z2m_garage/garage-contact-mandoor', payload '{"contact":false,"tamper":false,"battery_low":false,"linkquality":39}'
```
Seems to report 33 no matter what I do to it. Very odd.

Any ideas for troubleshooting?

Koenkk, 2020-01-27:
Looks like an issue of the device itself, try factory resetting and repairing it.

luckypants123, 2020-01-27:
I'll do some experimenting tonight - I have 5 of these devices currently not paired so I should be able to tell quickly if the device is suffering some type of fault.

luckypants123, 2020-01-28:
I setup a complete new instance - different pi, coordinator, and fresh install. Seems to be working well.

I'll leave this open for a couple more days while I try and reproduce the issue. If I cant, or discover its a hardware fault i'll close this.

luckypants123, 2020-01-29:
I'm unable to reproduce this. Chalk it up to a misconfiguration perhaps.

luckypants123, 2020-01-31:
Ok - Its started up again...this time on a different SZ-DWS04 sensor.

I enabled zigbee_herdsman_debug to capture that section of the log when changing the status of the sensor.

It was quite difficult - the sensor is either spamming the log or status is being requested nearly every 10 seconds.... 

I tampered the switch, and recorded that section of the log: https://pastebin.com/eMGZAmQ0
I untampered the switch and recorded that section of the log: https://pastebin.com/McC18XVp
And lastly a section of log spam, where no events are occurring: https://pastebin.com/qy5Xg161

I am unsure how to troubleshoot...do you have any ideas?

Koenkk, 2020-01-31:
Unfortunately I don't have an idea, it seems that the sensors goes crazy and starts reporting incorrect values. Do you have any other zigbee hub to test this device with?

luckypants123, 2020-01-31:
Ok - thank you.
I'll continue to troubleshoot and if I find anything worth reporting, I will reopen. I may just bin the whole lot of them.

closed 2020-01-31 by luckypants123