Koenkk/zigbee2mqtt #2827: Failed to interview '0x00158d000...', device has not successfully been paired

sh4jonny, 2020-01-26:
# Bug Report
## What happened
Xiaomi temperature sensor [WSDCGQ01LM] is joined to the network
Zigbee2MQTT is starting to interview

controller:device Interview - node descriptor request failed for '0x00158d00041e5d53', attempt 5
adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":35282,"nwkaddrofinterest":35282}
adapter:zStack:unpi:writer --> frame [254,4,37,2,210,137,210,137,35]
adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
adapter:zStack:unpi:parser --- parseNext []
controller:device Interview - active endpoints request failed for '0x00158d0004102e7d', attempt 1
adapter:zStack:znp:SREQ --> ZDO - activeEpReq - {"dstaddr":64597,"nwkaddrofinterest":64597}
adapter:zStack:unpi:writer --> frame [254,4,37,5,85,252,85,252,36]
adapter:zStack:unpi:parser <-- [254,1,101,5,0,97]
adapter:zStack:unpi:parser --- parseNext [254,1,101,5,0,97]
adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 5 - [0] - 97
adapter:zStack:znp:SRSP <-- ZDO - activeEpReq - {"status":0}
adapter:zStack:unpi:parser --- parseNext []

zigbee2mqtt:error 2020-01-26 13:48:03: Failed to interview '0x00158d0004102e7d', device has not successfully been paired
zigbee2mqtt:info  2020-01-26 13:48:03: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x00158d0004102e7d"}}'
2020-01-26T10:48:10.337Z zigbee-herdsman:controller:device Interview - active endpoints request failed for '0x00158d00041e5d53', attempt 1
2020-01-26T10:48:10.339Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - activeEpReq - {"dstaddr":35282,"nwkaddrofinterest":35282}
2020-01-26T10:48:10.341Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,5,210,137,210,137,36]
2020-01-26T10:48:10.350Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,5,0,97]
2020-01-26T10:48:10.351Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,5,0,97]
2020-01-26T10:48:10.360Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 5 - [0] - 97
2020-01-26T10:48:10.361Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - activeEpReq - {"status":0}
2020-01-26T10:48:10.361Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-01-26T10:48:20.344Z zigbee-herdsman:controller:device Interview - active endpoints request failed for '0x00158d00041e5d53', attempt 2
2020-01-26T10:48:20.352Z zigbee-herdsman:controller:device Interview - failed for device '0x00158d00041e5d53' with error 'Error: Interview failed because can not get active endpoints ('0x00158d00041e5d53')
    at Device.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/device.js:273:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/device.js:6:65)'
2020-01-26T10:48:20.358Z zigbee-herdsman:controller:error Interview failed for '0x00158d00041e5d53 with error 'Error: Interview failed because can not get active endpoints ('0x00158d00041e5d53')'

I have 3 new sensors and no one can be paired

## How to reproduce it (minimal and precise)
Reset the sensor by pressing the button for >4s
LED starts flashing, getting into paring mode,
But directly after the 3 flashes, it flash once more very quick.
Flashes after a while once more.

## Debug Info
zigbee2mqtt version: 1.9.0-dev
CC253X firmware version: CC2531_DEFAULT_20190608

Koenkk, 2020-01-26:
Can you try disconnecting the battery for a few minutes and pair again? Also keep holding the reset button until the LED stops flashing.

sh4jonny, 2020-01-28:
> Can you try disconnecting the battery for a few minutes and pair again? Also keep holding the reset button until the LED stops flashing.

hi @Koenkk! thank you for your reply. i try 3 different sensors. i try to replace battery and disconnect it also.

Koenkk, 2020-01-29:
Please try with the latest zigbee2mqtt dev branch, if it still fails please provide the herdsman debug logging.

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

sh4jonny, 2020-01-29:
zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,200,182,164,65,109,3,0,141,21,0,0,0,238] +26s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,200,182,164,65,109,3,0,141,21,0,0,0,238] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [200,182,164,65,109,3,0,141,21,0,0,0] - 238 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":46792,"extaddr":"0x00158d00036d41a4","parentaddr":0} +26s
  zigbee-herdsman:controller:log Device '0x00158d00036d41a4' joined +26s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +7ms
  zigbee-herdsman:controller:log Device '0x00158d00036d41a4' accepted by handler +3ms
  zigbee-herdsman:controller:log New device '0x00158d00036d41a4' joined +1ms
  zigbee-herdsman:controller:log Creating device '0x00158d00036d41a4' +1ms
zigbee2mqtt:info  2020-01-30 01:19:45: Device '0x00158d00036d41a4' joined
zigbee2mqtt:info  2020-01-30 01:19:45: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x00158d00036d41a4"}}'
  zigbee-herdsman:controller:log Interview '0x00158d00036d41a4' start +19ms
zigbee2mqtt:info  2020-01-30 01:19:45: Starting interview of '0x00158d00036d41a4'
zigbee2mqtt:info  2020-01-30 01:19:45: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00158d00036d41a4"}}'
  zigbee-herdsman:controller:device Interview - start device '0x00158d00036d41a4' +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":46792,"nwkaddrofinterest":46792} +26s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,200,182,200,182,35] +26s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102] +48ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102] +2ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0} +26s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:controller:device Interview - node descriptor request failed for '0x00158d00036d41a4', attempt 1 +10s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":46792,"nwkaddrofinterest":46792} +10s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,200,182,200,182,35] +10s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102] +10s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0} +10s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:controller:device Interview - node descriptor request failed for '0x00158d00036d41a4', attempt 2 +10s

sh4jonny, 2020-01-29:
so as i can see #2844 #2851 all of us have proubles with 2531
now i try to disable rtscts ("rtscts": false) and LEDs on my WXKG02LM blink more intensive

Koenkk, 2020-01-30:
Can you try reflashing your CC2531 and pairing it immediately after that? Also check https://www.zigbee2mqtt.io/information/FAQ.html#no-logging-is-shown-at-all

closed 2020-04-07 by stale[bot]