Koenkk/zigbee2mqtt #2830: Why "occupancy":false before "occupancy":true with Xiaomi RTCGQ11LM?

glsf91, 2020-01-26:
I' am running 1.9.0 version.
With the Xiaomi motion sensor I see the following in the logfile:

When motion detected:
`Jan 26 20:52:57 ubuntu1 npm[23856]: zigbee2mqtt:info  2020-01-26 20:52:57: MQTT publish: topic 'zigbee2mqtt/KeukenMotionSensorMS4', payload '{"battery":97,"voltage":2995,"illuminance":0,"linkquality":86,"occupancy":false}'`
`Jan 26 20:52:57 ubuntu1 npm[23856]: zigbee2mqtt:info  2020-01-26 20:52:57: MQTT publish: topic 'zigbee2mqtt/KeukenMotionSensorMS4', payload '{"battery":97,"voltage":2995,"illuminance":0,"linkquality":86,"occupancy":true}'`
`Jan 26 20:54:02 ubuntu1 npm[23856]: zigbee2mqtt:info  2020-01-26 20:54:02: MQTT publish: topic 'zigbee2mqtt/KeukenMotionSensorMS4', payload '{"battery":97,"voltage":2995,"illuminance":0,"linkquality":86,"occupancy":false}'`

When detecting motion again:

`Jan 26 20:54:32 ubuntu1 npm[23856]: zigbee2mqtt:info  2020-01-26 20:54:32: MQTT publish: topic 'zigbee2mqtt/KeukenMotionSensorMS4', payload '{"battery":97,"voltage":2995,"illuminance":0,"linkquality":86,"occupancy":false}'`
`Jan 26 20:54:32 ubuntu1 npm[23856]: zigbee2mqtt:info  2020-01-26 20:54:32: MQTT publish: topic 'zigbee2mqtt/KeukenMotionSensorMS4', payload '{"battery":97,"voltage":2995,"illuminance":0,"linkquality":84,"occupancy":true}'`

So when detecting motion it is first sending an "occupancy":false, even if it is already sent before.
This sometimes causing in Domoticz the light is going on and out again at the same second. I think because of the "occupancy":false in the same second.

I don't expect a resend of the already send "occupancy":false when motion is detected.

My configuration for the device:

devices:
  '0x00158d0001643f99':
    friendly_name: KeukenMotionSensorMS4
    occupancy_timeout: 65

Koenkk, 2020-01-27:
Please provide the log with zigbee2mqtt debug enabled. (https://www.zigbee2mqtt.io/information/debug.html#zigbee2mqtt-debug-logging)

glsf91, 2020-01-27:
> Please provide the log with zigbee2mqtt debug enabled. (https://www.zigbee2mqtt.io/information/debug.html#zigbee2mqtt-debug-logging)

https://pastebin.com/gmmifkEb

At 2020-01-27 20:03:26 motion was detected. At 2020-01-27 20:04:31 timeout 65s occurred.
New motion at 2020-01-27 20:04:58.

Koenkk, 2020-01-27:
Sensor sends illuminance value, occupancy is still false at this point, zigbee2mqtt sends new illuminance value together with current state (e.g. `occupancy`)
```
debug 2020-01-27 20:04:58: Received Zigbee message from 'KeukenMotionSensorMS4', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":0}' from endpoint 1 with groupID 0
info  2020-01-27 20:04:58: MQTT publish: topic 'zigbee2mqtt/KeukenMotionSensorMS4', payload '{"battery":97,"voltage":2995,"illuminance":0,"linkquality":86,"occupancy":false}'
```

In the same second, the sensor send the occupancy detection message, zigbee2mqtt sends occupancy true
```
debug 2020-01-27 20:04:58: Received Zigbee message from 'KeukenMotionSensorMS4', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 1 with groupID 0
info  2020-01-27 20:04:58: MQTT publish: topic 'zigbee2mqtt/KeukenMotionSensorMS4', payload '{"battery":97,"voltage":2995,"illuminance":0,"linkquality":84,"occupancy":true}'
```

So the reason for this is that right before the sensor send occupancy it send the illuminance report. In case you don't want this you can use the `debounce` option with value `1` (https://www.zigbee2mqtt.io/information/configuration.html#all-devices) note that this will delay your occupancy events by 1 second.

Koenkk, 2020-01-28:
This is needed for home assistant integration (requires all attributes), if you dont want this set `cache_state: false`. https://www.zigbee2mqtt.io/information/configuration.html

glsf91, 2020-01-29:
This is working better (I hope). Indeed the "occupancy":false is not send anymore.

closed 2020-01-29 by glsf91