Koenkk/zigbee2mqtt #2831: Forgot to disable permit join, now my network went crazy

Roeliooo, 2020-01-26:
# Bug Report
After setting up my Zigbee network, I **forgot** to set permit join to FALSE (stupid stupid stupid). My neighbors added some philips hue zigbee devices to their own zigbee network but some of them are also picked up by my zigbee2mqtt. And now, zigbee2mqtt won't start anymore with some random errors.

## What did you expect to happen
Normal operation.

## How to reproduce it (minimal and precise)
Set your zigbee2mqtt network on permit join = true. Buy a Philips Hue Bridge and some Hue bulbs and begin adding them to your hue bridge. This is exactly what happenend  by my neighbor. After a few minutes zigbee2mqtt will also picked them up and is going crazy.

## Debug Info
zigbee2mqtt version: 1.8
CC253X firmware version: CC2531_DEFAULT_20190608

In the YAML config file I can see there are 4 device from my neighbor added. I can't remove them from my zigbee network, so I added them to BAN but nothing helped.
```
ban:
  - '0x000b57fffea6ff65'
  - '0x0017880108413ae2'
  - '0x00178801084140e0'
  - '0x00178801084c3ab2'
```

If I start Zigbee2mqtt, the following errors appeared;
```
Jan 26 21:08:54 raspberrypi systemd[1]: Started zigbee2mqtt.
Jan 26 21:08:56 raspberrypi npm[1259]: > zigbee2mqtt@1.8.0 start /opt/zigbee2mqtt
Jan 26 21:08:56 raspberrypi npm[1259]: > node index.js
Jan 26 21:08:58 raspberrypi npm[1259]: zigbee2mqtt:info  2020-01-26 21:08:58: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-01-26.21-08-58'
Jan 26 21:08:59 raspberrypi npm[1259]: zigbee2mqtt:info  2020-01-26 21:08:59: Starting zigbee2mqtt version 1.8.0 (commit #da4d26a)
Jan 26 21:08:59 raspberrypi npm[1259]: zigbee2mqtt:info  2020-01-26 21:08:59: Starting zigbee-herdsman...
Jan 26 21:09:21 raspberrypi npm[1259]: (node:1274) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - startupFromApp after 20000ms
Jan 26 21:09:21 raspberrypi npm[1259]:     at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
Jan 26 21:09:21 raspberrypi npm[1259]:     at ontimeout (timers.js:436:11)
Jan 26 21:09:21 raspberrypi npm[1259]:     at tryOnTimeout (timers.js:300:5)
Jan 26 21:09:21 raspberrypi npm[1259]:     at listOnTimeout (timers.js:263:5)
Jan 26 21:09:21 raspberrypi npm[1259]:     at Timer.processTimers (timers.js:223:10)
Jan 26 21:09:21 raspberrypi npm[1259]: (node:1274) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
Jan 26 21:09:21 raspberrypi npm[1259]: (node:1274) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
Jan 26 21:10:01 raspberrypi npm[1259]: zigbee2mqtt:error 2020-01-26 21:10:01: Error while starting zigbee-herdsman
Jan 26 21:10:01 raspberrypi npm[1259]: zigbee2mqtt:error 2020-01-26 21:10:01: Failed to start zigbee
Jan 26 21:10:01 raspberrypi npm[1259]: zigbee2mqtt:error 2020-01-26 21:10:01: Exiting...
Jan 26 21:10:01 raspberrypi npm[1259]: zigbee2mqtt:error 2020-01-26 21:10:01: Error: AREQ - ZDO - stateChangeInd after 60000ms
Jan 26 21:10:01 raspberrypi npm[1259]:     at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
Jan 26 21:10:01 raspberrypi npm[1259]:     at ontimeout (timers.js:436:11)
Jan 26 21:10:01 raspberrypi npm[1259]:     at tryOnTimeout (timers.js:300:5)
Jan 26 21:10:01 raspberrypi npm[1259]:     at listOnTimeout (timers.js:263:5)
Jan 26 21:10:01 raspberrypi npm[1259]:     at Timer.processTimers (timers.js:223:10)
Jan 26 21:10:01 raspberrypi npm[1259]: npm ERR! code ELIFECYCLE
Jan 26 21:10:01 raspberrypi npm[1259]: npm ERR! errno 1
Jan 26 21:10:01 raspberrypi npm[1259]: npm ERR! zigbee2mqtt@1.8.0 start: `node index.js`
Jan 26 21:10:01 raspberrypi npm[1259]: npm ERR! Exit status 1
Jan 26 21:10:01 raspberrypi npm[1259]: npm ERR!
Jan 26 21:10:01 raspberrypi npm[1259]: npm ERR! Failed at the zigbee2mqtt@1.8.0 start script.
Jan 26 21:10:01 raspberrypi npm[1259]: npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
Jan 26 21:10:01 raspberrypi npm[1259]: npm ERR! A complete log of this run can be found in:
Jan 26 21:10:01 raspberrypi npm[1259]: npm ERR!     /home/pi/.npm/_logs/2020-01-26T20_10_01_484Z-debug.log
Jan 26 21:10:01 raspberrypi systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Jan 26 21:10:01 raspberrypi systemd[1]: zigbee2mqtt.service: Unit entered failed state.
Jan 26 21:10:01 raspberrypi systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Jan 26 21:10:01 raspberrypi systemd[1]: zigbee2mqtt.service: Service hold-off time over, scheduling restart.
Jan 26 21:10:01 raspberrypi systemd[1]: Stopped zigbee2mqtt.

```

Can somebody help me how to fix zigbee2mqtt? If more info is needed please let me know.

sjorge, 2020-01-26:
Not sure about your bug, but wouldn't this only happen if you and your neighbor are using the same zigbee channel (+ network key)?

trongdong, 2020-01-27:
@Roeliooo not sure about your bug, but if you don't mind paring all your device again, remove database, change pan id + network key will help!

Roeliooo, 2020-01-27:
> 
> 
> Not sure about your bug, but wouldn't this only happen if you and your neighbor are using the same zigbee channel (+ network key)?

Yes, I think he used the same zigbeechannel when this happend. For now he changed his zigbeechannel to 25. All his devices are working correctly.

> 
> 
> @Roeliooo not sure about your bug, but if you don't mind paring all your device again, remove database, change pan id + network key will help!

I looked in the database located in `/opt/zigbee2mqtt/data` and I see the 4 listed spooky devices with ID 17, 18, 19 and 20, he added also an ikea power socket which is listed in my database:

> {"id":17,"type":"Router","ieeeAddr":"0x000b57fffea6ff65","nwkAddr":29502,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb GU10 W 400lm","epList":[1],"endpoints":{"1":{"profId":49246,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[5,25,32,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI bulb GU10 W 400lm","manufacturerName":"IKEA of Sweden","powerSource":1,"zclVersion":1,"appVersion":17,"stackVersion":87,"hwVersion":1,"dateCode":"20170302","swBuildId":"1.2.214"}}},"binds":[]}},"appVersion":17,"stackVersion":87,"hwVersion":1,"dateCode":"20170302","swBuildId":"1.2.214","zclVersion":1,"interviewCompleted":true,"meta":{}}
> `
> {"id":18,"type":"Router","ieeeAddr":"0x0017880108413ae2","nwkAddr":8780,"manufId":4107,"epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":268,"inClusterList":[0,3,4,5,6,8,4096,64514,768],"outClusterList":[25],"clusters":{},"binds":[]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[]}},"interviewCompleted":false,"meta":{}}
> 
> {"id":19,"type":"Router","ieeeAddr":"0x00178801084140e0","nwkAddr":26015,"manufId":4107,"epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":268,"inClusterList":[0,3,4,5,6,8,4096,64514,768],"outClusterList":[25],"clusters":{},"binds":[]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[]}},"interviewCompleted":false,"meta":{}}
> 
> {"id":20,"type":"Router","ieeeAddr":"0x00178801084c3ab2","nwkAddr":27434,"manufId":4107,"epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":268,"inClusterList":[0,3,4,5,6,8,4096,64514,768],"outClusterList":[25],"clusters":{},"binds":[]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[]}},"interviewCompleted":false,"meta":{}}
> 

If I removed these 4 devices from the Database file, nothing happend, still getting the same errors and zigbee2mqtt herdsman will not start. Is there another database ?

Roeliooo, 2020-01-27:
Okay little update. What I did is reflash my CC2351 Zigbeestick. From now on, Zigbee-herdsman will start, list all the devices in my network, but after i get a lot of errors. So i think a little step further... ??
can somebody help me?

```

Jan 27 19:44:17 raspberrypi npm[970]: zigbee2mqtt:info  2020-01-27 19:44:17: Starting zigbee2mqtt version 1.8.0 (commit #da4d26a)
Jan 27 19:44:17 raspberrypi npm[970]: zigbee2mqtt:info  2020-01-27 19:44:17: Starting zigbee-herdsman...
Jan 27 19:44:18 raspberrypi npm[970]: zigbee2mqtt:info  2020-01-27 19:44:18: zigbee-herdsman started
Jan 27 19:44:18 raspberrypi npm[970]: zigbee2mqtt:info  2020-01-27 19:44:18: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
**Jan 27 19:44:18 raspberrypi npm[970]: zigbee2mqtt:warn  2020-01-27 19:44:18: Banned device is connected (0x000b57fffea6ff65), removing...
Jan 27 19:44:18 raspberrypi npm[970]: zigbee2mqtt:warn  2020-01-27 19:44:18: Banned device is connected (0x0017880108413ae2), removing...
Jan 27 19:44:18 raspberrypi npm[970]: zigbee2mqtt:warn  2020-01-27 19:44:18: Banned device is connected (0x00178801084140e0), removing...
Jan 27 19:44:18 raspberrypi npm[970]: zigbee2mqtt:warn  2020-01-27 19:44:18: Banned device is connected (0x00178801084c3ab2), removing...**
Jan 27 19:44:18 raspberrypi npm[970]: zigbee2mqtt:info  2020-01-27 19:44:18: Currently 19 devices are joined:
Jan 27 19:44:18 raspberrypi npm[970]: zigbee2mqtt:info  2020-01-27 19:44:18: AqaraTempSensor03 (0x00158d0002b883f0): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
Jan 27 19:44:18 raspberrypi npm[970]: zigbee2mqtt:info  2020-01-27 19:44:18: AqaraTempSensor04 (0x00158d0002480273): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
Jan 27 19:44:18 raspberrypi npm[970]: zigbee2mqtt:info  2020-01-27 19:44:18: AqaraTempSensor02 (0x00158d0002786bd1): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
Jan 27 19:44:18 raspberrypi npm[970]: zigbee2mqtt:info  2020-01-27 19:44:18: AqaraMotionSensor1 (0x00158d000200e377): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
Jan 27 19:44:18 raspberrypi npm[970]: zigbee2mqtt:info  2020-01-27 19:44:18: Philips_klein_arau (0x00178801001cc1c0): 7099860PH - Philips LivingColors Aura (Router)
Jan 27 19:44:18 raspberrypi npm[970]: zigbee2mqtt:info  2020-01-27 19:44:18: Philips_groot_iris (0x00178801000dedd2): 7199960PH - Philips Hue Iris (Router)
Jan 27 19:44:18 raspberrypi npm[970]: zigbee2mqtt:info  2020-01-27 19:44:18: ZigbeeRouter (0x00124b001949f082): CC2530.ROUTER - Custom devices (DiY) [CC2530 router](http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/) (Router)
Jan 27 19:44:18 raspberrypi npm[970]: zigbee2mqtt:info  2020-01-27 19:44:18: Innr_1 (0xccccccfffe2c5cfb): RF 263 - Innr E27 filament bulb dimmable (Router)
Jan 27 19:44:18 raspberrypi npm[970]: zigbee2mqtt:info  2020-01-27 19:44:18: Innr_2 (0xccccccfffe2c636d): RF 263 - Innr E27 filament bulb dimmable (Router)
Jan 27 19:44:18 raspberrypi npm[970]: zigbee2mqtt:info  2020-01-27 19:44:18: PhilipsOutdoorPIR (0x001788010645814c): 9290019758 - Philips Hue motion outdoor sensor (EndDevice)
Jan 27 19:44:18 raspberrypi npm[970]: zigbee2mqtt:info  2020-01-27 19:44:18: PhilipsDimmerSwitch (0x0017880106732d69): 324131092621 - Philips Hue dimmer switch (EndDevice)
Jan 27 19:44:18 raspberrypi npm[970]: zigbee2mqtt:info  2020-01-27 19:44:18: Ikea_outlet_1 (0x086bd7fffe021f94): E1603/E1702 - IKEA TRADFRI control outlet (Router)
Jan 27 19:44:18 raspberrypi npm[970]: zigbee2mqtt:info  2020-01-27 19:44:18: Ikea_outlet_2 (0x000b3cfffef6be4c): E1603/E1702 - IKEA TRADFRI control outlet (Router)
Jan 27 19:44:18 raspberrypi npm[970]: zigbee2mqtt:info  2020-01-27 19:44:18: Relais_ewelink_box_1 (0x00124b001d436ada): SA-003-Zigbee - eWeLink Zigbee smart plug (Router)
Jan 27 19:44:18 raspberrypi npm[970]: zigbee2mqtt:info  2020-01-27 19:44:18: AqaraTempSensor01 (0x00158d000247cb11): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
**Jan 27 19:44:18 raspberrypi npm[970]: zigbee2mqtt:info  2020-01-27 19:44:18: 0x000b57fffea6ff65 (0x000b57fffea6ff65): LED1650R5 - IKEA TRADFRI LED bulb GU10 400 lumen, dimmable (Router)
Jan 27 19:44:18 raspberrypi npm[970]: zigbee2mqtt:info  2020-01-27 19:44:18: 0x0017880108413ae2 (0x0017880108413ae2): Not supported (Router)
Jan 27 19:44:18 raspberrypi npm[970]: zigbee2mqtt:info  2020-01-27 19:44:18: 0x00178801084140e0 (0x00178801084140e0): Not supported (Router)
Jan 27 19:44:18 raspberrypi npm[970]: zigbee2mqtt:info  2020-01-27 19:44:18: 0x00178801084c3ab2 (0x00178801084c3ab2): Not supported (Router)**
Jan 27 19:44:18 raspberrypi npm[970]: zigbee2mqtt:info  2020-01-27 19:44:18: Zigbee: disabling joining new devices.
Jan 27 19:44:18 raspberrypi npm[970]: zigbee2mqtt:info  2020-01-27 19:44:18: Connecting to MQTT server at mqtt://localhost
Jan 27 19:44:19 raspberrypi npm[970]: zigbee2mqtt:info  2020-01-27 19:44:19: Connected to MQTT server
Jan 27 19:44:19 raspberrypi npm[970]: zigbee2mqtt:info  2020-01-27 19:44:19: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Jan 27 19:44:19 raspberrypi npm[970]: zigbee2mqtt:info  2020-01-27 19:44:19: MQTT publish: topic 'zigbee2mqtt/AqaraTempSensor03', payload '{"temperature":19.99,"linkquality":26,"humidity":43.01,"pressure":1009,"battery":86,"volt [... cut ...]
Jan 27 19:44:38 raspberrypi npm[970]:     at tryOnTimeout (timers.js:300:5)
Jan 27 19:44:38 raspberrypi npm[970]:     at listOnTimeout (timers.js:263:5)
Jan 27 19:44:38 raspberrypi npm[970]:     at Timer.processTimers (timers.js:223:10)
Jan 27 19:44:38 raspberrypi npm[970]: (node:985) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 4)
```

Roeliooo, 2020-01-27:
Okay, great news. It looks like I fixed it. What I did;

- Reflash C2351 USB stick
- Run Zigbee2mqtt
- You are still getting errors and other shit (see post above)
- Just play with Permit_join set to False / True (via Mqtt server or config yaml file)
- After a while all devices will be active again, be patient.

Last two questions @Koenkk :
- Is it possible to set a timer or something else on Permit Join = True ?
- Can I remove the 4 spooky devices from de database file located in `/opt/zigbee2mqtt/data` without any problems?

9SMTM6, 2020-02-12:
zigbee2mqtt also keeps a record of paired devices in data/configuration.yaml/devices

Koenkk, 2020-02-12:
> Is it possible to set a timer or something else on Permit Join = True ?

not yet

> Can I remove the 4 spooky devices from de database file located in /opt/zigbee2mqtt/data without any problems?

Yes you can remove them manually from `database.db`

carldebilly, 2020-02-12:
@Roeliooo There's a feature in Zigbee2MqttAsssistant to turn it off automatically after a certain delay.

https://github.com/yllibed/Zigbee2MqttAssistant#settings (`AllowJoinTimeout`)

closed 2020-04-19 by stale[bot]