Koenkk/zigbee2mqtt #2832: Xiaomi Aqara Smart Door Lock (S1) - ZNMS11LM - lumi.lock.aq1

jestempablo, 2020-01-26:
Hello, I'm partially asking for support but other than that I hope someone will stumble upon this issue and maybe share their progress? Cos it's very hard to find any threads about this lock on the web.

[This is the lock in question](https://i.imgur.com/cIQG6DD.png)

Recently I've been reviewing list of my devices in HA Zigbee map and noticed I must have paired lock successfully (while I thought I didn't). After some digging, the only mention about this lock in terms of open sourced zigbee networks I've found is in this [Zigbee ioBroker issue](https://github.com/ioBroker/ioBroker.zigbee/issues/404). Thats how I've found the lock id. I've added it to devices.js by copying and renaming both `lumi.lock.v1` (Vima) and the `lumi.lock.acn02` (S2) lock config to `lumi.lock.aq1` and added extension for it (ZNMS11LM) to show on Home Assistant and ok, its there. The problem is that the states are mostly not populated and the device itself doesn't report anything on debug log of zigbee2mqtt. 
(Mostly not populated means: when running on Vima lock config copy, it populates one of four states: `"binary_sensor.lock_lock"` as **"Locked"** which is something, yeah? For comparison: S2 config - 2 states, both "unknown")

I'm not sure if `lumi.lock.aq1` is correct id, I currently assume yes if it showed in Home Assistant but I've might just forced it by assigning this id and device name ZNMS11LM in extensions.

"How to support new devices" page mentions that if device doesn't report anything it might need additional configuration, but no configuration exists for any of Xiaomi locks. I'm not sure how to progress. If anyone could give me a hint what I could try next, it would be awesome.

When connecting via Xiaomi Aqara Gateway I had access to attributes `number of failed entries`, `last action` (the id of user who did last open with success) and `state` of always locked so it must be somewhere, right? But where?

Thanks for your time!

dzungpv, 2020-02-04:
You could not use other lock code for your lock, because it may have differece packet. Start over and following step by step in "How to support new devices" and post here for help.

jestempablo, 2020-02-04:
Sorry for being not clear, but apart from my findings which I shared, I've
also followed the guide step by step which resulted in no messages being
sent by lock at all.

closed 2020-04-11 by stale[bot]

mikeymagic, 2025-02-08:
I've had my znms11lm installed for a while but have never managed to get any status from it or control it from Home Assistant. Has anyone had any success?