Koenkk/zigbee2mqtt #2836: Aqara relay controller - LLKZMK11LM - different device information

trongdong, 2020-01-27:
Hi there, 
First of all, thank you for such a great software.  
Not sure if this is a bug but I would like to ask anyway!  
I have a bunch of Aqara relay controller (LLKZMK11LM) in my network, all of them are wired and connect with DC source (for sure).  
I've noticed that when trying to get device list, they appear different like below:  
```
  {
    "ieeeAddr": "0x00158d00044be07d",
    "type": "Router",
    "networkAddress": 9318,
    "model": "LLKZMK11LM",
    "friendly_name": "0x00158d00044be07d",
    "manufacturerID": 4151,
    "manufacturerName": "LUMI",
    "powerSource": "DC Source",
    "modelID": "lumi.relay.c2acn01",
    "hardwareVersion": 18,
    "dateCode": "09-20-2018",
    "lastSeen": 1580054120265
  },
  {
    "ieeeAddr": "0x00158d00044ad8dd",
    "type": "EndDevice",
    "networkAddress": 48953,
    "model": "LLKZMK11LM",
    "friendly_name": "0x00158d00044ad8dd",
    "manufacturerID": 4151,
    "manufacturerName": "LUMI",
    "powerSource": "Battery",
    "modelID": "lumi.relay.c2acn01",
    "lastSeen": 1580054141474
  },
```
All of them are working flawlessly!

Koenkk, 2020-01-27:
These differences probably occur because the interview procedure didn't fully complete for the `LLKZMK11LM`, for one it did, for one it didn't (0x00158d00044ad8dd). Interviewing of Xiaomi is quite hard as these device fall asleep early.

This is only a cosmetic issue (doesn't affect functionality), if you want to fix it you can try re-pairing `0x00158d00044ad8dd`.

trongdong, 2020-01-28:
Thanks @Koenkk for your information.  
I also noticed that the one that didn't fully complete the interview procedure also got some weird behavior right after paired. Ex: I can only control the left channel but not the right one. It tooks around 30 minutes for me to start controlling the second channel successfully.
Closing issue for now!

closed 2020-01-28 by trongdong