Koenkk/zigbee2mqtt #2837: Converting zigbee illuminance value to lux

Erelen-Laiquendi, 2020-01-27:
According to [Zigbee doc](https://web.archive.org/web/20190825040942/http://www.zigbee.org/wp-content/uploads/2014/11/docs-07-5123-04-zigbee-cluster-library-specification.pdf) (4.2.2.2.1.1 and 4.3.2.4.1) and Wireshark sources (wireshark-3.2.1/epan/dissectors/packet-zbee-zcl-meas-sensing.c, line 192).

> lx = 10^(value/10000)-1

So, if sensor return 15'000 - real illuminance is 30.6 lx.

Please, implement this for [MiJia Light intensity sensor (GZCGQ01LM)](https://github.com/Koenkk/zigbee2mqtt/issues/2702)

Erelen-Laiquendi, 2020-01-27:
<details>
  <summary>100 lowest values of MiJia Light intensity sensor</summary>

```sql
SELECT DISTINCT
  CAST(state AS INT) AS `value`,
  ROUND(POWER(10, CAST(state AS INT)/10000)-1, 5) AS lx
FROM
  states
WHERE entity_id = 'sensor.kitchen_illuminance'
ORDER BY `value`
LIMIT 100;
```

| value | lx |
| ------------- | ------------- |
| 0 | **0.0** 0000 |
| 3011 | **1.0** 0032 |
| 4772 | **2.0** 0054 |
| 6021 | **3.0** 0037 |
| 6990 | **4.0** 0035 |
| 7782 | **5.0** 0067 |
| 8451 | **6.0** 0003 |
| 9031 | **7.0** 0018 |
| 9543 | **8.0** 0119 |
| 10001 | **9.0** 0230 |
| 10414 | **10.0** 0019 |
| 10792 | **11.0** 0052 |
| 11140 | **12.0** 0170 |
| 11462 | **13.0** 0232 |
| 11761 | **14.0** 0030 |
| 12042 | **15.0** 0295 |
| 12305 | **16.0** 0200 |
| 12553 | **17.0** 0114 |
| 12788 | **18.0** 0203 |
| 13011 | **19.0** 0322 |
| 13223 | **20.0** 0390 |
| 13425 | **21.0** 0392 |
| 13618 | **22.0** 0382 |
| 13803 | **23.0** 0491 |
| 13980 | **24.0** 0345 |
| 14150 | **25.0** 0160 |
| 14314 | **26.0** 0225 |
| 14472 | **27.0** 0271 |
| 14624 | **28.0** 0013 |
| 14772 | **29.0** 0544 |
| 14914 | **30.0** 0273 |
| 15052 | **31.0** 0369 |
| 15186 | **32.0** 0654 |
| 15315 | **33.0** 0165 |
| 15441 | **34.0** 0258 |
| 15564 | **35.0** 0808 |
| 15683 | **36.0** 0837 |
| 15798 | **37.0** 0144 |
| 15911 | **38.0** 0318 |
| 16021 | **39.0** 0369 |
| 16128 | **40.0** 0152 |
| 16233 | **41.0** 0490 |
| 16335 | **42.0** 0312 |
| 16435 | **43.0** 0479 |
| 16533 | **44.0** 0907 |
| 16628 | **45.0** 0447 |
| 16721 | **46.0** 0023 |
| 16813 | **47.0** 0650 |
| 16902 | **48.0** 0044 |
| 16990 | **49.0** 0345 |
| 17076 | **50.0** 0350 |
| 17161 | **51.0** 1157 |
| 17243 | **52.0** 0294 |
| 17324 | **53.0** 0078 |
| 17404 | **54.0** 0473 |
| 17482 | **55.0** 0154 |
| 17559 | **56.0** 0330 |
| 17635 | **57.0** 0962 |
| 17709 | **58.0** 0652 |
| 17782 | **59.0** 0674 |
| 17854 | **60.0** 0986 |
| 17924 | **61.0** 0119 |
| 17994 | **62.0** 0862 |
| 18062 | **63.0** 0295 |
| 18130 | **64.0** 1297 |
| 18196 | **65.0** 0852 |
| 18261 | **66.0** 0389 |
| 18326 | **67.0** 1426 |
| 18389 | **68.0** 0809 |
| 18451 | **69.0** 0032 |
| 18513 | **70.0** 0681 |
| 18574 | **71.0** 1119 |
| 18634 | **72.0** 1297 |
| 18693 | **73.0** 1164 |
| 18751 | **74.0** 0669 |
| 18809 | **75.0** 1512 |
| 18865 | **76.0** 0164 |
| 18921 | **77.0** 0097 |
| 18977 | **78.0** 1326 |
| 19031 | **79.0** 0184 |
| 19085 | **80.0** 0279 |
| 19139 | **81.0** 1627 |
| 19191 | **82.0** 0419 |
| 19243 | **83.0** 0401 |
| 19295 | **84.0** 1587 |
| 19345 | **85.0** 0031 |
| 19396 | **86.0** 1618 |
| 19445 | **87.0** 0351 |
| 19494 | **88.0** 0205 |
| 19543 | **89.0** 1191 |
| 19591 | **90.0** 1228 |
| 19638 | **91.0** 0258 |
| 19685 | **92.0** 0365 |
| 19732 | **93.0** 1562 |
| 19778 | **94.0** 1671 |
| 19823 | **95.0** 0636 |
| 19868 | **96.0** 0631 |
| 19913 | **97.0** 1668 |
| 19957 | **98.0** 1477 |
| 20001 | **99.0** 2303 |
</details>

As you can see, internally sensor works with integer lux value.
So, after converting to lux, we can round result to integer.

Koenkk, 2020-01-27:
This can be implemented here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/fromZigbee.js#L955

How to get to `formZigbee.js`: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

Can you give it a shot?

Koenkk, 2020-01-29:
Merged, thanks!

closed 2020-01-29 by Koenkk