Koenkk/zigbee2mqtt #2838: TRADFRI LED bulb (LED1546G12) - 'MAC no ack' (233 )

xefil, 2020-01-27:
# Bug Report
## What happened

After moving from a connected CC2531 USB Stick to a CC2530+CC2591 remote via ESP8266, I've a problem ONLY with this bulb. 
I was able to pair it the first time, but, then, moved to his location, stopped to work.
If I turn it on, it's announgin itself. Then I've a routine (via home assistant) that turns the light on as soon a sensor is triggered (pir). There the log:

```
debug 2020-01-27 14:09:16: Device '0x000d6ffffe17ea41' announced itself
debug 2020-01-27 14:12:48: Received MQTT message on 'XefilHome/0x000d6ffffe17ea41/set' with data '{"state": "OFF"}'
debug 2020-01-27 14:12:48: Publishing 'set' 'state' to '0x000d6ffffe17ea41'
error 2020-01-27 14:12:48: Publish 'set' 'state' to '0x000d6ffffe17ea41' failed: 'Error: Data request failed with error: 'MAC no ack' (233)'
info  2020-01-27 14:12:48: MQTT publish: topic 'XefilHome/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to '0x000d6ffffe17ea41' failed: 'Error: Data request failed with error: 'MAC no ack' (233)'","meta":{"friendly_name":"0x000d6ffffe17ea41"}}'
```

## What did you expect to happen

That it works as previously.

## How to reproduce it (minimal and precise)

Simply turning the light on. Tried different time the re-pairing, but I've to be near to the antenna. After paired, I need to turn itoff and place it to the end point. Then it starts over again with the issues.

## Debug Info
zigbee2mqtt version: zigbee2mqtt version 1.9.0 (commit #287f843) - zigbee-herdsman
CC253X firmware version:
CC2530+CC2591 = CC2530_CC2591_DEFAULT_20190608 (CC2530ZNP-Prod) [1.2 afaik]

Seems that something is not working as expected only with this antenna, possibile?
Any way to debug?

Here my conf:

```
homeassistant: true
permit_join: true
mqtt:
  base_topic: XefilHome
  server: 'mqtt://192.168.1.13'
  user: <<CUT>>
  password: <<CUT>>
  client_id: zigbee2mqtt
serial:
  port: 'tcp://192.168.3.23:1775'
advanced:
  rtscts: false
#  report: true
  homeassistant_discovery_topic: XefilHome
  network_key:
<<CUT>>
  log_level: debug
map_options:
  graphviz:
    colors:
      fill:
        enddevice: '#fff8ce'
        coordinator: '#e04e5d'
        router: '#4ea3e0'
      font:
        coordinator: '#ffffff'
        router: '#ffffff'
        enddevice: '#000000'
      line:
        active: '#009900'
        inactive: '#994444'
devices:
<<CUT>>
  '0x000d6ffffe17ea41':
    friendly_name: '0x000d6ffffe17ea41'
```

odx, 2020-01-31:
I am currently struggling with a similar issue that appeared after moving to CC1352P-2: #2789
So far I have no clear idea on how to proceed in debugging. As I have another bulb next to it that works perfectly.

xefil, 2020-02-05:
So actually no solution... seems so strange that one device can handle it (CC2531) and the other not :-(

Koenkk, 2020-02-05:
> Simply turning the light on. Tried different time the re-pairing, but I've to be near to the antenna. After paired, I need to turn itoff and place it to the end point. Then it starts over again with the issues.

Can you control it when having it close to the antenna?

xefil, 2020-02-23:
Hello @Koenkk !
Yes, this morning, again, tested. Fresh removed from configuration/database and status, the restart the service and repaired succesfully (after some tries). Then, Until it's close on the antenna, it works.
Yesterday I've added two CC2531 as routers in the house, btw actually I've no idea how to check if they are used for traffic. BTW, about the light bulb, always the same.
Afer placed into end posizion I can see:
` Device '0x000d6ffffe17ea41' announced itself`
But after, nothing. If I try to turn it off via HomeAssistant (Mqtt):

```
debug 2020-02-23 18:42:16: Received MQTT message on 'XefilHome/0x000d6ffffe17ea41/set' with data '{"state": "OFF"}'
debug 2020-02-23 18:42:16: Publishing 'set' 'state' to '0x000d6ffffe17ea41'
error 2020-02-23 18:42:16: Publish 'set' 'state' to '0x000d6ffffe17ea41' failed: 'Error: Command 0x000d6ffffe17ea41/1 genOnOff.off({}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'MAC no ack' (233))'
debug 2020-02-23 18:42:16: Error: Command 0x000d6ffffe17ea41/1 genOnOff.off({}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'MAC no ack' (233))
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:347:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:94:5)
info  2020-02-23 18:42:16: MQTT publish: topic 'XefilHome/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to '0x000d6ffffe17ea41' failed: 'Error: Command 0x000d6ffffe17ea41/1 genOnOff.off({}, {\"timeout\":10000,\"defaultResponseTimeout\":15000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Data request failed with error: 'MAC no ack' (233))'","meta":{"friendly_name":"0x000d6ffffe17ea41"}}'
```

Ideas?

Thanks!

Koenkk, 2020-02-23:
@xefil I've pushed a possible fix to the latest dev branch, please try and let me know.

xefil, 2020-02-23:
Hello!
I've the setup on a Synology NAS via Docker. I've to download the latest-dev, right?
I need to understand how to put the content of the latest-dev over the latest setup ;-)

xefil, 2020-02-23:
Hi @Koenkk ,
The same, trying to power it off:

```
debug 2020-02-23 23:26:30: Received MQTT message on 'XefilHome/0x000d6ffffe17ea41/set' with data '{"state": "OFF"}'
debug 2020-02-23 23:26:30: Publishing 'set' 'state' to '0x000d6ffffe17ea41'
error 2020-02-23 23:26:34: Publish 'set' 'state' to '0x000d6ffffe17ea41' failed: 'Error: Command 0x000d6ffffe17ea41/1 genOnOff.off({}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'MAC no ack' (233))'
debug 2020-02-23 23:26:34: Error: Command 0x000d6ffffe17ea41/1 genOnOff.off({}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'MAC no ack' (233))
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:344:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:97:5)
info  2020-02-23 23:26:34: MQTT publish: topic 'XefilHome/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to '0x000d6ffffe17ea41' failed: 'Error: Command 0x000d6ffffe17ea41/1 genOnOff.off({}, {\"timeout\":6000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Data request failed with error: 'MAC no ack' (233))'","meta":{"friendly_name":"0x000d6ffffe17ea41"}}'
```

Should I try to delete the pair, repair it near to the main antenna and the move it back to the end place? That should be the same as now btw having it already registered before.
Thanks!

Simon

Koenkk, 2020-02-24:
@xefil can you provide the log when running with herdsman debug logging? https://www.zigbee2mqtt.io/information/debug.html#docker

xefil, 2020-02-24:
Hello @Koenkk ,
Having the docker on a NAS container, I'm not able so fast to add the debug property. Have added it as ENVIRONMENT due I've not found where to add the `"-e DEBUG=*" parameter, neither have idea how to check from the command line (ssh) how the container was executed.
Actually I see only:

```
# docker ps --no-trunc -a
koenkk/zigbee2mqtt:latest-dev             "./run.sh"                                                                        15 hours ago        Up 4 minutes                                       zigbee2mqtt-dev
```

If the `-e DEBUG=*` is a parameter then it would be great to take this value from the Environment. Passing the parameter, seems I need to export the settings, delete the container and add it again:

https://community.synology.com/enu/forum/17/post/115537?reply=378518

But if there is a way to do it via command line, glad to test it.

Ideas?

Menawhile I'm searching it

xefil, 2020-02-24:
Hi again, ok, the -e *is* an environment variable, so seems added correctly, btw I cannot find nothing interesting. Here the env:
```
# docker inspect zigbee2mqtt-dev
(...)
            "Env": [
                "DEBUG=*",
                "PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin",
                "NODE_VERSION=12.16.1",
                "YARN_VERSION=1.22.0",
                "TZ=Europe/Rome"
            ],
```
BTW the logs are showing the same if I try to turn the light on via MQTT:
```
debug 2020-02-24 14:58:39: Received MQTT message on 'XefilHome/0x000d6ffffe17ea41/set' with data '{"state": "OFF"}'
debug 2020-02-24 14:58:39: Publishing 'set' 'state' to '0x000d6ffffe17ea41'
error 2020-02-24 14:58:43: Publish 'set' 'state' to '0x000d6ffffe17ea41' failed: 'Error: Command 0x000d6ffffe17ea41/1 genOnOff.off({}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'MAC no ack' (233))'
debug 2020-02-24 14:58:43: Error: Command 0x000d6ffffe17ea41/1 genOnOff.off({}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'MAC no ack' (233))
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:344:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:97:5)
info  2020-02-24 14:58:43: MQTT publish: topic 'XefilHome/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to '0x000d6ffffe17ea41' failed: 'Error: Command 0x000d6ffffe17ea41/1 genOnOff.off({}, {\"timeout\":6000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Data request failed with error: 'MAC no ack' (233))'","meta":{"friendly_name":"0x000d6ffffe17ea41"}}'
```

If I should try simething else, let meknow!
Thanks!

Simon

Koenkk, 2020-02-24:
An alternative would be to sniff the traffic when executing the on/off commands (may even give some more information): https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

xefil, 2020-02-24:
@Koenkk Would it make sense re-register the bulb?

I don't think so, btw, I've a CC2531 as spare I could attach on my laptop and sniff some traffic, then let you know.
Thanks, Simon

xefil, 2020-02-28:
Hello @Koenkk , I've moved back to latest, not dev, actually on version 1.11.0 (commit #31e5678). Is this version including the fix you've pushed days ago on dev?
On this release, I've again the problems on `Failed to execute LQI`, don't know if related to this version.
Thanks,
Simon

Koenkk, 2020-02-28:
@xefil it's also included there, for LQI there is already an issue open: #2802.

closed 2020-05-06 by stale[bot]