Koenkk/zigbee2mqtt #2839: Error while starting zigbee-herdsman

Rohmilchkaese, 2020-01-27:
# Bug Report
Run Zigbee2mqtt for a while, wait till it fails.
## What happened
zigbee2mqtt:error 2020-01-27 19:13:11: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-01-27 19:13:11: Failed to start zigbee
zigbee2mqtt:error 2020-01-27 19:13:11: Exiting...
zigbee2mqtt:error 2020-01-27 19:13:11: Error: SRSP - SYS - version after 6000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at listOnTimeout (internal/timers.js:531:17)
    at processTimers (internal/timers.js:475:7)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.9.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.9.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-01-27T18_13_46_723Z-debug.log
Using '/app/data' as data directory
## What did you expect to happen
Just start zigbee2mqtt

## How to reproduce it (minimal and precise)
Start with a clean install and run for while. I connected like 10 devices and then it started to fail.

## Debug Info
zigbee2mqtt version:
1.9 Docker
CC253X firmware version:
Sorry I don't know, how can I find out ? 

I run a CC2531.

Koenkk, 2020-01-27:
Try flashing the source routing firmware, this should improve stability: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

Rohmilchkaese, 2020-01-29:
@Koenkk Thank you, will try.

closed 2020-04-02 by Rohmilchkaese