Koenkk/zigbee2mqtt #2840: Upcoming Philips Hue lights/plug firmware update - reset with light switch

Kryzek, 2020-01-28:
From https://www2.meethue.com/en-gb/support/release-notes/lamps-luminaires
**January 23, 2020**
Software version: 1.65.7_hCDE64320

This update, available for all Philips Hue with Bluetooth lights and the Philips Hue smart plug, provides improvements to stability and security, as well as the ability to:
-     Perform a manual factory reset of lights by power cycling a light 5 times

Time to harness Hue Bridge to push updates to some bulbs then... Or even better, when we eventually get to enjoy https://github.com/Koenkk/zigbee-herdsman/issues/60 :)

sjorge, 2020-01-28:
Do we have a way to get to the hue firmware images? if not, we will probably still need a bridge even with Koenkk/zigbee-herdsman#60. But it is an interesting update none the less.

Kryzek, 2020-01-28:
Well there is some information regarding Hue firmware updates (https://github.com/dresden-elektronik/deconz-rest-plugin/issues/270#issuecomment-483386143) on deconz side...

closed 2020-04-04 by stale[bot]