Koenkk/zigbee2mqtt #2841: Niloé / Nereya switches x zigbee2mqtt

wagiotto, 2020-01-28:
Hello everyone.

I have legrand niloé / nereya switches and I can't connect them with the home assistant as they only communicate with each other.
With this zigbee2mqtt solution is it possible for me to be able to connect them with the raspberry pi for control via cell phone?
I leave here a link to the switches.

Please help me!

http://www.decorwatts.com.br/interruptor-zigbee.php

https://www.amazon.co.uk/Legrand-Niloé-Shutter-Switch-Zigbee®/dp/B00BD0EOSA

sjorge, 2020-01-28:
The legrand stuff might already work: https://github.com/Koenkk/zigbee2mqtt/issues/2399

wagiotto, 2020-01-28:
But my switches are of different models than these. They have no integration with the cell phone.

closed 2020-04-04 by stale[bot]