Koenkk/zigbee2mqtt #2842: Xiaomi device (WXKG01LM) binding support

DhaenensT, 2020-01-28:
Thank you Koenk for this great project.

Im am trying to bind the WXKG01LM Xiaomi switch to an ikea light bulb by using the following guide: https://www.zigbee2mqtt.io/information/binding.html
Nothing is happening in de log output and I am unable to control de ikea light bulb. 
I was able to bind the Ikea switch (E1743) to the light bulb (LED1733G7)
Does the Xiaomi switch have an GenOnOff cluster to bind to the ikea bulb? 

Thanks!

Koenkk, 2020-01-28:
The WXKG01LM cannot be bound (not supported by the device).

closed 2020-01-28 by Koenkk

rs-11, 2020-05-12:
Hi Koenkk, I recommend you write that on the product page at [link].(https://www.zigbee2mqtt.io/devices/WXKG01LM.html )
Thanks!

Koenkk, 2020-05-13:
Done, thanks

Se7enair, 2021-12-05:
How is ist possible that deconz is able to bind this device to a group?
I recently switched from deconz to z2m and was wondering  why binding is not possible.

Is deconz using a „fictional“ binding?