Koenkk/zigbee2mqtt #2843: Adding new devices needs reflashing?

kpmkpm, 2020-01-28:
Hello and first of all thanks for your awesome work.

I was wondering is normal behavior and a feature that I cannot add new devices if there are days between tries after pairing them? It has happened now twice. Say, I pair couple of devices today and after two weeks I need to pair other devices but nothing happens until I reflash the stick and after that I am able to pair the old and new devices with no problem at all. 

The paired device work ok, but cannot pair new ones until reflashing (the same) firmware.

What am I doing wrong?

Thank you in advance.

# Bug Report
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version:
CC253X firmware version:

Koenkk, 2020-01-29:
What firmware are you using?

kpmkpm, 2020-01-29:
CC2531_DEFAULT_20190608

Koenkk, 2020-01-29:
How many devices do you have in your network and what type of devices? Will be logged on zigbee2mqtt startup.

kpmkpm, 2020-01-29:
Thank you for your quick reply.

It has happened twice. First time I only tried with couple of Ikea motion sensors and remotes. After couple of weeks I tried to add two bulbs and couple or remotes more. Only after the reflash the new devices were found. Everything else worked though. Tried to stop and start and even reboot the server with no luck.

Today I tried to add new motion sensors and bulbs again and the same happened. The new devices were found after a reflash. I now have four trådfri bulbs, 3 Motion sensors, 4 Ikea remotes and 3 onoff-switches. I use hue bridge for most of the bulbs and these four are only to extend the zigbee2mqtt mesh.

After a reflash it is ok now, but I am awaiting for some Aqara sensors to arrive and it would be nice if I will be able to pair them without a reflash.

I have only used z2mqtt for about six weeks. I have also ordered a new stick with antenna and another stick without as this first one that I am using now. Perhaps I will flash the two as routers and the one with antenna as a coordinator, or is it better to leave the two as spare ones and use bulbs as coordinator instead?

Thank you again.

Koenkk, 2020-01-30:
Tradfri bulbs don't have really strong attenas so the extra routers should be fine.

Does stopping zigbee2mqtt, replugging the CC2531 USB and starting zigbee2mqtt fix the pairing issue?

kpmkpm, 2020-01-30:
Thanks, I will try that next time if the problem reoccurs.

MonsieurSander, 2020-02-15:
> Tradfri bulbs don't have really strong attenas so the extra routers should be fine.
> 
> Does stopping zigbee2mqtt, replugging the CC2531 USB and starting zigbee2mqtt fix the pairing issue?

Had the same issue, tried this and the problem was resolved. Thank you!

closed 2020-04-25 by stale[bot]