Koenkk/zigbee2mqtt #2844: Xiaomi WXCJKG11LM - interview failed

bogdaan2001, 2020-01-28:
Hi, i try install firmware Z-Stack 1.2 and Z-Stack 3.0 on cc2531 stick but i can't add the switch "aqara opple"  WXCJKG11LM  to mosquitto broker (get error interview_failed when try connect switch).
Can you please say where i did mistake ?

Other devices work correctly

sh4jonny, 2020-01-28:
hi! i have a same problem. Z-Stack 3.0 even not start to try pairing ? only 1.2 try to pair for me but it also fail. i write about here #2827

Koenkk, 2020-01-29:
I've just tested this with my WXCJKG12LM and it successfully interviewed on the latest zigbee2mqtt dev branch.

Please try with the latest zigbee2mqtt dev branch, if it still fails please provide the herdsman debug logging.

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

bogdaan2001, 2020-01-29:
> I've just tested this with my WXCJKG12LM and it successfully interviewed on the latest zigbee2mqtt dev branch.
>
> Please try with the latest zigbee2mqtt dev branch, if it still fails please provide [...]

Hi!
What is herdsman  ???
i use:
- raspberry pi 4
- zigbee2mqtt
- mosquito
- cc2531 stick with firmware https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.0.x/bin/CC2531_20190425.zip

Koenkk, 2020-01-29:
The underlying zigbee library, please run with instructions from https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

bogdaan2001, 2020-01-29:
> The underlying zigbee library, please run with instructions from https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

sudo systemctl start zigbee2mqtt DEBUG=zigbee-herdsman* 

Invalid unit name "DEBUG=zigbee-herdsman*" was escaped as "DEBUG\x3dzigbee-herdsman*" (maybe you should use systemd-escape?)

Koenkk, 2020-01-30:
Don't run via systemctl, stop zigbee2mqtt first and then start it with:

```
cd /opt/zigbee2mqtt
DEBUG=zigbee-herdsman:controller* npm start
```

bogdaan2001, 2020-01-30:
Hi!

zigbee2mqtt:info  2020-01-30 21:57:06: Connecting to MQTT server at mqtt://localhost
zigbee2mqtt:info  2020-01-30 21:57:06: Connected to MQTT server
zigbee2mqtt:info  2020-01-30 21:57:06: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
zigbee2mqtt:info  2020-01-30 21:57:06: MQTT publish: topic 'zigbee2mqtt/temp1', payload '{"battery":80,"voltage":2965,"temperature":22.13,"humidity":37.46,"linkquality":128}'
zigbee2mqtt:info  2020-01-30 21:57:06: MQTT publish: topic 'zigbee2mqtt/gerkon1', payload '{"battery":100,"voltage":3005,"linkquality":78,"contact":true}'
zigbee2mqtt:info  2020-01-30 21:57:06: MQTT publish: topic 'zigbee2mqtt/0x00158d00044d8f30', payload '{"battery":100,"voltage":3045,"linkquality":89}'
zigbee2mqtt:info  2020-01-30 21:57:06: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.9.0","commit":"287f843","coordinator":{"type":"zStack30x","meta":{"transportrev":2,"product":2,"majorrel":2,"minorrel":7,"maintrel":2,"revision":20190425}},"log_level":"debug","permit_join":true}'
  zigbee-herdsman:controller:log Device leave '0x04cf8cdf3c75c994' +16s
  zigbee-herdsman:controller:log Removing device from database '0x04cf8cdf3c75c994' +0ms
zigbee2mqtt:warn  2020-01-30 21:57:22: Device '0x04cf8cdf3c75c994' left the network
zigbee2mqtt:info  2020-01-30 21:57:22: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"left_network","meta":{"friendly_name":"0x04cf8cdf3c75c994"}}'
  zigbee-herdsman:controller:log Device '0x04cf8cdf3c75c994' joined +6s
  zigbee-herdsman:controller:log Device '0x04cf8cdf3c75c994' accepted by handler +1ms
  zigbee-herdsman:controller:log New device '0x04cf8cdf3c75c994' joined +1ms
  zigbee-herdsman:controller:log Creating device '0x04cf8cdf3c75c994' +0ms
zigbee2mqtt:info  2020-01-30 21:57:28: Device 'switch1' joined
zigbee2mqtt:info  2020-01-30 21:57:28: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"switch1"}}'
  zigbee-herdsman:controller:log Interview '0x04cf8cdf3c75c994' start +15ms
zigbee2mqtt:info  2020-01-30 21:57:28: Starting interview of 'switch1'
zigbee2mqtt:info  2020-01-30 21:57:28: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"switch1"}}'
  zigbee-herdsman:controller:device Interview - start device '0x04cf8cdf3c75c994' +0ms
  zigbee-herdsman:controller:log Device announce '0x04cf8cdf3c75c994' +784ms
zigbee2mqtt:debug 2020-01-30 21:57:29: Device 'switch1' announced itself
  zigbee-herdsman:controller:device Interview - got node descriptor for device '0x04cf8cdf3c75c994' +1s
  zigbee-herdsman:controller:device Interview - got active endpoints for device '0x04cf8cdf3c75c994' +981ms
  zigbee-herdsman:controller:device Interview - got simple descriptor for endpoint '1' device '0x04cf8cdf3c75c994' +967ms
  zigbee-herdsman:controller:device Interview - got simple descriptor for endpoint '2' device '0x04cf8cdf3c75c994' +981ms
  zigbee-herdsman:controller:device Interview - got simple descriptor for endpoint '3' device '0x04cf8cdf3c75c994' +985ms
  zigbee-herdsman:controller:device Interview - failed for device '0x04cf8cdf3c75c994' with error 'Error: AREQ - ZDO - simpleDescRsp after 10000ms
  zigbee-herdsman:controller:device     at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
  zigbee-herdsman:controller:device     at ontimeout (timers.js:436:11)
  zigbee-herdsman:controller:device     at tryOnTimeout (timers.js:300:5)
  zigbee-herdsman:controller:device     at listOnTimeout (timers.js:263:5)
  zigbee-herdsman:controller:device     at Timer.processTimers (timers.js:223:10)' +10s
  zigbee-herdsman:controller:error Interview failed for '0x04cf8cdf3c75c994 with error 'Error: AREQ - ZDO - simpleDescRsp after 10000ms' +0ms
zigbee2mqtt:error 2020-01-30 21:57:43: Failed to interview 'switch1', device has not successfully been paired
zigbee2mqtt:info  2020-01-30 21:57:43: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"switch1"}}'

Koenkk, 2020-01-30:
Thanks, I need a little bit more logging, can you run with

`DEBUG=zigbee-herdsman:* npm start`

bogdaan2001, 2020-01-30:
pi@raspberrypi:/opt/zigbee2mqtt $ DEBUG=zigbee-herdsman:* npm start

> zigbee2mqtt@1.9.0 start /opt/zigbee2mqtt
> node index.js

zigbee2mqtt:info  2020-01-30 22:21:58: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-01-30.22-21-58'
zigbee2mqtt:debug 2020-01-30 22:21:58: Removing old log directory '/opt/zigbee2mqtt/data/log/2020-01-28.23-46-36'
zigbee2mqtt:debug 2020-01-30 22:21:58: Loaded state from file /opt/zigbee2mqtt/data/state.json
zigbee2mqtt:info  2020-01-30 22:21:58: Starting zigbee2mqtt version 1.9.0 (commit #287f843)
zigbee2mqtt:info  2020-01-30 22:21:58: Starting zigbee-herdsman...
zigbee2mqtt:debug 2020-01-30 22:21:58: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"}}'
  zigbee-herdsman:adapter Path '/dev/ttyACM0' is valid for 'ZStackAdapter' +0ms
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json"}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyACM0 and {"baudRate":115200,"rtscts":true,"autoOpen":false} +0ms
  zigbee-herdsman:adapter:zStack:znp:log Serialport opened +7ms
  zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload +1ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239] +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - version - {} +0ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,2,35] +1s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,14,97,2,2,2,2,7,2,217,20,52,1,2,0,0,0,0,144] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,14,97,2,2,2,2,7,2,217,20,52,1,2,0,0,0,0,144] +2ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 14 - 3 - 1 - 2 - [2,2,2,7,2,217,20,52,1,2,0,0,0,0] - 144 +2ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - version - {"transportrev":2,"product":2,"majorrel":2,"minorrel":7,"maintrel":2,"revision":20190425} +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +9ms
  zigbee-herdsman:adapter:zStack:adapter Detected znp version 'zStack30x' ({"transportrev":2,"product":2,"majorrel":2,"minorrel":7,"maintrel":2,"revision":20190425}) +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":96,"offset":0,"len":1,"value":{"type":"Buffer","data":[85]}} +31ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,96,0,0,74] +30ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,97,8,0,1,85,62] +34ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,97,8,0,1,85,62] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 3 - 1 - 8 - [0,1,85] - 62 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvRead - {"status":0,"len":1,"value":{"type":"Buffer","data":[85]}} +40ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
  zigbee-herdsman:adapter:zStack:startZnp Item 'hasConfigured' is valid +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":96,"offset":0,"len":1,"value":{"type":"Buffer","data":[85]}} +34ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,96,0,0,74] +31ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,97,8,0,1,85,62] +27ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,97,8,0,1,85,62] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 3 - 1 - 8 - [0,1,85] - 62 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvRead - {"status":0,"len":1,"value":{"type":"Buffer","data":[85]}} +30ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
  zigbee-herdsman:adapter:zStack:startZnp Item 'hasConfigured' is valid +30ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":132,"len":4,"offset":0,"value":{"type":"Buffer","data":[0,8,0,0]}} +32ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,132,0,0,174] +33ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,6,97,8,0,4,0,8,0,0,99] +20ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,6,97,8,0,4,0,8,0,0,99] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 6 - 3 - 1 - 8 - [0,4,0,8,0,0] - 99 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvRead - {"status":0,"len":4,"value":{"type":"Buffer","data":[0,8,0,0]}} +22ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:startZnp Item 'channelList' is valid +20ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":99,"len":1,"offset":0,"value":{"type":"Buffer","data":[0]}} +17ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,99,0,0,73] +16ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,97,8,0,1,0,107] +18ms
[... cut ...]
  zigbee-herdsman:controller:device     at Timer.processTimers (timers.js:223:10)' +10s
  zigbee-herdsman:controller:error Interview failed for '0x04cf8cdf3c75c994 with error 'Error: AREQ - ZDO - simpleDescRsp after 10000ms' +0ms
zigbee2mqtt:error 2020-01-30 22:22:31: Failed to interview 'switch1', device has not successfully been paired
zigbee2mqtt:info  2020-01-30 22:22:31: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"switch1"}}'

bogdaan2001, 2020-01-31:
no idea ? :(

Koenkk, 2020-01-31:
I was just looking at this.

Please don't post logs here anymore but instead on pastebin.com and link it here.

Try:
- Please try repairing it once more with the debug logging (like you did in: https://github.com/Koenkk/zigbee2mqtt/issues/2844#issuecomment-580414196). I want to see if it fails at the same point again.
- Stop zigbee2mqtt
- Replace `/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/device.js` with: https://gist.github.com/Koenkk/131ee3e20fdabb6fa094a07f5c247083
- Try repairing it again and provide the debug logging.

bogdaan2001, 2020-01-31:
ok

bogdaan2001, 2020-01-31:
https://pastebin.com/26C8ykXi

Koenkk, 2020-01-31:
I've made a little mistake, please try again with: https://gist.github.com/Koenkk/131ee3e20fdabb6fa094a07f5c247083

bogdaan2001, 2020-01-31:
https://pastebin.com/AaWuEpVt

Koenkk, 2020-01-31:
Looks like you are still using the old `device.js`, make sure to update to the updated https://gist.github.com/Koenkk/131ee3e20fdabb6fa094a07f5c247083

bogdaan2001, 2020-01-31:
Now, i get this error :(

pi@raspberrypi:/opt/zigbee2mqtt $ DEBUG=zigbee-herdsman:* npm start

> zigbee2mqtt@1.9.0 start /opt/zigbee2mqtt
> node index.js

zigbee2mqtt:info  2020-02-01 01:33:25: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-02-01.01-33-25'
zigbee2mqtt:debug 2020-02-01 01:33:25: Removing old log directory '/opt/zigbee2mqtt/data/log/2020-01-28.23-58-13'
/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/device.js:387
    powerSource: { key: 'powerSource', set: (v, d) => { d.powerSource = v; } },
                                          ^

SyntaxError: Unexpected token :
    at Module._compile (internal/modules/cjs/loader.js:723:23)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
    at Module.require (internal/modules/cjs/loader.js:692:17)
    at require (internal/modules/cjs/helpers.js:25:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/index.js:6:34)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.9.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.9.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-01-31T22_33_25_562Z-debug.log

Koenkk, 2020-02-01:
Sorry made anther mistake: https://gist.github.com/Koenkk/131ee3e20fdabb6fa094a07f5c247083

bogdaan2001, 2020-02-01:
Hi!

https://pastebin.com/KriMZhmp

Koenkk, 2020-02-01:
Can you try with https://gist.github.com/Koenkk/131ee3e20fdabb6fa094a07f5c247083

Btw, is this the WXCJKG1**1**LM (1 band, 2 buttons) or the WXCJKG1**3**LM (3 bands, 6 buttons)?

bogdaan2001, 2020-02-01:
This is  WXCJKG11LM (1 band, 2 buttons)
[image]

bogdaan2001, 2020-02-01:
https://pastebin.com/s96Bw6Mk

Koenkk, 2020-02-01:
Thanks, try with https://gist.github.com/Koenkk/131ee3e20fdabb6fa094a07f5c247083

bogdaan2001, 2020-02-01:
Now working!!! but only single press action :(

https://pastebin.com/7mVnfc1j

Koenkk, 2020-02-01:
Cool! I think all actions have been added in the latest dev branch (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

bogdaan2001, 2020-02-01:
> Cool! I think all actions have been added in the latest dev branch (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Yes, it's work (but i replace the device.js file again)

Thank you !!!

Koenkk, 2020-02-02:
Great, I will implement this new interview procedure once zigbee2mqtt 1.10 is released (so we have some time to test) (wednesday/thursday) and ask you again to test this.

bogdaan2001, 2020-02-02:
ok

Koenkk, 2020-02-07:
Pushed some fixes to the latest dev branch, can you check and provide the log?

bogdaan2001, 2020-02-08:
> Pushed some fixes to the latest dev branch, can you check and provide the log?

Hi !
How i can install this version ?

Koenkk, 2020-02-08:
https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html

bogdaan2001, 2020-02-08:
https://pastebin.com/TK72pENG

Tiitan, 2020-02-08:
Hi here, I only have the single press recognized. I updated to 1.10 to fix that but it didn't. Then I switched to dev branch (zigbee2mqtt-edge addon on hass.io) and I still only have single press.
@bogdaan2001 you had the same problem last week but I noticed on your last log report that you actually do have double and hold messages now. is it really switching to dev that fixed it for you or did you change something else ?

bogdaan2001, 2020-02-08:
> Hi here, I only have the single press recognized. I updated to 1.10 to fix that but it didn't. Then I switched to dev branch (zigbee2mqtt-edge addon on hass.io) and I still only have single press.
> @bogdaan2001 you had the same problem last week but I noticed on your last log report that you actually do have double and hold messages now. is it really switching to dev that fixed it for you or did you change something else ?
> I'll open a new issue if it is not obvious

no, i only install the dev branch

Koenkk, 2020-02-08:
Looks good, interview is successful now!

closed 2020-02-08 by Koenkk