Koenkk/zigbee2mqtt #2845: SPZB0001 variant not working?

keesj, 2020-01-28:
## Debug Info
zigbee2mqtt version:dev branch

```
2e958f863f22659750ea26e735bc4318d47a8283 (HEAD -> dev, origin/dev)
Author: Koen Kanters <koenkanters94@gmail.com>
Date:   Sat Jan 18 23:05:05 2020 +0100

    Update herdsman.
```

CC253X firmware version:   2020-01-28 19:43:30: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'

## Description

I have previously ordered a SPZB0001 and this one worked mostly fine.As I was happy I now ordered this one https://www.conrad.nl/p/eurotronic-spirit-zigbee-draadloze-radiatorthermostaat-elektronisch-1559294

[image]
[image]
[image]

It looks very similar (and is from the same vendor) but it does not want to pair and remains in Jin

One of the differences I see is that the documented reset procedure is different from the one documented on the https://www.zigbee2mqtt.io/devices/SPZB0001.html side (namely press power  and +) but perhaps.. I don't know

I can not see anything related to pairing but paiting is enable and in fact I have also pair a different device today  (here is the log.. there was an error meesage but still worked)

```
zigbee2mqtt:info  2020-01-28 17:14:36: Successfully interviewed '0x000d6f000d83b3e4', device has successfully been paired
zigbee2mqtt:info  2020-01-28 17:14:36: Device '0x000d6f000d83b3e4' is supported, identified as: Bitron Wireless wall thermostat with relay (AV2010/32)
zigbee2mqtt:info  2020-01-28 17:14:36: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x000d6f000d83b3e4","model":"AV2010/32","vendor":"Bitron","description":"Wireless wall thermostat with relay","supported":true}}'
zigbee2mqtt:info  2020-01-28 17:14:36: Configuring '0x000d6f000d83b3e4'
zigbee2mqtt:error 2020-01-28 17:14:42: Failed to configure '0x000d6f000d83b3e4', attempt 1 (Error: Data request failed with error: 'MAC transaction expired' (240)
    at ZStackAdapter.<anonymous> (/home/why/projects/zigbee/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:584:27)
    at Generator.next (<anonymous>)
    at fulfilled (/home/why/projects/zigbee/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:5:58))
```

sjorge, 2020-01-28:
This is interesting `"model":"AV2010/32","vendor":"Bitron"` it won't match the existing eurotronic's vendor we have...

But it does seem to match another device... I'm not sure how to deal with that... maybe the other device is similar? Or worse, Bitron rebranded a Eurotronic one and reused the vendor/model of one of there other models.

keesj, 2020-01-29:
Hello,

> This is interesting `"model":"AV2010/32","vendor":"Bitron"` it won't match the existing eurotronic's vendor we have...

This I can explain. it is indeed the AV2010/32 that successfully paired just a few minutes before I first tried pairing the Spirit so no "big" problems here! I merely added the log to prove I was accepting pairing and that the log does **not** show the Spirit trying to pair

sjorge, 2020-01-29:
But the failure you showed was for 0x000d6f000d83b3e4, whic hwas the Bitron...

As why it is not even attempting to pair, mine had to have a few tries too.
I basically had to reset it once, then it would pair as normal. So try resetting the TRV and try to join again.

keesj, 2020-01-29:
Hi, 

I had similar problems with the first SPZB0001 (I had to find a way to reset the device) but when in "pairing mode" the reset is not accepted.

SBTephan, 2020-01-30:
Other question,

How are you controlling the bitron video thermostat? with homeassistant or...?
An if so, what did you had to do to make it work?

keesj, 2020-01-30:
Hello,

My plan is to use zigbeemqtt (possibly with home assistant) to determine when the heating needs to be turned on. 
* Use the konke temperature/moisture sensor to measure the quality in the rooms
* Control the temperature in rooms by setting the SPZB0001 target temperature
* Turn on the head production controlling the AV2010
* I also have a mechanical ventilation system that i might want to hook up

At this point I only have one SPZB0001 working and the 3 others do not want to pair (first bought one and because that worked got 3 more but thoses do not pair somehow). I by now tried different usb dongles and installed on different machines and tested the both the dev and master branch. 

Unless some node dependency broken the whole system I see no reason to question my setup but I am not very happy about how it currently behaves :p

redge76, 2020-01-31:
I just got a brand new SPZB0001 and it refused to pair using the batteries supplied in the box (new and still in plastic wrap)
I read somewhere else that the SPZB0001 needed good batteries so I installed brand new "energ1zer" batteries and ... voila... It paired imediately.
The SPZB0001  still does not work... It only pairs but I don't receive anthing. I also can't send any command. I just get some notification that the communication is still alive in debug mode.

keesj, 2020-02-01:
I am willing to put some time in figuring out more on this subject. I will probably need to start by dumping the zigbee data

SpookyFM, 2020-02-02:
Just wanted to check with you here, I was having issues with pairing a SPZB0001. I have 2, one was pairing every time (tested removing it) without issues, the other one I spent the last 2 hours reading up and resetting it. I just had success pairing it almost instantly after changing the zigbee channel to one overlapping with fewer wifi networks. 

However, I'm not sure if this really was the deciding factor, after changing the channel back to the default of 11 the things seem to pair just as well.

keesj, 2020-02-03:
Hi,
I am continuing the research (I have 3 non functional that I .. kinda would like to get working). 

This morning I dumped what the device is sending out(outside of home to get a more clear.. signal).
From using wireshark I see the following:
The device  **is** sending out a beacon request and the dongle **is** sending a response... 
[image]

keesj, 2020-02-05:
Here is a trace with the working device
[image]

and adding the wireshark captures
[caps.zip](https://github.com/Koenkk/zigbee2mqtt/files/4160966/caps.zip)

keesj, 2020-02-23:
After buying a conbee II dongle and also having problems there I updated zigbee2mqtt and gave it an other go.
I am now on the dev branch at commit 409df4c193adcd37c2eef75ac234728ff2e2473b and things all of a sudden all work (no timeouts any more)

notable changes:
* Upgrade of os (Ubuntu eoan) , node , and source code.
* Moved from channel 15 (that I configured for my konke devices) to channel 25
* Learned to reset the Spirit devices (even when not paired) . It turns out you *need* to press the 3 buttons (preferably by starting with the minus button)

Adding a log (with no warnings) and the system is starting to rock!

[log.zip](https://github.com/Koenkk/zigbee2mqtt/files/4242309/log.zip)

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1098
https://tweakers.net/productreview/222266/eurotronic-spirit-thermostaat-kraan-zigbee.html

keesj, 2020-02-23:
closing this issue at it works4me now

closed 2020-02-23 by keesj