Koenkk/zigbee2mqtt #2846: New version of Osram Gardenpole Mini? Unsupported

IcedEarth, 2020-01-28:
Hey,

I have an Osram Gardenpole Mini here at home. However, it appears as an unknown device and is not properly supported. So I did some research and found the zigbeeModel: ['Gardenpole Mini RGBW OSRAM'] in the devices.js with model: 'AC0363900NJ'. However, Zigbee alliance also lists other gardenpoles with the extension Z3 and the models 4058075047853 and 4058075147126.
So i though it would be enough to copy the definition of the AC0363900NJ and change with one of the new model IDs.
Doing this, The device was apparently successfully interview, however, labeled as unsupported.

After going through the tutorial I was not able to figure this out by my own, so I have to leave it to the pro's here.
Here is a log entry for this device: {"id":25,"type":"Router","ieeeAddr":"0xf0d1b800000fd8e4","nwkAddr":56509,"manufId":4489,"epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,2821,64513],"outClusterList":[25],"clusters":{},"binds":[]}

I really though it would be just adding this new device in the devices.js, but this is apparently not the case.
Here is the link to the zigbee alliance page:
https://zigbeealliance.org/zigbee_products/gardenpole-mini-rgbw-z3/

I hope someone can help me out with this. It is also weird, that no one brought this up before, since this device is not really new, i guess.

best regards

Koenkk, 2020-01-29:
Can you post your `database.db`? (I need the `modelId` attribute)

IcedEarth, 2020-01-29:
Here you go. Sorry for not including it in the beginning
`{"id":25,"type":"Router","ieeeAddr":"0xf0d1b800000fd8e4","nwkAddr":4922,"manufId":4489,"manufName":"LEDVANCE","powerSource":"Mains (single phase)","modelId":"Gardenpole RGBW Z3","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,2821,64513],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"Gardenpole RGBW Z3","manufacturerName":"LEDVANCE","powerSource":1,"zclVersion":2,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"","swBuildId":""}}},"binds":[]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[]}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"","swBuildId":"","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1580240604783}`

Koenkk, 2020-01-29:
Should be supported in the latest zigbee2mqtt dev branch.

IcedEarth, 2020-01-30:
Hey again,
tried to pair the device (and use it), but got some problems:
The devide is successfully interviewed:
`{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"Osram_Gardenpole","model":"4058075047853","vendor":"OSRAM","description":"Smart+ gardenpole 4W RGBW","supported":true}}`

However, when i try to send a command I get an timeout error:
`zigbee2mqtt:error 2020-01-30 18:27:58: Publish 'set' 'state' to 'Osram_Gardenpole' failed: 'Error: Timeout - 53125 - 1 - 6 - 6 - 11 after 15000ms'
Jan 30 18:27:58 PiFHEM npm[21415]: zigbee2mqtt:info  2020-01-30 18:27:58: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'Osram_Gardenpole' failed: 'Error: Timeout - 53125 - 1 - 6 - 6 - 11 after 15000ms'","meta":{"friendly_name":"Osram_Gardenpole"}}'
`

Not sure what the problem is though.
The database.js entry says,t he interview was not completed:
`{"id":26,"type":"Router","ieeeAddr":"0xf0d1b800000fd8e4","nwkAddr":19073,"manufId":4489,"manufName":"LEDVANCE","modelId":"Gardenpole RGBW Z3","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,2821,64513],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"Gardenpole RGBW Z3","manufacturerName":"LEDVANCE"}}},"binds":[]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[]}},"interviewCompleted":false,"meta":{},"lastSeen":1580404664538}
`

Any ideas on that?

Koenkk, 2020-01-30:
Should be fixed in latest dev branch.

IcedEarth, 2020-01-30:
Indeed, it is. Thank you. Now I need a stronger transmitter, apperently ;-)
Thanks you again!

closed 2020-01-30 by IcedEarth