Koenkk/zigbee2mqtt #2849: Dimmer IKEA battery status

antti365, 2020-01-29:
## Debug Info
zigbee2mqtt version: 1.9.0-dev
CC253X firmware version: 20190608
 
Hi,
I installed the latest version of the software and the latest stick firmware, connected dimmer ikea and found out that there is no battery value in the transmitted values ​​from the dimmer.  It seems that everything worked on older versions and the value was transferred - has something changed?  From the Xiaomi temperature sensor, the battery value comes without a problem ...

mr2c12, 2020-01-29:
I am facing the same problem with an Ikea Remote (e1524). The battery percentage disappeared at some point.
Even when I had to restart from zero (changed coordinator fw, repaired every device) this has not been solved.

I am pretty sure it used to work.

antti365, 2020-01-29:
I’ll check the switch E1743 in the evening, I’d like to check the motion sensor, I don’t have them unfortunately ...

mr2c12, 2020-01-29:
[image]

E1743 and motion sensor are currently reporting the battery percentage in my network.

antti365, 2020-01-29:
All the same, the dimmer came to his senses and sent me the battery value, before that it was necessary to wait for almost a day, apparently nothing has changed since then ... :)

Koenkk, 2020-01-30:
@antti365 it indeed will take up to 1 day for the device to report it's battery level so this is expected.

closed 2020-01-30 by Koenkk